lvgl/lvgl #7386: `test_image` Failing Intermittently in CI Tests

vwheeler63, 2024-12-01:
### LVGL version

v9.3

### Platform

Ubuntu Linux -- the CI test platform

### What happened?

In my last 8 or so pushes to PRs (new and existing), `test_image` runs as part of the CI test called `C/C++ CI / Build ESP IDF ESP32S3 (pull_request)`, and it has been failing in about 25-40% of the cases.  When it fails, simply pushing a dummy commit causes it to run again, and in more than half the cases so far, it has succeeded the next time it runs, when the change in the commit was not related to what it tests.  I don't know what causes that, but I am intensely interested since it is interfering with production.

### How to reproduce?

Just push any commits to any PRs -- new or existing, and note that this test will fail in approximately the above % of cases.

closed 2024-12-02 by liamHowatt

liamHowatt, 2024-12-02:
Let's see if it's fixed after merging #7387. If you still see it happening, please reopen the issue.

> I don't know what causes that, but I am intensely interested since it is interfering with production, plus I am very interested to know how it can be that a test is not consistent when running in a digital environment.

The sanitizers do not always catch the bug every run. If there's an intermittent CI failure it's usually because there's a memory access violation somewhere and the instrumentation only catches it some of the time, but there is a real bug there and it's there every run, whether or not it's caught.

vwheeler63, 2024-12-03:
> Let's see if it's fixed after merging #7387. If you still see it happening, please reopen the issue.

Will do!

> The sanitizers do not always catch the bug every run.

What is a "sanitizer"?  I found these 2 things which appear to be really nice tools:

https://github.com/google/sanitizers/wiki/AddressSanitizerLeakSanitizer

https://github.com/google/sanitizers/wiki/AddressSanitizer

Are we using these in the CI tests?

liamHowatt, 2024-12-03:
Yes, those are them. Enabled [here](https://github.com/lvgl/lvgl/blob/5bccba9946440b30c4f369a891a0a5a023ac5146/tests/CMakeLists.txt#L61-L63).

It can catch out-of-bounds reads in addition to writes.

```
...
AddressSanitizer: heap-buffer-overflow
...
READ of size 71
...
located 0 bytes to the right of 8531-byte region
...
```

https://github.com/lvgl/lvgl/actions/runs/12106478635/job/33752268846

I really can't pretend to know how it works. I think it uses a special `memcpy` that knows what memory should and should not be accessed by a correct program.

PR #7387 (2024-12-02): fix(draw_sw): wrong image buffer calculation
Fix #7386

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code-format.py` (`astyle v3.4.12` nee [... cut]