lvgl/lvgl #7390: Problem with exmaple from 9.0.0 version keep restarting

Matys-check, 2024-12-01:
### LVGL version

LVGL 9.0.0

### Platform

Arudino IDE esp32 3248s035c YCD

### What happened?

Hi, I'm a new user who has been struggling for a few days to implement a simple program that will allow me to fully utilize the device I have. I would like to write a simple example where I can click on the screen. I've been jumping between different versions of LVGL because that's what the tutorials use. The most examples I found are for version 9.0.0, but I'm unable to even compile a simple example for that version.

21:07:53.813 -> rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
21:07:53.813 -> configsip: 0, SPIWP:0xee
21:07:53.813 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:07:53.813 -> mode:DIO, clock div:1
21:07:53.813 -> load:0x3fff0030,len:1344
21:07:53.813 -> load:0x40078000,len:13964
21:07:53.813 -> load:0x40080400,len:3600
21:07:53.813 -> entry 0x400805f0
21:07:53.984 -> E (337) esp_core_dump_f�f�� Core dump data check failed:
21:07:53.984 -> Calculated checksum='57b8c3c0'
21:07:53.984 -> Image checksum='ffffffff'

### How to reproduce?

I just took the example from:
https://github.com/lvgl/lvgl/blob/release/v9.0/examples/arduino/LVGL_Arduino/LVGL_Arduino.ino
[lv_conf.txt.txt](https://github.com/user-attachments/files/17970220/lv_conf.txt.txt)
[User_options.txt](https://github.com/user-attachments/files/17970221/User_options.txt)

liamHowatt, 2024-12-02:
I understand your frustration. Are you able to increase the stack size? It may be a stack overflow. It's hard to know without more information about the crash. Can you add `Serial.println` logs to find out where/when it crashes?

Matys-check, 2024-12-02:
@liamHowatt i get now something like this:
16:54:43.161 -> Hello Arduino! V9.0.0
16:54:43.161 -> 
16:54:43.161 -> ***ERROR*** A stack overflow in task loopTask has been detected.
16:54:43.194 -> 
16:54:43.194 -> 
16:54:43.194 -> Backtrace: 0x40083571:0x3ffae440 0x40088299:0x3ffae460 0x4008b1ad:0x3ffae480 0x40089b9f:0x3ffae500 0x400883a8:0x3ffae530 0x40088358:0x00000000 |<-CORRUPTED

[image]

liamHowatt, 2024-12-19:
Sorry for the delay. It is a stack overflow. Search "arduino esp32 increase loop task stack size"

I found some potential ways to increase the stack size here: https://esp32.com/viewtopic.php?t=28250

closed 2025-01-17 by lvgl-bot