lvgl/lvgl #7393: When there are too many data points or the screen window size is not sufficient, the line width in a line chart may not be rendered correctly

lhdjply, 2024-12-02:
### LVGL version

master

### Platform

vscode SDL on linux

### What happened?

https://github.com/user-attachments/assets/9ce84946-7230-44d5-a2c5-c5db6f169e81

### How to reproduce?

#include "lvgl/lvgl.h"

#define CHART_POINT_COUNT 800
void chart_test(void)
{
  lv_obj_t *chart = lv_chart_create(lv_screen_active());
  lv_obj_set_size(chart, lv_pct(100), lv_pct(100));
  lv_chart_set_point_count(chart, CHART_POINT_COUNT);
  lv_obj_set_style_line_width(chart, 3, LV_PART_ITEMS);
  lv_chart_set_div_line_count(chart, 11, 11);
  lv_obj_center(chart);

  lv_chart_series_t *ser_1= lv_chart_add_series(chart, lv_palette_main(LV_PALETTE_RED), LV_CHART_AXIS_PRIMARY_Y);

  lv_chart_set_range(chart, LV_CHART_AXIS_PRIMARY_Y, -100, 100);

  for(uint16_t i=0; i< CHART_POINT_COUNT; i++)
  {
    lv_chart_set_next_value(chart, ser_1, 0);
  }
}

int main(int argc, char **argv)
{
  lv_display_t *display;
  lv_init();

  display = lv_sdl_window_create(800, 480);
  lv_sdl_window_set_title(display, "LVGL Simulator");
  lv_sdl_mouse_create();

  chart_test();

  while (1)
  {
    uint32_t time_until_next = lv_timer_handler();
    lv_delay_ms(time_until_next);
  }

  return 0;
}

liamHowatt, 2024-12-02:
The effect you're seeing is a result of [this](https://docs.lvgl.io/master/details/widgets/chart.html#handling-large-numbers-of-points):

> #### Handling large numbers of points
>
> On LINE charts, if the number of points is greater than the pixels horizontally, the Chart will draw only vertical lines to make the drawing of large amount of data effective. If there are, let's say, 10 points to a pixel, LVGL searches the smallest and the largest value and draws a vertical lines between them to ensure no peaks are missed.

When N points > screen width it switches to a mode where it draws vertical lines instead of horizontal.

One solution in `lv_chart.c` would be to use the line width property to extend the vertical lines up and down an amount that would make it thicker, but it would be misrepresent data on the chart since the lines would extend past the correct position.

lhdjply, 2024-12-03:
Even when drawing a vertical line, shouldn't the correct line width be maintained?

liamHowatt, 2024-12-03:
The width is being maintained, but the width of vertical lines extends horizontally. A width-10, length-2 vertical line is 10 pixels wide and 2 pixels high.

closed 2024-12-12 by lhdjply