lvgl/lvgl #7394: CMake install fails to find lv_conf.h

zylalx1, 2024-12-02:
### LVGL version

v9.3.0

### Platform

Windows / Linux

### What happened?

CMake install fails to find lv_conf.h

if i run CMake with LV_CONF_SKIP=OFF and LV_CONF_PATH not defined, the building process is ok but the following error occurs in the installing process:

```
...
-- Installing: (something)/install/include/lvgl/lvgl.h
CMake Error at build/cmake_install.cmake:50 (file):
  file INSTALL cannot find "(something)/lvgl/lv_conf.h": No such file or directory.
```

i think the reason may be this part of code in /env_support/custom.cmake

```
if(NOT LV_CONF_SKIP)
  if (LV_CONF_PATH)
    list(APPEND LVGL_PUBLIC_HEADERS
    ${LV_CONF_PATH})
  else()
    list(APPEND LVGL_PUBLIC_HEADERS
    "${CMAKE_SOURCE_DIR}/lv_conf.h")
  endif()
endif()
```

the code tries to find "lv_conf.h" in the "lvgl/" directory but it isn't there.
i guess the correct logic may be:

```
if(NOT LV_CONF_SKIP)
  if (LV_CONF_PATH)
    list(APPEND LVGL_PUBLIC_HEADERS
    ${LV_CONF_PATH})    # we assume it's at the specified path
  elseif(LV_CONF_INCLUDE_SIMPLE)
    list(APPEND LVGL_PUBLIC_HEADERS
    "${CMAKE_SOURCE_DIR}/src/lv_conf.h")    # we assume it's in the "lvgl/src/" directory
  else()
    list(APPEND LVGL_PUBLIC_HEADERS
    "${CMAKE_SOURCE_DIR}/../lv_conf.h")    # we assume it's in the "lvgl/../" directory
  endif()
endif()
```

this solves the problem but another issue remains: 
"lv_conf.h" is always installed to the "include/lvgl/" directory, not the corresponding directory specified by the LV_CONF_PATH or LV_CONF_INCLUDE_SIMPLE

for LV_CONF_INCLUDE_SIMPLE we can install to either "include/lvgl/src" or "include/" based on the value
but for LV_CONF_PATH it's a little bit complex since it can be anything.
if it's an absolute path, nothing to do since it's already there.
if it's a relative path, the final destination may be outside the "include/" directory. i'm not sure if it's a good idea to copy a file to a path that is outside our control. 

any suggestions?

### How to reproduce?

```
cmake -S./lvgl -B./build -G"Ninja Multi-Config" -DBUILD_SHARED_LIBS=ON -DLV_CONF_INCLUDE_SIMPLE=OFF -DLV_LVGL_H_INCLUDE_SIMPLE=OFF -DLV_CONF_SKIP=OFF
cmake --build ./build
cmake --install ./build --config debug --prefix ./install
```

kisvegabor, 2024-12-22:
Hi,

Sorry the late reply. @etag4048 could you take a look at it? It seems like something that we need to take care of in https://github.com/lvgl/lvgl/issues/7223

kisvegabor, 2025-01-09:
@etag4048 gentle ping

etag4048, 2025-01-17:
Hello @zylalx1  Thank you for the bug report,
thanks for your proposal and the detailed replication steps, but could you please explain your use case in more detail ?

I suppose you have the ``lv_conf.h`` configuration file, adjacent to the ``./lvgl`` directory ?

I am trying to understand what you are trying to achieve with this command :
```
cmake -S./lvgl -B./build -GNinja Multi-Config -DBUILD_SHARED_LIBS=ON -DLV_CONF_INCLUDE_SIMPLE=OFF -DLV_LVGL_H_INCLUDE_SIMPLE=OFF -DLV_CONF_SKIP=OFF
```

You are setting the directory of the library itself as the source directory.
Not using ``LV_CONF_INCLUDE_SIMPLE``, not defining a custom ``LV_CONF_PATH`` and also setting ``LV_CONF_SKIP=OFF``

Also please note that dynamic linking is not yet fully supported and not documented,
it is currently being worked on - there are still many problems to solve...

I tried the command, and I deduce that you must have the ``lv_conf.h`` file adjencent to the ``lvgl`` directory, otherwise you would have a preprocessor error at ``lv_conf_internal.h:60`` where it includes "../../lv_conf.h"

Instead of modifying the ``custom.cmake``, I would suggest
that you move ``lv_conf.h`` into the ``lvgl/`` directory since the your source dir is LVGL

Now your command will run and everything works, ``lv_conf.h`` is now present in the source directory and in ``lv_conf_internal.h:48`` ``LV_CONF_INCLUDE_SIMPLE`` will get defined.

so the command can even be shortened to:
`
cmake -S./lvgl -B./build -GNinja Multi-Config -DBUILD_SHARED_LIBS=ON -DLV_CONF_SKIP=OFF`

The installation is successful but there is a catch

once you try to include ``<lvgl/lvgl.h>`` you will get errors because ``lv_conf_internal.h``
will expect it in ``{prefix}/include`` and not in ``{prefix}/include/lvgl`` were it will end up.

> "lv_conf.h" is always installed to the "include/lvgl/" directory, not the corresponding directory specified by the LV_CONF_PATH or LV_CONF_INCLUDE_SIMPLE

``LV_CONF_INCLUDE_SIMPLE`` is not used to specify a directory where to install ``lv_conf.h``
When defined, ``lv_conf_internal.h`` will include ``lv_conf.h`` using quotes ``#include "lvgl.h"`` the developer then passes an extra include directory via the ``-I`` argument to the compiler or with the ``target_include_directories`` directive in the project's ``CMakeLists.txt`` file.

``LV_CONF_PATH`` is also **not used to specify the destination** of ``lv_conf.h`` but rather where to find. And it must be double quoted value or there will be an error in ``lv_conf_internal.h`` It is then copied by cmake during installation to the prefix directory so specifying a relative or absolute path doesn't matter, it will not end up in outside of the directory specified by the cmake prefix.

zylalx1, 2025-01-22:
Hello @etag4048 and thanks for your reply! 

> I suppose you have the lv_conf.h configuration file, adjacent to the ./lvgl directory ?

Yes. Because I think this is the recommended practice in the doc ([https://docs.lvgl.io/master/intro/add-lvgl-to-your-project/configuration.html](url)), and should be supported. By separating config file (provided by the user) from code (provided by the lvgl team), it achieves a clearer project structure and allows easy replacement/update of either part. (Or it will require additional efforts on code management, such as editing the `.gitignore` file. )

So I think maybe we can modify the `custom.cmake` to support this practice by default?

For other practices, seems like I misunderstood the use of `LV_CONF_INCLUDE_SIMPLE` and `LV_CONF_PATH`. I read the doc again, and hope I got it right this time. 

`LV_CONF_INCLUDE_SIMPLE` does not specify the path to `lv_conf.h`, but changes the code to `#include "lv_conf.h"` so the compiler will try to find it under the given include paths. 
Putting the `lv_conf.h` in the `/lvgl` directory may be the most common practice when defining `LV_CONF_INCLUDE_SIMPLE`, but we can not assume it, since the user can put the `lv_conf.h` anywhere as long as the compiler is able to find it under one of the include paths. So where to find the `lv_conf.h` on installation is a problem. 
Also we can not assume its destination the user want is the `include/lvgl` directory under the installation path. If that path is not in the include path when compiling the executable (that relies lvgl library), `lv_conf_internal.h` will throw an error, like you mentioned. 

Similar situation when user defines `LV_CONF_PATH`. 

In my opinion, if the user defines `LV_CONF_INCLUDE_SIMPLE` or `LV_CONF_PATH`, it probably means he has his own idea on where to put and how to include the config file. So in that case maybe we'd better skip the copying of the config file on installation, and show a message to inform the user that he has to manage the path and include of the config file manually?

closed 2025-02-15 by lvgl-bot

krzysobo, 2025-07-03:
When trying to build LVGL 9.3, I've had a similar problem with lv_internal_conf.h. Thanks to Grok and my persistence, I've found out that:
- adding DLV_LVGL_H_INCLUDE_SIMPLE to definitions:
add_definitions(-DLV_LVGL_H_INCLUDE_SIMPLE)

...and simply adding quotation marks around the path definition in CMakeLists.txt solved the problem with lv_internal_conf.h,  since the latter file requested  "path" (string) or <path> and without quotation marks it wasn't treated as a string:
# lack of quotation marks "" below was the source of the problem in lv_internal_conf.h:
add_definitions(-DLV_CONF_PATH="${LV_CONF_PATH}")   # <-- this is OK
add_definitions(-DLV_CONF_PATH=${LV_CONF_PATH})   # <-- this was NOT OK

Anyway, it seems that the problem described by this topic's author still remains. I made a workaround by soft linking the lv_conf.h (which I have in the upper directory), to the application code, ie:
app -> my application
lvgl -> lvgl library
lv_conf.h - the config that I linked as app/lv_conf.h

vwheeler63, 2025-07-03:
@AndreCostaaa , I wonder if this is the same issue we were just discussing...

AndreCostaaa, 2025-07-04:
Hi @krzysobo 

A couple of questions:

1. What is your exact project structure? 
2. Are you also building your application using CMake or just LVGL ? If so, where are the other CMakeLists.txt?
3. Where did you add this ?

```cmake
add_definitions(-DLV_CONF_PATH=${LV_CONF_PATH})
```