lvgl/lvgl #7400: Docs feedback

kisvegabor, 2024-12-03:
### Introduce the problem

We got 2 feedback about our docs n the Forum:
- https://forum.lvgl.io/t/setup-lvgl-in-a-simulator-with-4-clicks-in-2-minutes/11317/20
- https://forum.lvgl.io/t/simple-hello-world-first-lvgl-program/4697/5

@vwheeler63 do you see a way to resolve these?

vwheeler63, 2024-12-03:
Hi, Gábor!  ( @kisvegabor )

On the first one, I'm not sure exactly what the issue is.  Can you tell me a bit more?  Is he trying to work with the `lv_port_pc_visual-studio` repository?  I do remember going through that recently and having to use some knowledge I already had in order to get it working....  Which could be handled by adding a step-by-step setting-it-up guide, but I would want to be addressing the correct problem.

On the 2nd one, that is interesting!  I find myself wishing I could be in the same room with `AnalogWeapon` and be there as he runs into barriers and asks questions, as it would be highlight what is missing.  If I am understanding this correctly, he is looking for something that applies to his circumstances + his knowledge, but I am missing what his circumstances are.  Or if he is just looking for something called "Getting Started" in which case the "Going Deeper" button (and section it leads to) might be re-named that, since it is about jumping in and starting to work with LVGL code.

kisvegabor, 2024-12-03:
In case of the first one please check out the [first comment of that topic](https://forum.lvgl.io/t/setup-lvgl-in-a-simulator-with-4-clicks-in-2-minutes/11317?u=kisvegabor). It's about GitHub Codespaces which is the simplest was to try LVGL. 

Regarding the second: could you ask him on the Forum to better understand his use case?

michaesc, 2024-12-04:
The [LVGL Basics page](https://docs.lvgl.io/latest/en/html/intro/basics.html#basic-examples) is great for many things, because it supports **both visual output as well as mouse input** for a variety of widgets.

Copying the _Show C code_ fragments is very useful, but...

There is no next step to this workflow on getting started, due to lack of documentation with relevance to a modern (2024) Debian based distribution.

It's easy enough to get visual output once a _lv_port_pc_linux build_ completes, but more or less no chance at all for a beginner to have input working when `bin/lvglsim` runs.

michaesc, 2024-12-12:
Is there any progress on documenting the method of getting a linux simulator working?

As stated above, the current documentation does not work with systems since about 2024.

vwheeler63, 2024-12-12:
> Is there any progress on documenting the method of getting a linux simulator working?
> 
> As stated above, the current documentation does not work with systems since about 2024.

This certainly is relevant, and I'm happy to update that basics page.  @kisvegabor, what would be the most efficient way to add what a Debian user will need to the LVGL Basics page?  Do you want me to set up a Debian WSL and work through setting up LVGL and document it?

vwheeler63, 2024-12-13:
Hi, Gábor!  ( @kisvegabor )

Sorry for the long delay.

Re item number 2 -- for sure we could have a short page that has a full set of code to get started.  (There are some things the user would have to fill in, like the details of pushing pixels to the display panel, since that hardware will vary, but I could use an actual example of that from one of my projects.)  I LOVE the idea!!

Since "getting started" is not now used as a title in the documentation, it would seem an appropriate title for such a page.  Especially since AnalogWeapon appears to have been searching for it.

By the way, is there a mapping from `https://forum.lvgl.io/u/AnalogWeapon` to a GitHub ID anywhere?

vwheeler63, 2024-12-13:
I'm guessing we're going to need some really good documentation for the LVGL Editor too....

kisvegabor, 2024-12-17:
Maybe "LVGL Basics" Could be restructured to be very condensed and contain practical information about how to have get the first project up an running. I imagine it without any links to further reading, just very targeted: do this, and see LVGL running on 
1. PC
2. Embedded hardware

I think the [project creator](https://lvgl.io/tools/project-creator) would be very useful here. We will work on it again in January to clean things up add more projects.

vwheeler63, 2024-12-17:
> Maybe "LVGL Basics" Could be restructured to be very condensed and contain practical information about how to have get the first project up an running.

I like the idea.  If there were no links, it would at least need a paragraph at the top telling the user why there are no links and how to find further details.

vwheeler63, 2025-01-03:
not stale

vwheeler63, 2025-01-26:
not stale

michaesc, 2025-01-26:
I would love to help reduce the confusion caused by lack of documentation and broken outdated interfaces causing failure with desktop simulation of embedded platform(s.) I also respect everyone's (at LVGL) work a lot, and I know most of you don't have time.

I didn't find any results of @kisvegabor 's January Project Creator design, but when using the mid October (1.1.1) release I can set up three simulator projects using cmake. Building with the included makefile fails on all three, but when building the Cmake way (mkdir buildit && cd buildit && make) two of the three (lv_port_linux framebuffer and SDL) actually successfully build.

Unfortunately running bin/lvglsim displays nothing, I guess the Linux framebuffer and SDL are not properly initialising or something else is wrong. Please confirm that running bin/lvglsim should display an example LVGL application?

username@host:~/projectcreat/lv_port_linux$ ./bin/lvglsim
[Warn]  (0.000, +0)      lv_init: Memory integrity checks are enabled via LV_USE_ASSERT_MEM_INTEGRITY which makes LVGL much slower lv_init.c:280
[Warn]  (0.000, +0)      lv_init: Object sanity checks are enabled via LV_USE_ASSERT_OBJ which makes LVGL much slower lv_init.c:284
[Warn]  (0.000, +0)      lv_init: Style sanity checks are enabled that uses more RAM lv_init.c:288

The OS is a fully updated Ubunto with all the needed development libraries installed.

I run the binary from a command line while having all the needed permissions:

uid=****(username) gid=1001(username) Gruppen=1001(username),4(adm),5(tty),10(uucp),20(dialout),24(cdrom),27(sudo),30(dip),44(video),46(plugdev),50(staff),100(users),121(lpadmin),132(lxd),133(sambashare),137(libvirt)

So I will keep waiting until a provisional document or page appears that gives clear instruction on how to install and configure a desktop simulation environment that really works with the latest LVGL technology. Thank you!

vwheeler63, 2025-01-28:
Hi, @michaesc !  Good write-up.  Thank you!  @kisvegabor , what do you see as the best way to address the above?  I don't (yet) have the expertise on the Linux side to take this on, but I certainly could add it to my list for Windows if needed.

kisvegabor, 2025-01-28:
Although the project creator is released, internally with @liamHowatt, @uLipe, and @fbiego we are working on putting the current projects in shape and advertising the project creator more.

 > I can set up three simulator projects using cmake. Building with the included makefile fails on all three, but when building the Cmake way (mkdir buildit && cd buildit && make) two of the three (lv_port_linux framebuffer and SDL) actually successfully build.

It should be an issue in the projects themselves. Which repos where these exactly?

michaesc, 2025-01-28:
I agree that the problem is in the project(s) themselves, but also related to the lack of current documentation as well as the project creator.

The repository in question is **lv_port_linux** and I assume you @kisvegabor will ask me to research and post a bug report in that repository to continue this discussion. Is that what is preferred in this case?

By the way when choosing _LVGL on Linux_, the three projects created by project creator depending on the three display driver choices all differ their lv_config files very little:

LV_USE_LINUX_FBDEV
LV_USE_LINUX_DRM
LV_USE_SDL

...so I guess it's not a good idea to post any bug report to the **lvgl_project_creator** repository.

We have three repositories (lvgl, lv_port_linux, and lvgl_project_creator) that are relating to the simulation problems described in this thread.

etag4048, 2025-01-29:
@michaesc Help is very much welcome, documentation is important. I am currently cleaning up ``lv_port_linux`` but it has been a secondary task due to lack of time. My goal is to have support for multiple backends, the current implementation of lv_port_linux only supports one backend at time. And you have to recompile it. 

The end goal of this refactoring, would be also to fix the installation problems, so that ``lv_port_linux`` will generate shared objects that are then installed on the system.   This would remove the need to lvgl as submodule when writing programs, this is the first step to creating packages for popular GNU/Linux distributions.

kisvegabor, 2025-02-04:
> The repository in question is lv_port_linux and I assume you @kisvegabor will ask me to research and post a bug report in that repository to continue this discussion. Is that what is preferred in this case?

You know me very well. 😊 Yes, if it's related to `lv_port_linux` let's continue the discussion there.

closed 2025-03-02 by lvgl-bot