lvgl/lvgl #7402: LVGL Lottie draw buffer cannot allocate from external memory esp32

MR-VENOM, 2024-12-03:
### LVGL version

9.2.2

### Platform

ESP_IDF , eps32s3 with 8MB external psram

### What happened?

The lv_lottie_set_buffer is not able to set more than 180 *180 buffer even though psram is enabled and set to allocate by malloc()

### How to reproduce?

### Code :

```

lvgl_port_cfg_t lvgl_port_cfg = {
        .task_priority = CONFIG_BSP_DISPLAY_LVGL_TASK_PRIORITY,
        .task_stack = 6144,
        .task_affinity = 1,
        .task_max_sleep_ms = CONFIG_BSP_DISPLAY_LVGL_MAX_SLEEP,
        .timer_period_ms = CONFIG_BSP_DISPLAY_LVGL_TICK,
    };
    ESP_ERROR_CHECK(lvgl_port_init(&lvgl_port_cfg));

    bsp_display_config_t bsp_disp_cfg = {
        .max_transfer_sz = BSP_LCD_DRAW_BUFF_SIZE * sizeof(uint16_t),
    };
    ESP_ERROR_CHECK(bsp_display_new(&bsp_disp_cfg, &this->panel_handle, &this->io_handle));

    esp_lcd_panel_disp_on_off(this->panel_handle, true);

    lvgl_port_display_cfg_t disp_cfg = {
        .io_handle = this->io_handle,
        .panel_handle = this->panel_handle,
        .buffer_size = BSP_LCD_DRAW_BUFF_SIZE * 4,
        .double_buffer = 1,
        .hres = BSP_LCD_H_RES,
        .vres = BSP_LCD_V_RES,
        .monochrome = false,
        /* Rotation values must be same as used in esp_lcd for initial settings of the screen */
        .rotation = {
            .swap_xy = false,
            .mirror_x = false,
            .mirror_y = false,
        },
        .flags = {
            .buff_dma = false,
            .buff_spiram = true,
#if LVGL_VERSION_MAJOR >= 9
            .swap_bytes = (BSP_LCD_BIGENDIAN ? true : false),
#endif
        }
    };
    
    this->disp = lvgl_port_add_disp(&disp_cfg);
    assert(this->disp);

    bsp_display_backlight_on();

    this->gfx_buffer = (uint8_t *)heap_caps_malloc(240 * 240 * 4, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
    assert(this->gfx_buffer != nullptr);
    bsp_display_lock(0);
    this->lottie_obj = lv_lottie_create(lv_screen_active());
    bsp_display_unlock();
    
    
    bsp_display_lock(0);
    lv_lottie_set_src_file(this->lottie_obj, path.c_str());
    lv_lottie_set_buffer(this->lottie_obj, 10, 10, this->gfx_buffer);
    lv_obj_center(this->lottie_obj);
    bsp_display_unlock();
    

```

### SDKCONFIG
[sdkconfig.txt](https://github.com/user-attachments/files/17990640/sdkconfig.txt)

kisvegabor, 2024-12-03:
It seems more like an ESP issue, but maybe @uLipe @fbiego or @liamHowatt will have some idea.

uLipe, 2024-12-06:
@kisvegabor yes, even though the PSRAM is enabled on ESP32, the allocation from there via malloc is not the default option, one of the ways of dealing to it is to replace the plain malloc to the esp-idf specifc function `heap_caps_malloc` and pass `MALLOC_CAP_SPIRAM` as the capabillty desired.

MR-VENOM, 2024-12-07:
@uLipe 

> it is to replace the plain malloc to the esp-idf specifc function `heap_caps_malloc` and pass `MALLOC_CAP_SPIRAM` as the capabillty desired.

I am already allocating it using heap_caps_malloc see here 

> this->gfx_buffer = (uint8_t *)heap_caps_malloc(240 * 240 * 4, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
> assert(this->gfx_buffer != nullptr);

uLipe, 2024-12-07:
Please try without the MALLOC_CAP_8BIT , it will force the multi heap to allocate only from SPIRAM and reject other potentially source that offer similar caps.

MR-VENOM, 2024-12-08:
@uLipe my problem is not with allocating buffer for it the heapsmalloc works fine and able.to allocate 240*240*4 bytes . The problem is when I call  

> lv_lottie_set_buffer(this->lottie_obj, 240, 240, this->gfx_buffer);

MR-VENOM, 2024-12-08:
@uLipe 

## LOG

```

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x403835c1  PS      : 0x00060e33  A0      : 0x80388dda  A1      : 0x3fca2840  
--- 0x403835c1: spinlock_acquire at /home/nikhi/esp/v5.2.3/esp-idf/components/esp_hw_support/include/spinlock.h:103
 (inlined by) xPortEnterCriticalTimeout at /home/nikhi/esp/v5.2.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:474

A2      : 0x00000031  A3      : 0xffffffff  A4      : 0x3fca75b4  A5      : 0x00060e23  
A6      : 0x00000000  A7      : 0x0000cdcd  A8      : 0x00060e20  A9      : 0x3fca2840  
A10     : 0x00060e23  A11     : 0x00000000  A12     : 0x00060e20  A13     : 0x3fca2d4c  
A14     : 0x00000001  A15     : 0x000000e7  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000031  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff  
--- 0x400570e8: memset in ROM
0x400570f3: memset in ROM

Backtrace: 0x403835be:0x3fca2840 0x40388dd7:0x3fca2870 0x40388e16:0x3fca2890 0x4037a71d:0x3fca28b0 0x4037a74a:0x3fca28d0 0x4037a4ec:0x3fca2900 0x4038a631:0x3fca2920 0x420384a5:0x3fca2940 0x4203867e:0x3fca2960 0x420387d9:0x3fca2980 0x420397f7:0x3fca7720 0x42037629:0x3fca7750 0x4203b036:0x3fca77a0 0x42037c37:0x3fca77c0 0x42037caf:0x3fca77e0 0x4202dffe:0x3fca7810 0x4202e465:0x3fca7850 0x4202e2aa:0x3fca7900 0x4202e486:0x3fca7950 0x4202e2aa:0x3fca7a00 0x4202e486:0x3fca7a50 0x4202e583:0x3fca7b00 0x4202e4a7:0x3fca7b80 0x4205d6a5:0x3fca7c30 0x4205d93d:0x3fca7c70 0x4205d95f:0x3fca7ca0 0x420231bf:0x3fca7cd0 0x4204b171:0x3fca7d00 0x42008efa:0x3fca7d30 0x42008dd0:0x3fca7d50 0x421220cf:0x3fca7d90 0x403833d1:0x3fca7dc0
--- Error: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
--- Waiting for the device to reconnect.
--- 0x403835be: xt_utils_get_raw_core_id at /home/nikhi/esp/v5.2.3/esp-idf/components/xtensa/include/xt_utils.h:51
 (inlined by) spinlock_acquire at /home/nikhi/esp/v5.2.3/esp-idf/components/esp_hw_support/include/spinlock.h:87
 (inlined by) xPortEnterCriticalTimeout at /home/nikhi/esp/v5.2.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:474
0x40388dd7: xPortEnterCriticalTimeoutSafe at /home/nikhi/esp/v5.2.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:581
 (inlined by) vPortEnterCriticalSafe at /home/nikhi/esp/v5.2.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:588
 (inlined by) multi_heap_internal_lock at /home/nikhi/esp/v5.2.3/esp-idf/components/heap/multi_heap.c:165
0x40388e16: multi_heap_malloc_impl at /home/nikhi/esp/v5.2.3/esp-idf/components/heap/multi_heap.c:209
0x4037a71d: heap_caps_malloc_base at /home/nikhi/esp/v5.2.3/esp-idf/components/heap/heap_caps_base.c:131
0x4037a74a: heap_caps_realloc_base at /home/nikhi/esp/v5.2.3/esp-idf/components/heap/heap_caps_base.c:159
0x4037a4ec: heap_caps_realloc_default at /home/nikhi/esp/v5.2.3/esp-idf/components/heap/heap_caps.c:151
0x4038a631: realloc at /home/nikhi/esp/v5.2.3/esp-idf/components/newlib/heap.c:34
0x420384a5: _genSpan(SwRleData*, SwSpan const*, unsigned long) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRle.cpp:317
0x4203867e: _sweep(RleWorker&) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRle.cpp:434 (discriminator 1)
0x420387d9: rleRender(SwRleData*, SwOutline const*, SwBBox const&, bool) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRle.cpp:994
0x420397f7: shapeGenStrokeRle(SwShape*, tvg::RenderShape const*, tvg::Matrix const*, SwBBox const&, SwBBox&, SwMpool*, unsigned int) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwShape.cpp:617
0x42037629: SwShapeTask::run(unsigned int) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRenderer.cpp:170
0x4203b036: tvg::TaskSchedulerImpl::request(tvg::Task*) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgTaskScheduler.cpp:189
 (inlined by) tvg::TaskScheduler::request(tvg::Task*) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgTaskScheduler.cpp:217
0x42037c37: tvg::SwRenderer::prepareCommon(SwTask*, tvg::RenderTransform const*, tvg::Array<void*> const&, unsigned char, tvg::RenderUpdateFlag) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRenderer.cpp:765
0x42037caf: tvg::SwRenderer::prepare(tvg::RenderShape const&, void*, tvg::RenderTransform const*, tvg::Array<void*>&, unsigned char, tvg::RenderUpdateFlag, bool) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgSwRenderer.cpp:809
0x4202dffe: tvg::Shape::Impl::update(tvg::RenderMethod*, tvg::RenderTransform const*, tvg::Array<void*>&, unsigned char, tvg::RenderUpdateFlag, bool) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgShape.h:110
0x4202e465: tvg::Paint::Impl::update(tvg::RenderMethod*, tvg::RenderTransform const*, tvg::Array<void*>&, unsigned char, tvg::RenderUpdateFlag, bool) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs/thorvg/tvgPaint.cpp:293 (discriminator 2)
0x4202e2aa: tvg::Scene::Impl::update(tvg::RenderMethod*, tvg::RenderTransform const*, tvg::Array<void*>&, unsigned char, tvg::RenderUpdateFlag, bool) at /home/nikhi/disk/pc_ai_acc/managed_components/lvgl__lvgl/src/libs [... cut ...]
0x403833d1: vPortTaskWrapper at /home/nikhi/esp/v5.2.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: c5856b30c

--- Error: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
--- Waiting for the device to reconnect.
--- Error: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
--- Waiting for the device to reconnect.
--- Error: device reports readiness to read but 

```

uLipe, 2024-12-11:
@MR-VENOM What is the PSRAM size of your target?

You are suffering a Load Prohibited exception, on xtensa based chips this occurs very often when trying to load data to invalid memory area, or bad memory size transaction, but PSRAM accepts byte boundart access. So for some reason after setting this buffer the address obstined from the PSRAM seems to be invalid.

MR-VENOM, 2024-12-11:
> @MR-VENOM What is the PSRAM size of your target?

8MB psram. IC ESP32N8R8.

> You are suffering a Load Prohibited exception, on xtensa based chips this occurs very often when trying to load data to invalid memory area, or bad memory size transaction, but PSRAM accepts byte [...]

I know what the error is but whys is it happening that my concern about this :cry:

MR-VENOM, 2024-12-11:
@uLipe Here is my heap log for the same.

```

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x568
load:0x403c9700,len:0x4
load:0x403c9704,len:0xb5c
load:0x403cc700,len:0x2d10
entry 0x403c98f0
I (36) cpu_start: Multicore app
I (37) octal_psram: vendor id    : 0x0d (AP)
I (37) octal_psram: dev id       : 0x02 (generation 3)
I (40) octal_psram: density      : 0x03 (64 Mbit)
I (45) octal_psram: good-die     : 0x01 (Pass)
I (50) octal_psram: Latency      : 0x01 (Fixed)
I (55) octal_psram: VCC          : 0x01 (3V)
I (60) octal_psram: SRF          : 0x01 (Fast Refresh)
I (66) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (72) octal_psram: BurstLen     : 0x01 (32 Byte)
I (77) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (83) octal_psram: DriveStrength: 0x00 (1/1)
I (89) MSPI Timing: PSRAM timing tuning index: 5
I (94) esp_psram: Found 8MB PSRAM device
I (98) esp_psram: Speed: 80MHz
I (423) esp_psram: SPI SRAM memory test OK
I (432) cpu_start: Pro cpu start user code
I (432) cpu_start: cpu freq: 240000000 Hz
I (432) cpu_start: Application information:
I (433) cpu_start: Project name:     ai-module
I (433) cpu_start: App version:      9116a8c-dirty
I (433) cpu_start: Compile time:     Dec  4 2024 14:22:21
I (434) cpu_start: ELF file SHA256:  5c4c662e9...
I (434) cpu_start: ESP-IDF:          v5.2.3-dirty
I (435) cpu_start: Min chip rev:     v0.0
I (435) cpu_start: Max chip rev:     v0.99 
I (435) cpu_start: Chip rev:         v0.1
I (436) heap_init: Initializing. RAM available for dynamic allocation:
I (437) heap_init: At 3FCB1718 len 00037FF8 (223 KiB): RAM
I (437) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (438) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (439) spi_flash: detected chip: generic
I (439) spi_flash: flash io: qio
W (441) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (448) sleep: Configure to isolate all GPIO pins in sleep state
I (449) sleep: Enable automatic switching of GPIO sleep configuration
I (450) coexist: coex firmware version: 27d8387
I (450) coexist: coexist rom version e7ae62f
I (451) main_task: Started on CPU0
I (452) esp_psram: Reserving pool of 16K of internal memory for DMA/internal allocations
I (453) main_task: Calling app_main()
W (453) S3-EYE: Warning: Long filenames on SD card are disabled in menuconfig!
I (453) MEM:    Biggest /     Free /    Total
          SRAM : [  180224 /   218111 /   268059]
         PSRAM : [ 8257536 /  8385916 /  8388608]
I (454) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (455) gpio: GPIO[38]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (456) gpio: GPIO[40]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (521) pp: pp rom version: e7ae62f
I (521) net80211: net80211 rom version: e7ae62f
I (523) wifi:wifi driver task: 3fcc2df0, prio:23, stack:6656, core=0
I (529) wifi:wifi firmware version: 82f54d2
I (530) wifi:wifi certification version: v7.0
I (530) wifi:config NVS flash: enabled
I (530) wifi:config nano formating: disabled
I (530) wifi:Init data frame dynamic rx buffer num: 128
I (531) wifi:Init static rx mgmt buffer num: 5
I (531) wifi:Init management short buffer num: 32
I (532) wifi:Init static tx buffer num: 32
[... 95 lines cut ...]
I (5256) WIFI-PROV: WAIT FOR WIFI CONNECT
I (5464) MEM:    Biggest /     Free /    Total
          SRAM : [   21504 /    56403 /   268059]
         PSRAM : [ 7864320 /  7901700 /  8388608]
I (5965) MEM:    Biggest /     Free /    Total
          SRAM : [   21504 /    56403 /   268059]
         PSRAM : [ 7864320 /  7901700 /  8388608]
I (6256) WIFI-PROV: WIFI DISCONNECTED
I (6256) WIFI-PROV: WAIT FOR WIFI CONNECT
I (6466) MEM:    Biggest /     Free /    Total
          SRAM : [   21504 /    56403 /   268059]
         PSRAM : [ 7864320 /  7901700 /  8388608]
I (6967) MEM:    Biggest /     Free /    Total
          SRAM : [   21504 /    56403 /   268059]
         PSRAM : [ 7864320 /  7901700 /  8388608]
I (7256) WIFI-PROV: WIFI DISCONNECTED
I (7256) WIFI-PROV: WAIT FOR WIFI CONNECT
I (7468) MEM:    Biggest /     Free /    Total
          SRAM : [   21504 /    56403 /   268059]
         PSRAM : [ 7864320 /  7901700 /  8388608]
I (7969) MEM:    Biggest /     Free /    Total
          SRAM : [   21504 /    56403 /   268059]
         PSRAM : [ 7864320 /  7901700 /  8388608]

```

uLipe, 2024-12-12:
@MR-VENOM could you please check on your KConfig options if the WIFI module is allowed to perform PSRAM allocations?

ReleaseBug, 2025-01-02:
@MR-VENOM  

hi，I'm having the same issue！

I tried it with ：
```
#define LV_MEM_SIZE (256 * 1024U)

lvgl\src\stdlib\builtin\lv_mem_core_builtin.c 
line : 82
#include "esp_heap_caps.h"
//static LV_ATTRIBUTE_LARGE_RAM_ARRAY MEM_UNIT work_mem_int[LV_MEM_SIZE / sizeof(MEM_UNIT)];
void* work_mem_int = heap_caps_malloc(LV_MEM_SIZE, MALLOC_CAP_SPIRAM);
```
and it supports 260*260 now.

MR-VENOM, 2025-01-03:
> @MR-VENOM
> 
> hi，I'm having the same issue！
> 
> I tried it with ：
> 
> ```
> lv_conf.h
> line: 57
> #define LV_MEM_SIZE (256 * 1024U)
> 
> lvgl\src\stdlib\builtin\lv_mem_core_builtin.c 
> line : 82
> #include "esp_heap_caps.h"
> //static LV_ATTRIBUTE_LARGE_RAM_ARRAY MEM_UNIT work_mem_int[LV_MEM_SIZE / sizeof(MEM_UNIT)];
> void* work_mem_int = heap_caps_malloc(LV_MEM_SIZE, MALLOC_CAP_SPIRAM);
> ```
> 
> and it supports 260*260 now.

Okay will try this . Thanks.

MR-VENOM, 2025-01-07:
> @MR-VENOM
>
> hi，I'm having the same issue！
>
> I tried it with ：
>
> ```
> lv_conf.h
> line: 57
> #define LV_MEM_SIZE (256 * 1024U)
>
> lvgl\src\stdlib\builtin\lv_mem_core_builtin.c
> line : 82
> #include [...]

The issue is not solved YET still facing the same problem

@ReleaseBug  Which LVGL versin do you sue and can you share me your LVGL sdkconfig parameters ?

MR-VENOM, 2025-01-07:
> @MR-VENOM could you please check on your KConfig options if the WIFI module is allowed to perform PSRAM allocations?

YES. the problem is LVGL is not allocating from PSRAM it is trying to allocate from internal RAM.

closed 2025-02-03 by lvgl-bot

MR-VENOM, 2025-03-06:
After days of debugging, I finally got it to work and display a 240x240 lottie. The main issue was that LVGL doesn’t allocate one large chunk of memory at a time — instead, it requests lots of small chunks, usually under 100 bytes. In ESP-IDF, to make SPIRAM usable with the standard C library (`malloc` and friends), you need to enable the `SPIRAM_USE_MALLOC` config. But that brings its own set of problems.

There’s another config called `SPIRAM_MALLOC_ALWAYSINTERNAL`. This setting defines a size threshold — any allocation below this size will always be placed in internal memory. In my case, this was set to 1024 bytes, and since LVGL allocates in small chunks, all of them were ending up in internal RAM instead of external RAM. The fix was to set `SPIRAM_MALLOC_ALWAYSINTERNAL` to 0, so everything, regardless of size, could go into external RAM.

On top of that, to render a 240x240 lottie, you need a minimum stack size of **24,576 bytes**.

ha5dzs, 2025-03-27:
I had a very similar issue, but on the Arduino framework and with different hardware.

I am using a 480x480 screen where I need to have the entire display as a frame buffer.
Even if I manually allocated the frame buffer to the SPI RAM and set it to direct render mode, all I got was garbage on the screen. If I intentionally reduced (halved, quartered, decimated, etc.) the size of the frame buffer, then the same garbage repeated on the screen.

Lines in question are:

```C
void *buffer = heap_caps_malloc(DISPLAY_BUFFER_SIZE, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
if(buffer == NULL)
{
   Serial.println("Couldn't allocate memory for the display buffer.");
   while(1);
}

lv_display_set_buffers(disp, buffer, NULL, DISPLAY_BUFFER_SIZE, LV_DISPLAY_RENDER_MODE_DIRECT);

Serial.println("We got here, good.");
```

For future reference, if anyone finds this, I used a:

Guition ESP32-4848S040

...with the ST7701 480x480 display.

For now, I reverted to lvgl 8.4.0, where the 450 kB frame buffer allocated to the SPI RAM works fine with my hardware.

Probably this doesn't worth adding a separate issue about this, but hopefully someone will not spend a day and a half tracing this down like I did.