lvgl/lvgl #7406: Optimization Issue in lv_memcpy Due to Macro Usage

1saeed, 2024-12-04:
### LVGL version

v9.2,2

### Platform

STM32H755 Platform
GCC (-O3)
-ffunction-sections
-fdata-sections

### What happened?

The current implementation of lv_memcpy uses macros like _COPY and _REPEAT8, which can lead to undefined or unintended behavior under high compiler optimization levels  (-O3). This results in hard faults in some cases. 

Here’s the refactored version of the function using inline functions:

```
#include <stdint.h>
#include <stddef.h>

#define _ALIGN_MASK (sizeof(uint32_t) - 1)

/* Inline function to copy a single byte */
static inline void copy_byte(volatile uint8_t **dst, const uint8_t **src) {
    **dst = **src;
    (*dst)++;
    (*src)++;
}

/* Inline function to copy a single 32-bit word */
static inline void copy_word(volatile uint32_t **dst, const uint32_t **src) {
    **dst = **src;
    (*dst)++;
    (*src)++;
}

/* Inline function for repeated byte copying */
static inline void repeat_byte_copy(volatile uint8_t **dst, const uint8_t **src, size_t times) {
    for (size_t i = 0; i < times; i++) {
        copy_byte(dst, src);
    }
}

/* Inline function for repeated word copying */
static inline void repeat_word_copy(volatile uint32_t **dst, const uint32_t **src, size_t times) {
    for (size_t i = 0; i < times; i++) {
        copy_word(dst, src);
    }
}

void lv_memcpy(void *dst, const void *src, size_t len) {
    volatile uint8_t *d8 = (volatile uint8_t *)dst;
    const uint8_t *s8 = (const uint8_t *)src;

    /* Simplify for small memories */
    if (len < 16) {
        while (len) {
            copy_byte(&d8, &s8);
            len--;
        }
        return;
    }

    uintptr_t d_align = (uintptr_t)d8 & _ALIGN_MASK;
    uintptr_t s_align = (uintptr_t)s8 & _ALIGN_MASK;

    /* Byte copy for unaligned memories */
    if (s_align != d_align) {
        while (len > 32) {
            repeat_byte_copy(&d8, &s8, 32);
            len -= 32;
        }
        while (len) {
            copy_byte(&d8, &s8);
            len--;
        }
        return;
    }

    /* Align the memories */
    if (d_align) {
        d_align = _ALIGN_MASK + 1 - d_align;
        while (d_align && len) {
            copy_byte(&d8, &s8);
            d_align--;
            len--;
        }
    }

    volatile uint32_t *d32 = (volatile uint32_t *)d8;
    const uint32_t *s32 = (const uint32_t *)s8;

    while (len > 32) {
        repeat_word_copy(&d32, &s32, 8);
        len -= 32;
    }

    d8 = (volatile uint8_t *)d32;
    s8 = (const uint8_t *)s32;
    while (len) {
        copy_byte(&d8, &s8);
        len--;
    }
}
```

1saeed, 2024-12-05:
@liamHowatt @uLipe . Thanks a lot for your replies. So, the code switches between uint8_t* and uint32_t* pointers for optimization, but O3 might make aggressive assumptions about pointer aliasing that could break the alignment handling logic. The compiler might incorrectly assume these pointers can't alias to the same memory. Also, The manual loop unrolling (using that _REPEAT8 macro) could conflict with O3's auto-vectorization. The compiler might try to further optimize these already-unrolled loops, potentially causing cache-unfriendly access patterns or excessive register pressure. And last one is that O3 might reorder memory operations since there aren't explicit barriers. The _COPY macro (which does *d = *s; d++; s++;) could have its loads/stores reordered in ways we don't want. 

So, I'm guessing that what I'm having is that O3's aggressive optimizations might not properly respect my platform's alignment requirements. 

About the solution, that's a fair assumption you mentioned. One thing we can do is adding volatile qualifiers or memory barriers at critical points. It also fixed the issue on my end. 

```c
volatile uint8_t* restrict d8 = dst;
volatile const uint8_t* restrict s8 = src;
```

onecoolx, 2024-12-05:
> @liamHowatt @uLipe非常感谢您的回复。因此，代码在 uint8_t* 和 uint32_t* 指针之间切换以进行优化，但 O3 可能会对指针别名做出激进的假设，这可能会破坏对齐处理逻辑。编译器可能会错误地假设这些指针不能别名到相同的内存。此外，手动循环展开（使用 _REPEAT8 宏）可能会与 O3 的自动矢量化相冲突。编译器可能会尝试进一步优化这些已经展开的循环，这可能会导致缓存不友好的访问模式或过度的寄存器压力。最后一个是 O3 可能会重新排序内存操作，因为没有明确的障碍。_COPY 宏（执行 *d = *s; d++; s++;）的加载/存储可能会以我们不想要的方式重新排序。
> 
> 因此，我猜测我遇到的问题是 O3 的积极优化可能没有正确尊重我的平台的对齐要求。
> 
> 关于解决方案，您提到的假设很合理。我们可以做的一件事是在关键点添加 volatile 限定符或内存屏障。这也解决了我的问题。
> 
> ```c
> volatile uint8_t* restrict d8 = dst;
> volatile const uint8_t* restrict s8 = src;
> ```

Can your problem be solved by adding the GCC compiler option `-fno-strict-aliasing`?

onecoolx, 2024-12-05:
> @liamHowatt @uLipe . Thanks a lot for your replies. So, the code switches between uint8_t* and uint32_t* pointers for optimization, but O3 might make aggressive assumptions about pointer aliasing that could break the alignment handling logic. The compiler might incorrectly assume these pointers can't alias to the same memory. Also, The manual loop unrolling (using that _REPEAT8 macro) could conflict with O3's auto-vectorization. The compiler might try to further optimize these already-unrolled loops, potentially causing cache-unfriendly access patterns or excessive register pressure. And last one is that O3 might reorder memory operations since there aren't explicit barriers. The _COPY macro (which does *d = *s; d++; s++;) could have its loads/stores reordered in ways we don't want.
> 
> So, I'm guessing that what I'm having is that O3's aggressive optimizations might not properly respect my platform's alignment requirements.
> 
> About the solution, that's a fair assumption you mentioned. One thing we can do is adding volatile qualifiers or memory barriers at critical points. It also fixed the issue on my end.
> 
> ```c
> volatile uint8_t* d8 = dst;
> volatile const uint8_t* s8 = src;
> ```

Can your problem be solved by adding the GCC compiler option -fno-strict-aliasing?

1saeed, 2024-12-05:
> > @liamHowatt @uLipe . Thanks a lot for your replies. So, the code switches between uint8_t* and uint32_t* pointers for optimization, but O3 might make aggressive assumptions about pointer aliasing that could break the alignment handling logic. The compiler might incorrectly assume these pointers can't alias to the same memory. Also, The manual loop unrolling (using that _REPEAT8 macro) could conflict with O3's auto-vectorization. The compiler might try to further optimize these already-unrolled loops, potentially causing cache-unfriendly access patterns or excessive register pressure. And last one is that O3 might reorder memory operations since there aren't explicit barriers. The _COPY macro (which does *d = *s; d++; s++;) could have its loads/stores reordered in ways we don't want.
> > So, I'm guessing that what I'm having is that O3's aggressive optimizations might not properly respect my platform's alignment requirements.
> > About the solution, that's a fair assumption you mentioned. One thing we can do is adding volatile qualifiers or memory barriers at critical points. It also fixed the issue on my end.
> > ```c
> > volatile uint8_t* d8 = dst;
> > volatile const uint8_t* s8 = src;
> > ```
> 
> Your problem can be solved by adding the GCC compiler option "-fno-strict-aliasing"?

I could, but I would lose optimization opportunities that strict aliasing would enable in other parts of the code. It's a global flag that effects the entire codebase.

1saeed, 2024-12-05:
> Your problem can be solved by adding the GCC compiler option "-fno-strict-aliasing"?

The -fstrict-aliasing option is enabled at levels -O2, -O3, -Os.

However, code that only works with GCC with -fno-strict-aliasing is not correct C.

liamHowatt, 2024-12-10:
Regarding the pointer aliasing assumption issue, it only dereferences a `uint32_t *` if it's aligned. Are you saying the aggressive optimization will generate machine code that eagerly accesses unaligned memory as though it's aligned, before it confirms it's aligned?

kisvegabor, 2024-12-18:
@1saeed Thank you for the detailed reply. 

Have you really experienced a crash because of it or is it more like a theoretical issue? 

These functions might heavily affect the performance so I wonder if you have checked if there is no regression.

1saeed, 2024-12-20:
Hello @kisvegabor. Sorry for the late reply! Had a super busy week.

I can confirm that I’ve replicated the issue with unaligned pointers and O3 optimization using the following setup:

c
Copy code
uint8_t* src = (uint8_t*)0xD0000001;
uint8_t* dst = (uint8_t*)0xD0000063;
uint32_t N = 99;

lv_memcpy(dst, src, N);

kisvegabor, 2024-12-22:
> Sorry for the late reply! Had a super busy week.

No worries, I know how it goes :slightly_smiling_face: 

> Using volatile here would save it:

I suggest going with this less invasive fix for now. Could you send  a PR with some comments to explain why `volatile` is required?

liamHowatt, 2025-01-08:
Please open the PR in this repo.

I tested the benchmark on STM32U5 with SW rendering, `-O3` with and without `volatile` and there's no performance regression. It's too bad there isn't a better way to fix it, though. Using `volatile` feels like the wrong approach but I don't have a better idea.

uLipe, 2025-01-08:
volatile here would prevent the variable to be removed during loop optimisation step, which would enforce the usage of 16/32bit move opérations on 32bit archs, which might be a problem to deal with unaligned memory.

So I suppose the volatile here is a way to tell the compiler to not try to optimize byte copy opération by keeping byte type placeholders.

It just a guess but I suppose volatile here would not impact on perfomance here.

It would be a nice case to put a test dealing with data moving with unaligned addresses to check the behavior.

closed 2025-02-03 by uLipe