lvgl/lvgl #7408: Exception in lv_draw_label_iterate_characters

FrankBJensen, 2024-12-04:
### LVGL version

LVGL 9.2.2

### Platform

Visual studio code, where I use LVGL as a component. I use CST816 and ST7789 libraries. I keep all LVGL operations in the main LVGL task, to keep it simple and thread safe. Only main loop (that contains nearly nothing) and LVGL is running on core 0, main application runs on core 1, and it posts data to LVGL main task via xQueueReceive.

### What happened?

I have en exception. It happens rarely, so I dont really know where to look. When it happens, the ESP32-S3 do not reboot, it just gets into a loop, where this exception occurs every few seconds, and the display in inresponsive. Rest of the application keeps running. It can happen quickly after reboot, it can also happen after several days of perfect operation. I monitor the heap and memory, there do not seem to be anything leaking away.

```
E (1242187) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (1242187) task_wdt:  - IDLE1 (CPU 1)
E (1242187) task_wdt: Tasks currently running:
E (1242187) task_wdt: CPU 0: IDLE0
E (1242187) task_wdt: CPU 1: lvglDraw
E (1242187) task_wdt: Print CPU 1 backtrace

Backtrace: 0x403776FE:0x3FC9A260 0x4037A581:0x3FC9A280 0x42011F81:0x3FCBFD70 0x4201262B:0x3FCBFDD0 0x42031131:0x3FCBFEF0 0x4202E5D6:0x3FCBFF10 0x4202E62A:0x3FCBFF30 0x4202E6B4:0x3FCBFF50 0x4201AAE3:0x3FCBFF70 0x4037EE8D:0x3FCBFF90
  #0  0x403776FE in esp_crosscore_isr at C:\Users\Frank\.platformio\packages\framework-espidf\components\esp_system/crosscore_int.c:73
  #1  0x3FC9A260 in _xt_exception_table at ??:?
  #2  0x4037A581 in _xt_lowint1 at C:\Users\Frank\.platformio\packages\framework-espidf\components\xtensa/xtensa_vectors.S:1240
  #3  0x3FC9A280 in _xt_exception_table at C:\Users\Frank\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa/port.c:67
  #4  0x42011F81 in draw_letter at components\lvgl\src\draw/lv_draw_label.c:538 (discriminator 1)
  #5  0x4201262B in lv_draw_label_iterate_characters at components\lvgl\src\draw/lv_draw_label.c:434
  #6  0x42031131 in lv_draw_sw_label at components\lvgl\src\draw\sw/lv_draw_sw_letter.c:59
  #7  0x4202E5D6 in execute_drawing at components\lvgl\src\draw\sw/lv_draw_sw.c:268
  #8  0x4202E62A in execute_drawing_unit at components\lvgl\src\draw\sw/lv_draw_sw.c:132
  #9  0x4202E6B4 in render_thread_cb at components\lvgl\src\draw\sw/lv_draw_sw.c:243
  #10 0x4201AAE3 in prvRunThread at components\lvgl\src\osal/lv_freertos.c:433
```

### How to reproduce?

```
bool MyDisplay::notify_flush_ready(esp_lcd_panel_io_handle_t panel_io, esp_lcd_panel_io_event_data_t *edata, void *user_ctx)
{
    lv_display_t *disp = (lv_display_t *)user_ctx;
    lv_display_flush_ready(disp);  // I have tried to change this to the global display variable, no change
    return false;
}

void MyDisplay::flush_cb(lv_display_t* disp, const lv_area_t* area, uint8_t* px_map)
{
    int x1 = area->x1;
    int x2 = area->x2;
    int y1 = area->y1;
    int y2 = area->y2;
    
    lv_draw_sw_rgb565_swap(px_map, (x2 + 1 - x1) * (y2 + 1 - y1));

    esp_lcd_panel_draw_bitmap((esp_lcd_panel_handle_t)lv_display_get_user_data(disp), x1, y1, x2 + 1, y2 + 1, px_map);
}

esp_err_t MyDisplay::lvgl_init(void)
{
    lv_init();

    display = lv_display_create(DISP_HOR_RES, DISP_VER_RES);

    buf1 = heap_caps_calloc(1, BUFFER_SIZE, MALLOC_CAP_INTERNAL |  MALLOC_CAP_DMA);
    buf2 = heap_caps_calloc(1, BUFFER_SIZE, MALLOC_CAP_INTERNAL |  MALLOC_CAP_DMA);

    lv_display_set_buffers(display, buf1, buf2, BUFFER_SIZE, LV_DISPLAY_RENDER_MODE_PARTIAL);

    lv_display_set_user_data(display, panel_handle);
    lv_display_set_color_format(display, LV_COLOR_FORMAT_RGB565);

[... 181 lines cut ...]
                    break;
                case SCR_PAUSE:
                    _ui_screen_change(&ui_screenPause, LV_SCR_LOAD_ANIM_FADE_ON, 500, 0, &ui_screenPause_screen_init);
                    break;
                case SCR_STAT:
                    _ui_screen_change(&ui_screenSTAT, LV_SCR_LOAD_ANIM_FADE_ON, 500, 0, &ui_screenSTAT_screen_init);
                    break;
                case SCR_OTA:
                    _ui_screen_change(&ui_screenOTA, LV_SCR_LOAD_ANIM_FADE_ON, 500, 0, &ui_screenOTA_screen_init);
                    break;
                case SCR_SCREEN2:
                    _ui_screen_change(&ui_Screen2, LV_SCR_LOAD_ANIM_FADE_ON, 500, 0, &ui_Screen2_screen_init);
                    break;
                case SCR_SCREEN3:
                    _ui_screen_change(&ui_Screen3, LV_SCR_LOAD_ANIM_FADE_ON, 500, 0, &ui_Screen3_screen_init);
                    break;
                case SCR_SCREEN4:
                    _ui_screen_change(&ui_Screen4, LV_SCR_LOAD_ANIM_FADE_ON, 500, 0, &ui_Screen4_screen_init);
                    break;
                }
            }
        }
    }
}
```

FrankBJensen, 2024-12-04:
Posting this code above, I just noticed, that lvglDraw is running on core 1. LVGL main task is running on core 0. Is that an issue? Can/should I pin lvglDraw to core 0?

FrankBJensen, 2024-12-04:
USE_OS was set to FreeRTOS. I have changed this to none, and running test with that. The processor used is ESP32-S3 n16r8v. I dont know, if this could be the cause.

kisvegabor, 2024-12-22:
I'm sorry that I have missed this comment so long. Unfortunately even after this long delay I can't say any useful. From LVGL's point of view all your code looks good.  It seems more like and ESP specific issue. Maybe the ESP guys can give more hints. 🙁

Just a few ideas:
- Enable [LVGL's logging](https://docs.lvgl.io/master/details/debugging/log.html)
- Enable `[LV_USE_ASSERT_...](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L378-L383)`s
- Is there a default task stack size somewhere? If so, it'd be interesting to see if changing it  would make any difference.

FrankBJensen, 2024-12-29:
Hi Gabor.

My solution was to select no operation system, thus only using one core. No problems since I did that.

Venligst / Best regards
Frank Jensen

Futuresales
Salgsafdelingen / Sales
+45 63 40 18 00   telephone
+45 63 40 18 01   direct/cell

Email:      ***@***.******@***.***>
Web:       http://www.futuresales.eu<http://www.futuresales.eu/>
CVR/VAT: 26990408
Damsbovej 7, 5492 Vissenbjerg, Denmark

closed 2025-01-21 by lvgl-bot