lvgl/lvgl #7410: roller widget display not correct when set a large option list.

xuanmumy, 2024-12-04:
### LVGL version

8.3.10

### Platform

arm cortex m33 with rt-thread os.

### What happened?

when I set the option with 60 items/montserrat 48/LV_ROLLER_MODE_INFINITE, the roller display wrong selected item.
the bug seems caused by the content hight larger than max obj hight. 
the max label hight by default is 8192, with LV_ROLLER_INF_PAGES set to 3, 60 items content hight is (48+16)*3*60 = 11520.
label_sel_y caculated in draw_main is wrong.  this also cause bg part display problem.  if use default LV_ROLLER_INF_PAGES 7,  the situation is more serious.
I see 
#define LV_COORD_MAX            ((1 << _LV_COORD_TYPE_SHIFT) - 1)
and 
#define _LV_COORD_TYPE_SHIFT    (13U)
modify like bellow can fix if inf pages is set to 3. but it is not a good solution. and I can not estimate the side effect.
#define LV_COORD_MAX            ((1 << (_LV_COORD_TYPE_SHIFT+1)) - 1)

### How to reproduce?

const char min_sec_num_table[] = {"0\n1\n2\n3\n4\n5\n6\n7\n8\n9\n10\n11\n12\n13\n14\n15\n16\n17\n18\n19\n20\n21\n22\n23\n24\n25\n26\n27\n28\n29\n30\n31\n32\n33\n3"
                                  "4\n35\n36\n37\n38\n39\n40\n41\n42\n43\n44\n45\n46\n47\n48\n49\n50\n51\n52\n53\n54\n55\n56\n57\n58\n59"};

void roller_1(void)
{
    lv_obj_t *roller1 = lv_roller_create(lv_scr_act());
    lv_obj_set_size(roller1, 250, 100);
  //  lv_obj_set_style_text_font(roller1, font_chn_large.font, LV_PART_MAIN);
    lv_obj_set_style_text_font(roller1, &lv_font_montserrat_48, LV_PART_MAIN);
    lv_obj_set_style_text_color(roller1, lv_color_hex(0xffffff), LV_PART_MAIN);
    lv_obj_set_style_bg_color(roller1, lv_color_hex(0x000000), LV_PART_MAIN);
    lv_obj_set_style_bg_color(roller1, lv_color_hex(0xff0000), LV_PART_SELECTED);
    lv_roller_set_options(roller1, min_sec_num_table, LV_ROLLER_MODE_INFINITE);
    lv_roller_set_visible_row_count(roller1, 4);
    lv_obj_center(roller1);
}

xuanmumy, 2024-12-04:
I see there is a issue record https://github.com/lvgl/lvgl/issues/5006. but it is not a solution, it just hide this problem with a larger obj max hight setting.

kisvegabor, 2024-12-22:
Hi,

In v8.3 you should enable [LV_USE_LARGE_COORD](https://github.com/lvgl/lvgl/blob/release/v8.3/lv_conf_template.h#L356) in this case.

closed 2025-01-16 by lvgl-bot