lvgl/lvgl #7411: Clickable Flow-of-Control / Flow-of-Data Diagram

vwheeler63, 2024-12-04:
### Introduce the problem

In reviewing the diagram shown near the top of the [LVGL Basics](https://docs.lvgl.io/master/intro/basics.html) page, I found myself wanting to click parts of the diagram to lead to documentation about it.  But there is no image map currently to make this happen.

### Proposal

- lv_timer_handler() click could lead to the [Timer Handler](https://docs.lvgl.io/master/intro/add-lvgl-to-your-project/timer_handler.html#timer-handler) documentation.
- lv_tick_inc() click could lead to the [Tick Interface](https://docs.lvgl.io/master/intro/add-lvgl-to-your-project/connecting_lvgl.html#tick-interface) documentation.
- Widget Tree click could lead to [Widget Basics](https://docs.lvgl.io/master/details/base-widget/obj.html#widget-basics) documentation since that documentation spells out how Widgets form a tree structure.
- Rendering Logic click could lead to Drawing API documentation once it exists.
- etc. as more documentation gets added.

Having found myself intuitively wanting to click parts of the diagram, I thought I would share the idea since I have no doubt it is do-able in the documentation, and would make that diagram that much more useful.

kisvegabor, 2024-12-05:
I'm not sure that it's a high priority thing, but I wonder if you have any idea how to to do it.

vwheeler63, 2024-12-05:
> I'm not sure that it's a high priority thing, but I wonder if you have any idea how to to do it.

I know how HTML image maps work and can work it out if we want to do it.

liamHowatt, 2024-12-06:
@vwheeler63 take a look at the Mermaid diagram I embedded in the docs some time ago. It's interactive. You can zoom and scroll it. Maybe Mermaid diagrams support hyperlinks and one could be used in LVGL Basics?

https://docs.lvgl.io/master/details/main-components/fs.html#lv-fs-read-behavior-when-cache-is-enabled

vwheeler63, 2024-12-06:
> take a look at the Mermaid diagram

Those are very cool!  I will look into it!  (And thank you for turning me onto Mermaid diagrams.  Not sure what their capabilities are yet.)  Though I am unclear on how PR #7414 is going to help....

vwheeler63, 2024-12-06:
> > Though I am unclear on how PR #7414 is going to help....
> 
> Instant feedback loop! I can make a change, rebuild the docs, repeat, with a shorter wait time. The Mermaid diagram is just a Sphinx extension.

Ah, yes!  For sure!

vwheeler63, 2024-12-23:
not stale

liamHowatt, 2025-01-08:
Victor, fyi, you can remove `:zoom:` from a Mermaid diagram source to make it not zoom and scroll, I believe. All you're looking for is embedded hyperlinks in a diagram, after all.

vwheeler63, 2025-01-08:
> Victor ...

Since it is just a bitmap at the moment, all I was thinking was turning a reST `.. image::` directive into raw HTML and using a hand-typed image map.  Really simple.  I cannot (yet) envision a Mermaid diagram showing all the details that the current diagram does (though I still know little about them).  Can you?

liamHowatt, 2025-01-08:
I think you should go with what you know! Even if it is possible with Mermaid, your idea sounds more flexible.

vwheeler63, 2025-01-08:
> I think you should go with what you know!

And I openly admit, it's the shortest path to the result with my current knowledge and tool set.  :-)

kisvegabor, 2025-01-09:
@vwheeler63  Unless you are very passionate about the clickable diagrams, please prioritize the proof reading instead. 🙂

vwheeler63, 2025-01-09:
> please prioritize the proof reading instead.

Agreed.  I always held this at low priority.

vwheeler63, 2025-01-26:
not stale

vwheeler63, 2025-02-12:
still on my radar

vwheeler63, 2025-03-24:
Not stale.

vwheeler63, 2025-04-19:
Not stale.

vwheeler63, 2025-05-12:
not stale

vwheeler63, 2025-05-31:
not stale

vwheeler63, 2025-07-07:
not stale

closed 2025-07-28 by kisvegabor

vwheeler63, 2025-07-28:
> I close it now. Let's reopen when there is time to work on it.

Fair enough.