lvgl/lvgl #7412: LVGL Directory Structure Change Could Trigger Dependant Projects to Update

vwheeler63, 2024-12-04:
### Introduce the problem

I have noticed that when files are added and/or removed from the LVGL repository that the `lv_port_pc_visual_studio` repository lags behind, sometimes for quite a while.  And until it gets updated, the project will not compile without modification.

One of the projects contained in that repository is `LVGL.MaintainerTools.sln` (solution file) => `LvglProjectFileUpdater.vcxproj` (project file), which, when run, uses the updated LVGL sub-module's working directory tree to update the list of files in the project (unfortunately, Visual Studio doesn't do that automatically for C/C++ projects).  This then makes the main projects (such as `LvglWindowsSimulator.vcxproj`) compile again.

I am doing that manually now when the directory structure gets updated.

The trouble is:  currently, due to this delay, new LVGL users have an (unwanted) high probability of running into this repository and trying it out during a time when it is not compiling, and this can unnecessarily damage LVGL public relations and goodwill among potential new users.

### Proposal

So my thoughts were some kind of trigger to get the `lv_port_pc_visual_studio` repository updated to match is needed in order to reduce the delay between the directory-structure update and the dependent repository(ies) being updated.  It can be anything that works:

- email to maintainer
- perhaps a GitHub notification of some sort to maintainer, e.g. with a new issue for that repository, so it gets timely attention(?)
- something automated (preferably not after every commit, but only when there is a change in the working directory tree)

If there are other similarly-dependent projects out there (I know Eclipse updates its projects automatically), then they might need such a trigger as well.

I'm guessing the maintainer of that repository would like to say what his needs and wants are for such a notification:  what would work well for him?

## Thoughts on Design

Something that is able to build a string representation of the directory structure (easy with Python `os.walk()`) could generate a hash (or "digest") of that string which would remain the same from commit to commit in the `lvgl/lvgl` repository, and when a file is added or removed, the hash would change and could thus provide a decision-point to trigger whatever kind of notification is most helpful.  Other than the script, the only thing required to store would be a file to store of the generated hash when it changes.

cc:  @kisvegabor @MouriNaruto

vwheeler63, 2024-12-05:
> Can't we make Visual Studio to glob all the C and cpp files from the `lvgl/src` `lvgl/demos` `lvgl/examples` folder?

Unfortunately, Visual Stud8io is kind of stupid in that respect.  I think it is from a mis-guided effort on Microsoft's part to make the "tree structure" of files in the project to be INDEPENDENT of the actual directory structure.  Sadly, it is listed/discussed as a "feature" in forums about this issue.  That's why MouriNaruto had to write that "LvglProjectFileUpdater" program -- to do the job that Visual Studio isn't doing -- when it is dealing with a project whose source file structure can change frequently, and the changes are not done from within the VS project.

vwheeler63, 2024-12-05:
> Unfortunately, Visual Stud8io is kind of stupid in that respect. I think it is from a mis-guided effort on Microsoft's part to make the "tree structure" of files in the project to be INDEPENDENT of [...]

The result is that if, for example, a file `./lvgl/src/libs/path/to/lib/test_file.c` content is moved into another `.c` file and the file is deleted, suddenly the `lvgl/lv_port_pc_visual_studio` won't compile and requires its project being "fixed" by:

- manually by the user if he/she knows enough to do it [not intended], or
- by running the `LvglProjectFileUpdater` program (which updates 3 VS projects in the `lv_port_pc_visual_studio` repository.

MouriNaruto, 2024-12-05:
I think I will try to write an MSBuild task to automate these tasks when people open the project.

Kenji Mouri

vwheeler63, 2024-12-05:
> I think I will try to write an MSBuild task to automate these tasks when people open the project.
> 
> Kenji Mouri

Excellent!  Keep in mind that when updating the `lv_port_pc_visual_studio` repository locally, the update of the `lvgl` sub-module is sometimes automated and when the directory structure has changed, of course, it causes the 3 projects that rely on it to fail to compile with either "can't find file ..." or "symbol not found" errors depending on whether files went away, were added, or just moved or renamed.  So of course I am wondering if your idea keeps the projects in the repository updated, or whether it only updates the projects on the local workstations.  If the latter, I would prefer not having the delay and file re-writes every time I open it, since I open it frequently when no changes would apply (so it would be re-generating identical files over and over).

MouriNaruto, 2024-12-23:
Not stale

vwheeler63, 2025-01-03:
still not stale

MouriNaruto, 2025-01-19:
not stale

vwheeler63, 2025-01-19:
still not stale

vwheeler63, 2025-02-04:
Not stale

MouriNaruto, 2025-02-23:
Not stale

vwheeler63, 2025-02-23:
Hi, @MouriNaruto !  FYI, a doc-structure change is about to be submitted in the next 24-48 hours or so (might be delayed by waiting on pending PRs to be merged).  I will give you a heads-up here so you have good prediction as soon as I do.

I'm kind of thinking we should start by simply sending you an email whenever one of us sees or knows about:

- new files
- changed names
- changed paths
- old files going away.

But thinking about that also tells me that somewhere we could keep a record of file structure on every commit to `master` have a Python script check it for structure change and use THAT to send an email.  What do you think?

MouriNaruto, 2025-02-25:
@vwheeler63 

I think maybe the more better way in the current stage is providing batch scripts for users to update submodules.

I don't think it's a good idea for synchronizing from main repository for every changed commit because I think users may need the some stability.

Also, I think I need to write some tools help people migrate the lv_conf.h to new version. (I have found LVGL loves change this file a lot. I think if we need to provide users a up-to-date implementation, we should solve that issue.) If LVGL already has a prototype implementation for that will be much better.

Kenji Mouri

vwheeler63, 2025-02-25:
> I think maybe the more better way in the current stage is providing batch scripts for users to update submodules.
> 
> I don't think it's a good idea for synchronizing from main repository for every changed commit because I think users may need the some stability.

If I think of myself doing UI development in this project. I would for sure create my own branch and that by itself would give me stability.  Then when I was ready to get new content, I would re-base on top of _________ (latest, preferred version, etc.).

> Also, I think I need to write some tools help people migrate the lv_conf.h to new version.

Good point!  I can imagine a Python script that starts with an array of everything the project should turn on and it could parse the new versions of `lv_conf_template.h` and generate new versions of `lv_conf.h` therewith....  If I were doing it, I would model it from `lvgl/docs/config_builder.py` but enhance its logic to read what needs to get "turned on" (0 => 1) driven by an external config file, perhaps governed by what the LVGL examples need.  Just my thought based on what I have seen/learned thus far.

vwheeler63, 2025-03-18:
Not stale.

MouriNaruto, 2025-04-10:
I think it's not stale.

But also we can talk this issue in https://github.com/lvgl/lv_port_pc_visual_studio/issues/101.

Kenji Mouri

vwheeler63, 2025-05-04:
Not stale.

MouriNaruto, 2025-05-23:
Not stale.

vwheeler63, 2025-06-10:
Not stale.

MouriNaruto, 2025-07-22:
Not stale, I'm working on that.

Kenji Mouri

AndreCostaaa, 2025-07-22:
Hi @MouriNaruto 

The other week I discussed with @vwheeler63 about this. Would you consider using `CMake` to generate the Visual Studio solution? Seems like a very straight forward and easy solution for the problem you're trying to solve here

MouriNaruto, 2025-07-22:
@AndreCostaaa 

I don't consider CMake because we need proper out-of-box Visual Studio build experience.

Kenji Mouri

vwheeler63, 2025-07-22:
> I don't consider CMake because we need proper out-of-box Visual Studio build experience.

@AndreCostaaa , I concur with this, because so many end users will be very comfortable with Visual Studio, but will have no knowledge of CMAKE, and so be placed in a very bad situation if anything goes wrong that breaks the compile, whereas in the normal VS environment, they will be armed to fix it.

vwheeler63, 2025-08-12:
Not stale

AndreCostaaa, 2025-10-06:
I believe this issue belongs to https://github.com/lvgl/lv_port_pc_visual_studio as other port repositories don't suffer from the same problem

I still suggest using CMake to generate a visual studio solution, maybe based on a CI workflow

Let's reopen this if something comes up that should be done by the main repository

closed 2025-10-06 by AndreCostaaa