lvgl/lvgl #7416: The color with transparency drawn by lv-vector_csc_det_fill_inear_gradient is not displayed correctly

wu781521512, 2024-12-05:
### LVGL version

9.1

### Platform

SDL and DRM

### What happened?

When I use this function to draw polygons filled with gradient colors, setting white with transparency will display a bias towards black. I changed white to black and found no difference. White only works when I set it to be completely opaque. I see that my LVGL color is 32-bit, XRGB。
[image]

I would like to know how I can solve such a problem. Thank you very much
I have also tried transitioning from completely opaque white to transparent white, but the transition was also very unnatural

### How to reproduce?

```C++ 
   lv_vector_path_clear(path);
    lv_vector_dsc_identity(dsc);

    lv_grad_dsc_t grad;
    grad.dir = LV_GRAD_DIR_VER;
    grad.stops_count = 2;
    grad.stops[0].color = lv_color_hex(0xffffff);
    grad.stops[0].opa = LV_OPA_10;
    grad.stops[0].frac = 0;
    grad.stops[1].color = lv_color_hex(0xffffff);
    grad.stops[1].opa = LV_OPA_0;
    grad.stops[1].frac = 255;
    lv_vector_dsc_set_stroke_width(dsc, 0);
    lv_vector_dsc_set_fill_linear_gradient(dsc, &grad, LV_VECTOR_GRADIENT_SPREAD_PAD);
```

kisvegabor, 2024-12-22:
cc @FASTSHIFT

FASTSHIFT, 2024-12-27:
This is normal, because the vector renderer multiplies the RGB color with the Alpha when outputting, so that the RGB color can be displayed directly on the LCD (because the LCD does not process multiplication with Alpha). So when using white + 50% opa fill, the output color is 0x7E7E7E7E, which is displayed as gray, but you expect 0x7EFFFFFF.

FASTSHIFT, 2024-12-27:
You can use this code to achieve transparency gradient effect:

```c
static void alpha_blend(void)
{
    lv_draw_buf_t* draw_buf = lv_draw_buf_create(64, 64, LV_COLOR_FORMAT_A8, LV_STRIDE_AUTO);
    lv_draw_buf_clear(draw_buf, NULL);
    uint8_t* data = (uint8_t*)lv_draw_buf_goto_xy(draw_buf, 0, 0);

    for (int y = 0; y < 64; y++) {
        lv_memset(data, y, draw_buf->header.stride);
        data += draw_buf->header.stride;
    }

    lv_obj_t* obj = lv_obj_create(lv_scr_act());
    lv_obj_set_style_bg_color(obj, lv_palette_main(LV_PALETTE_BLUE), 0);

    lv_obj_t* img = lv_image_create(lv_scr_act());
    lv_obj_set_style_image_recolor(img, lv_color_white(), 0);
    lv_obj_set_style_image_opa(img, LV_OPA_COVER, 0);
    lv_image_set_src(img, draw_buf);
}
```

FASTSHIFT, 2024-12-27:
@kisvegabor 
I also noticed a problem. Since the image output by the LVGL/thorvg renderer has been multiplied by alpha, when the output image is used as the input image for secondary blending (**S * Sa + (1 - Sa) * D**), it may cause the alpha to be multiplied again, resulting in a darker color. 

It is necessary to consider introducing a SRC_OVER (**S + (1 - Sa) * D**) blend mode for the software renderer and use `LV_IMAGE_FLAGS_PREMULTIPLIED` to determine whether the original image has been alpha multiplied.

kisvegabor, 2025-01-09:
> the image output by the LVGL/thorvg renderer has been multiplied by alpha

Can ThorVG render in normal (not multiplied format)?

kisvegabor, 2025-01-30:
It is related to my suggestion [here](https://github.com/lvgl/lvgl/pull/7141#pullrequestreview-2581673010). In short if ThorVG finds that it cannot render directly to a layer, it should create a temporary buffer (layer) and draw there first and blend that buffer.

closed 2025-03-01 by lvgl-bot