lvgl/lvgl #7419: dropdown widget control has a bug！

xinsuinizhuan, 2024-12-06:
### LVGL version

v8.10 and v9.1

### Platform

I set the chinese dropdown list:
lv_dropdown_set_options(cur_dropdown_obj, "测试\n测试1\n测试2\n");

the select show is right,but the droplist is error,I have changed the chinese to utf8, but error show as:
[image]

### What happened?

the droplist also need to show right

M1NGS, 2024-12-06:
贴一下你的最小重现代码  Attach your code please.

xinsuinizhuan, 2024-12-06:
> 贴一下你的最小重现代码 Attach your code please.

    lv_obj_t* dd = lv_dropdown_create(lv_screen_active());
    lv_dropdown_set_options(dd, "test\n测试1\n测试2\n");

cheungxiongwei, 2024-12-06:
> > 贴一下你的最小重现代码 Attach your code please.
> 
> ```
> lv_obj_t* dd = lv_dropdown_create(lv_screen_active());
> lv_dropdown_set_options(dd, "test\n测试1\n测试2\n");
> ```

You also need to set the `lv_dropdown_get_list(dd)` object font.

`lv_obj_set_style_text_font(lv_dropdown_get_list(dd),font,LV_PART_MAIN);`

uLipe, 2024-12-06:
Please add the steps to reproduce the issue or please mention if the problem is not something that is not happen consistently.

xinsuinizhuan, 2024-12-09:
> lv_obj_set_style_text_font(lv_dropdown_get_list(dd),font,LV_PART_MAIN);

thank you, it is ok:

M1NGS, 2024-12-09:
> > lv_obj_set_style_text_font(lv_dropdown_get_list(dd),font,LV_PART_MAIN);
> 
> thank you, it is ok:

关闭issue啊，哥们儿， close issue please

uLipe, 2024-12-10:
Closing this one initially but if needed please feel free to reopen it.

closed 2024-12-10 by uLipe