lvgl/lvgl #7421: Port freetype2.13.3

jor1178, 2024-12-06:
### LVGL version

v9.2.2

### Platform

[image]

### What happened?

The emulator built with vscode ADAPTS to freetype, compiling times error, can not find the definition of these function interfaces。

cheungxiongwei, 2024-12-06:
It seems that your target_link_libraries target is wrong.
`lvgl` needs to link `freetype`, not the `main` target.
`target_link_libraries(lvgl PUBLIC freetype)`

If you include it as source code, you can use it like this, assuming the freetype source code is in the root directory.
```cmake
add_subdirectory(freetype)
target_link_libraries(lvgl PUBLIC freetype)
```

jor1178, 2024-12-06:
How to compile it into a dynamic library into the project, what should I do, do not want to use the source code

jor1178, 2024-12-06:
At present, the compilation has passed, but an error will be reported when running
[image]

jor1178, 2024-12-06:
At present, the compilation has passed, but an error will be reported when runni
[image]
ng

kisvegabor, 2024-12-22:
@jor1178 If the issue is resolved, please close it.

jor1178, 2024-12-23:
Thanks for answering. Closed

closed 2024-12-23 by jor1178