lvgl/lvgl #7422: glfw folder refactor

jwinarske, 2024-12-06:
### Introduce the problem

I'm looking to refactor the GLFW code.  The goal is to factor out the egl/GLES code into it's own folder, so the egl/GLES code can be reused for generic EGL/GLES use cases.

Does this sound reasonable?

jwinarske, 2024-12-07:
drmpp `drm-gbm-lvgl` demo running on Ubuntu 20 at 1920x1080 @ 60 FPS with above proposed changes.  No GLFW, no glew.
https://github.com/user-attachments/assets/8ee20a5e-f78f-4b2e-8b97-2c9168a4df24

liamHowatt, 2024-12-10:
Hi, thanks for mentioning it. When I picked up the OpenGL, I knew it was only a matter of time before this separation needed to occur. The logic is separated to a reasonable degree but not properly decoupled.

Do you want to open a PR or shall I?

jwinarske, 2024-12-10:
@liamHowatt Hi Liam.  Go for it.

My only request is to use an opaque data pointer for `window *`, opposed to a GLFW specific one, thus making the struct agnostic.  Then cast as needed per platform.  This way we can reuse the struct for any platform that supports EGL/GLES, opposed to having to redefine it for each platform.

kisvegabor, 2025-01-03:
Not stale.

liamHowatt, 2025-02-03:
Hey @jwinarske I'm working on it and if it weren't for your working demo, I would assume that the GL v4 functions LVGL is using are incompatible with EGL's GL ES version. It seems that functions like `glGenVertexArrays` should not be working on EGL without GLAD or GLEW. Do you know how you got it to work in this respect? I'm not opposed to using pure GL ES functions but if the v4 ones work with EGL somehow I'd like to know how!

closed 2025-03-10 by lvgl-bot