lvgl/lvgl #7423: NEMA GFX 9.3 performance vs DMA2D 8.3

tdjastrzebski, 2024-12-07:
### Introduce the problem

First of all, it really is not a problem per se.
Initial LVGL 9.3 NEMA GFX vs LVGL 8.3 with DMA2D performance appears to be quite good besides three particular tests: `Screen sized text`, `Containers with scrolling` and `Widgets demo`.
Even if FPS is often lower, CPU usage, render time and flush time is lower as well.
Flush time probably is lower since in my case DMA2D is exclusively available for the output buffer transfer and 32b->24b pixel format conversion done within flush_cb().
`Screen sized text` test is another issue, there may be a problem with 8.3.11 benchmark - it needs to be investigated.
I hope `Widgets demo` performance will be improved down the road.
Tested with STM32U5G9 MPU @160MHz, custom board with internal SRAM only, LCD 1024x600, LVGL 32b color, LCD 24b RGB888
LVGL DMA2D support was disabled since my tests showed no performance improvement.

[image]

kisvegabor, 2024-12-22:
Thank you for the benchmark. It's very useful that  we can see these results from an independent party. :slightly_smiling_face: 

- We should focus on Avg.time as the measurement of the others changed in the past, but the avg. time should be the same
- Screen sized text: Probably it wasn't scrolling in v8. 
- Container with opa_layer: the result of v8 seems incorrect. It should be one of the most difficult tests and the avg.time was 2 ms. Did it look good on the screen?
- Widgets demo: which parts was so slow in v9? Can you record a video?

tdjastrzebski, 2024-12-22:
Here are the most recent results using today's code (Dec 22) compared to previous ones. Some improvements are noticeable.
Containers with opa layer look good to me https://youtu.be/8dpTpYgUlNI?t=39
Containers with overlay do not look quite good https://youtu.be/8dpTpYgUlNI?t=36
Rotated images show some issues https://youtu.be/8dpTpYgUlNI?t=18
Demo slows down to just 2 fps when gauges are displayed https://youtu.be/8dpTpYgUlNI?t=53

[image]

kisvegabor, 2024-12-23:
Thank you! Do you use partial buffering? If so what is the size of the partial buffer?

kisvegabor, 2024-12-23:
cc @MarkIoanAmat

tdjastrzebski, 2024-12-23:
I realized that my previous tests still had issues.
1. Buffer sizes were not the same
2. 8.3 used hardware timers to trigger LVGL task and tick while 9.3 used software timer and thread. Software threads perform slower but not always. See the 2nd chart below.

I corrected the above issues and run tests again. Results are more pleasing. Although average 9.3 fpu performance is 13% lower, CPU usage is **39% lower**. I am wondering why fpu is lower when avg times and render times are shorter.
Now **partial** buffer sizes are equal width*height/10, in case of 9.3 multiplied by sizeof(lv_color_t). LV_MEM_SIZE is 150 kB.

Some tests show problems:
* 8.3 `Screen sized text` does not scroll
* 9.3 `Containers with overlay` and `Rotated images` do not look good
* 9.3 `Widgets demo` slows down a lot when gauges are displayed. 8.3 slows down as well but not that much.

Links to new videos:
* 8.3 https://youtu.be/4mM_YJ7NsKU
* 9.3 https://youtu.be/sP_BR_7sT4A

[image]

[image]

kisvegabor, 2024-12-23:
Can it happen that `lv_timer_handler()` is not called with the same period? Maybe in some case you wait 5ms after it and 10 ms in the other case. 

Anyway, please try with 1 ms wait time.

tdjastrzebski, 2024-12-23:
Ach, that is right, I found one more glitch. lv_tick_inc() was called with incorrect value 5 instead of 10.

Both Tick and Task are called from OS every about 10ms and this is the maximum frequency ThreadX OS allows.
Calling Tick and Task in OS context allows me for an easy synchronization with other threads updating GUI using simple global mutex and effectively use flushWaitCb. Ofc, this is not the only possible approach, but it is easy and effective. Otherwise, I would probably have to update GUI from within LV Task before or after LV operations to avoid conflicts - that is, to prevent LV objects modifications while they are being rendered. Effective flushWaitCb without OS is a separate topic but I do not think it could be done without significant LVGL modifications.

Anyway, here is another 8.3.11 vs 9.3 comparison with lv_tick_inc() proper value 10ms.
The 2nd chart presents 9.3 hardware 10ms timer vs 1ms. It shows avg 19% fps improvement, but I am not sure calling Tick and Task every 1ms would be a good idea from the overall system perspective. I think 10 ms is more than reasonable.

[image]

[image]

tdjastrzebski, 2025-01-04:
@kisvegabor I think it would be good to improve/fix 8.3.11 `lv_demo_benchmark()` to make sure we compare 'apples to apples'.
Probably, it would also make sense to move to 8.4 branch.

kisvegabor, 2025-01-09:
If there is a 10ms it might not be able to reach the set 30 FPS is rendering was e.g. 28 ms and there was a 10ms delay. So for the test I think it's better to go with a lower delay.

In v9 really there is an image rotation on the gauge but it was super fast in the earlier test.

tdjastrzebski, 2025-01-25:
@kisvegabor but why this particular part? what do your tests show?

kisvegabor, 2025-02-07:
With @liamHowatt we haven't seen much slowdown on he widgets demo with the boards we have tested recently.

tdjastrzebski, 2025-02-07:
Could you please share any test code for STM32U5G9 (any board) known to work well?
I can try to run it and compare.

kisvegabor, 2025-02-24:
Please take a look at this: https://github.com/lvgl/lv_port_riverdi_stm32u5

closed 2025-03-28 by lvgl-bot