lvgl/lvgl #7424: STM32U5 ARGB8888 format problem

Germ3, 2024-12-08:
### LVGL version

V9.3

### Platform

STM32U599NJH Custom board

### What happened?

I’m working with a 480x480 DSI 24-bit display, and while most things are functioning well, I’m running into trouble when trying to enable the ARGB8888 color format. Currently, the RGB565 color format works great, but I’d like to leverage the full potential of the display.

I’ve used [@liamHowatt's repository](https://github.com/liamHowatt/lv_port_riverdi_stm32u5/tree/nema) as a starting point, and it has been very helpful so far. However, when switching to ARGB8888, the library seems to hang in the following function: ```static void wait_for_flushing(lv_display_t * disp) ```

Here’s a summary of the changes I made to enable ARGB8888:
lv_conf.h: ```#define LV_COLOR_DEPTH 16 ``` to  ```#define LV_COLOR_DEPTH 32 ```
LTDC.c: 	```pLayerCfg.PixelFormat = LTDC_PIXEL_FORMAT_RGB565``` to ```pLayerCfg.PixelFormat = LTDC_PIXEL_FORMAT_ARGB8888``` 
lvgl_port_display.c:  ```DMA2D->OPFCCR = DMA2D_OUTPUT_RGB565;``` to ```DMA2D->OPFCCR = DMA2D_OUTPUT_ARGB8888;``` 
dma2d.c ```hdma2d.Init.ColorMode = DMA2D_OUTPUT_RGB565;*/``` to ```hdma2d.Init.ColorMode = DMA2D_OUTPUT_ARGB8888; ```

Updated the memory definitions in my linker script to accommodate the increased memory requirements (480x480 pixels × 4 bytes for ARGB). The updated configuration looks like this:
```/* Memories definition */
MEMORY
{
  FLASH	(rx)	: ORIGIN = 0x08000000, LENGTH = 4096K
  RAM2	(xrw)	: ORIGIN = 0x20000000, LENGTH = 1000K
  RAM	(xrw)	: ORIGIN = 0x200F4240, LENGTH = 1496K
} 
```
Despite these changes, the display behaves incorrectly when using ARGB8888. The bottom portion of the screen is never filled, as shown in the image below:
[image]

I hope this explanation is clear. If anyone has experience with enabling ARGB8888 on a similar setup or can point me in the right direction, I’d greatly appreciate your input.

Thank you in advance for your help!

### How to reproduce?

```c ```

kisvegabor, 2024-12-22:
@liamHowatt is already on holiday, but I'll ask him to take a look in January. 

A few noted from me: 
- It seems like all the rendering is correct just LVGL
   - The top part of the screen is ok
   - The bottom part is ok
   - The colors are ok
   - The shapes are ok 
So LVGL just thinks that the vertical resolution is smaller. **Have you changed [this too](https://github.com/liamHowatt/lv_port_riverdi_stm32u5/blob/nema/Core/Src/lvgl_port_display.c#L25)?**   
- FYI, `LV_COLOR_DEPTH` sets `LV_COLOR_FORTMAT_XRGB8888` for LVGL (Instead of ARGB8888). It's good because ARGB8888 rendring is slower.

liamHowatt, 2025-01-08:
I'll take a look soon.

> So LVGL just thinks that the vertical resolution is smaller. **Have you changed [this too](https://github.com/liamHowatt/lv_port_riverdi_stm32u5/blob/nema/Core/Src/lvgl_port_display.c#L25)?**

That seems like a good place to start.

Germ3, 2025-01-08:
Apologies for the delayed response—I’ve been really unwell over the past two weeks.

I’ve tried disabling the NEMA GFX, but unfortunately, that didn’t help. I’ve also experimented with various settings, including increasing the buffer size as you mentions to 2x, 3x, and even 4x. However, this didn’t resolve the issue; in fact, using a 3x or 4x buffer made things worse.

The only setting that works as expected is RGB565. With this configuration, everything functions correctly.

I’ve attached the relevant configurations for the LVGL library. To switch from RGB565 to RGB888, I make the following changes:

lv_conf.h: `#define LV_COLOR_DEPTH 24`
LTDC.cpp: `pLayerCfg.PixelFormat = LTDC_PIXEL_FORMAT_RGB888;`
lvgl_port_display.c:` DMA2D->OPFCCR = DMA2D_OUTPUT_RGB888;`
That should be all, right? ` DMA2D->FGPFCCR` automatically adjusts based on `LV_COLOR_DEPTH`.

However, with RGB888, the display output is incorrect. I’ve attached comparison pictures below:

- RGB565:

[image]

- RGB888

[image]

uLipe, 2025-01-08:
Hmm, just one question, since you are moving from a 16 to 24-bit RGB interface, did you check if the pinmux on the STM32 chip was updated? Please have a check on it, just updating the pixel format in the LTDC driver will not impact on the pinmux arrangement.

Also please check if your display controller is also being set to handle an ARGB8888 bus width, the controller must be configured to that format way before the LVGL and LTDC driver starts to enter in action

Germ3, 2025-01-09:
Well I am using a MIPI DSI screen that is always configured as RGB888, otherwise the LCD won't work at all. 
When using another GUI library like TouchGFX with the same initialization code, ARGB8888/RGB888 does work.
I will upload my repo this evening so you can have a look at the project if that helps.

uLipe, 2025-01-09:
That is great thanks for sharing that @Germ3

kisvegabor, 2025-01-13:
cc @liamHowatt

kisvegabor, 2025-01-29:
not stale

Germ3, 2025-02-20:
Is there any update on this? Still struggling with my project unfortunately.

liamHowatt, 2025-02-28:
Hi @Germ3, I got LTDC ARGB8888 to work on our STM32U5 port with minimal hassle. Can you see if there's anything majorly different with yours and mine?

https://github.com/lvgl/lv_port_riverdi_stm32u5/compare/master...liamHowatt:lv_port_riverdi_stm32u5:ltdc-argb8888

liamHowatt, 2025-02-28:
I see a potential issue in your linker script. The `K` in `LENGTH = 1000K` means 1024. You've told the linker that 1024000 bytes follow 0x20000000, but 0x200F4240 is only 1000000 bytes away from 0x20000000.

I don't think it's the root cause of the issues though, unfortunately, because the framebuffer area is still below 0x200F4240.

Nevertheless, try this (I don't expect this to fix it)
```
MEMORY
{
  FLASH	(rx)	: ORIGIN = 0x08000000, LENGTH = 4096K
  RAM2	(xrw)	: ORIGIN = 0x20000000, LENGTH = 900K
  RAM	(xrw)	: ORIGIN = 0x200e1000, LENGTH = 1596K
}
```

Germ3, 2025-03-04:
@liamHowatt I got it working thank you so much! 
The linker script did not work as you expected. But looking at your code you used a later version of LVGL. 
So now I also use the ST driver built in the library. 
And now ARGB8888 is working fine. 

Again thank you so much! Keep up the good work!