lvgl/lvgl #7425: SVG library: char-subscripts error from usage of ctype.h Macros

sidharth-458, 2024-12-08:
### LVGL version

v9.3.0

### Platform

ESP32 using ESP-IDF

### What happened?

When -Wall is used for compilation the error comes out,
`array subscript has type 'char' [-Werror=char-subscripts]` 
for the lines involving MACROS in ctype.h include.

### How to reproduce?

manually add LVGL master to idf.manifest and add LV_USE_SVG at project's base c-makefile, then at compile time these issues will arise.

I guess to use c- type.h the pointer passed needs to be recast as unsigned char

kisvegabor, 2024-12-22:
@onecoolx Could you take a look at this issue? 

@sidharth-458 Please add a longer compiler log to better see the problem.

onecoolx, 2024-12-30:
@sidharth-458 I'm not sure what the problem is, could you send me a complete compile log containing this error?

closed 2025-01-22 by lvgl-bot