lvgl/lvgl #7426: Just a suggestion regarding PR's

kdschlosser, 2024-12-09:
### Introduce the problem

Right now there seems to be a lot of bug fixes that gets made but the issue that has been opened to fix the bug might not be getting closed.

### Proposal

I recommend changing the template for a PR so it contains the "fixes" trigger keyword that GitHub looks for in the description of the PR.

Example is..

```
Title of PR...

This would be the PR's description. I recommend adding at the very bottom the following.

Fixes #7141 
```

where everything from the pound sign afterwards is set by the person opening the PR.. 

What this does is it tells GitHub to close the issue once  the PR has been merged. 

https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/using-keywords-in-issues-and-pull-requests

kisvegabor, 2024-12-16:
Good idea, PR opened #7462

kdschlosser, 2024-12-16:
It's good because it lets the person that opened the issue know that there is work that is being done to correct the problem they are having. It allows them to to go to the PR and make comments because it tells them that there is work being done on it. It also automatically closes the issue when the PR gets merged. It makes things easier for everyone I think.

kdschlosser, 2024-12-31:
not stale

PR #7462 (2025-01-02): chore(pr-template): add 'Fixes' to the template
fixes #7426

cc @kdschlosser 

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code-format.py [... cut]
closed 2025-01-02 by uLipe