lvgl/lvgl #7431: `lv_fs_drv_t` single field `letter` appears to assume that there is only one drive

vwheeler63, 2024-12-09:
### Introduce the problem

There is a hard-to-find conflict that *might* be worth resolving.

### Background Information

It is expected that the user only ever needs to call `lv_init()` once, which is as it should be.

While it is running, if `LV_USE_FS_WIN32` is configured as non-zero, then `lv_fs_win32_init()` runs exactly once, which is as it should be.

`lv_fs_drv_t` has a single field `char letter`, which means that one needs to call `lv_fs_drv_register` once for each storage drive on a system that LVGL might need to use, which can make sense as a design intention.

This observation is reinforced by the fact that `lv_global_t` has a single field `fsdrv_ll` which appears to indicate that the designer intended there to be possibly more than one registered drive in a system usable by LVGL, and that linked list ONLY stores a pointer to a single `lv_fs_drv_t` for each registered drive.  So *somewhere* in RAM there has to be a `lv_fs_drv_t` struct allocated for each drive that will be registered.

### The Visible Conflict in  Examples

`lv_global.h` has only one field `win32_fs_drv`, which -- according to the storage layout -- assumes the there is only one `lv_fs_drv_t` (member field `win32_fs_drv`), which limits a WIN32 system to having only one drive involved with `lv_fs`.

I can see how this might be a logical or practical assumption that LVGL images might only be stored on one drive, but this assumption conflicts with all the `lv_example_file_explorer_*.c`:  all 3 examples currently use 2 different drives.  This cannot work given the hard-coded single `win32_fs_drv` field in `lv_global_t`.

### The Underlying Design Conflict

It seems to me that, to be thorough, when LVGL is running on a WIN32 system, that there should be allowances to have LVGL `lv_fs` possibly deal with more than one drive.

If this is the case, then having a single field `win32_fs_drv` in `lv_global.h` is an error and instead it should be an array.

### Proposal

### Possible Solution

- When `LV_USE_FS_WIN32` is in `lv_conf.h` as non-zero, then `lv_conf.h` needs an additional  macro: `LV_FS_WIN32_DRIVE_COUNT` to be the number of drives to be registered.
- `lv_global.h::lv_global_t` instead of having a single field `win32_fs_drv` instead has an array:  `win32_fs_drives` whose capacity is determined by `LV_FS_WIN32_DRIVE_COUNT`.
- `lv_fs_win32.c` should enumerate through all the drives on the system, and register the first `LV_FS_WIN32_DRIVE_COUNT` drives that come up in the list during enumeration, using each element in the array in turn.

That would then make the 3 `lv_example_file_explorer_*.c` examples valid.

Kind regards,
Vic

liamHowatt, 2024-12-10:
I think this is a weakness that all the builtin FS drivers have. I see what you mean about how the examples emphasize the use of multiple drive letters.

In my opinion -- I know it's limiting and inflexible -- but I think there shouldn't be multiple of the same driver registered as multiple drive letters because there isn't a real need to, that I can think of. If a user wants `A` to be their permanent images and `B` to be their dynamic USB drive images, I think they should set `LV_FS_WIN32_PATH` such that `A` can reach both paths, and optionally use symlinks to shorten paths.

If they user multiple letters for the same device, they can implement the interface themselves and call `lv_fs_drv_register` multiple times.

Could the examples be made less misleading in your opinion?

vwheeler63, 2024-12-10:
> Could the examples be made less misleading in your opinion?

After I realized what the ACTUAL problem was in issue #7432, that first "drive letter" was not a drive letter at all, but a *driver-identifier-letter*, which is a whole different deal, and the same single driver (identified by that letter) can be used with all valid paths on a Windows system, and that means all valid drives.  So the misunderstanding about what that identifier letter was means there is no actual issue here, since Windows systems are not limited at all by having one FS driver.

closed 2024-12-10 by vwheeler63