lvgl/lvgl #7438: lv_demo_vector_graphic_not_buffered() demo hard fault

tdjastrzebski, 2024-12-10:
### LVGL version

9.3.0.dev

### Platform

STM32U5G9, GCC 13 compiler

### What happened?

Upon start `lv_demo_vector_graphic_not_buffered()` crashes.

`lv_demo_vector_graphic_buffered()` I was unable to run yet - I need to free up some SRAM memory or use the just announced STM32N6 MPU.

hard fault UFSR: 0x0000, BFSR: 0x82, MMFSR: 0x00, ABFSR: 0x00000000, HFSR: 0x40000000

[image]

[image]

uLipe, 2024-12-10:
@tdjastrzebski interesting, would you mind to replace the lv_memset optimized to the plain memset from libc and test if the thing actually work?

I'm suspecting of problem related to the addresses alignment which may cause usage/hard fault on these case, does the SRAM region have any alignment restrictions? (I.e.: access only in words or halfwords)

tdjastrzebski, 2024-12-11:
@uLipe after setting `#define LV_USE_STDLIB_STRING    LV_STDLIB_CLIB` in lv_conf.h the problem still persists.
All the other demos work OK, besides eBike but eBike seems to suffer from some unrelated issue.

liamHowatt, 2024-12-12:
> I need to free up some SRAM memory

Are you saying that a memory allocation failed and `dst` is `NULL` which is why there was a hard fault?

tdjastrzebski, 2024-12-12:
@liamHowatt Not exactly, `lv_demo_vector_graphic_not_buffered()` crashes with hard fault.
BFSR = 0x82 indicate that the fault was caused directly by the executing instruction, UFSR = 0 say that it was not caused e.g. by unaligned access operation. ref: https://interrupt.memfault.com/blog/cortex-m-hardfault-debug

`lv_demo_vector_graphic_buffered()` is a different story. Yes, buffer allocation fails, perhaps because I have 1800 kB LTDC buffer (1024x600 24b), NEMA_GFX requires 600 kB buffer and on top of that demo wants to create another, I believe 600 kB buffer. This may be a bit too much since there would be not SRAM left (total STM32U5G9 SRAM is 3MB).

liamHowatt, 2024-12-12:
So just to be clear, it is an out-of-memory issue?

tdjastrzebski, 2024-12-12:
@liamHowatt nope, this bug is related to lv_demo_vector_graphic_**not**_buffered() which should have enough memory.

uLipe, 2024-12-12:
@tdjastrzebski for the fault hints, it seems this is an invalid memory access fault, in your backtrace is it possible to see the parameters passed to the memset? Probably dest parameter has been set to an invalid location?

tdjastrzebski, 2024-12-12:
@uLipe Indeed, dst address is 0x0 but this can be caused by LVGL not being able to allocate more memory within `lv_ll_ins_tail()` function. Well, maybe the remaining 600 kB SRAM is not enough to run this demo. I suggest adding error trace.

[image]

uLipe, 2024-12-18:
It seems there is a malloc assertion, I'm curious to understand why it was not triggered.

tdjastrzebski, 2024-12-19:
[image]

uLipe, 2024-12-20:
NVM @tdjastrzebski I was looking at the wrong place, even though an assertion might be helpful there, it is a condition that should not happen.

uLipe, 2024-12-20:
Looking at lv_ll_ins_tail() it seems there is an allocation opération for the new node which returns null when it fails.

this allocation fails only when there is no memory available for that, would you are suffering from a OoM issue on your platform?

tdjastrzebski, 2024-12-20:
@uLipe Likely I do. Currently I test on STM32U5G9 with no external RAM and 1024x600 32b resolution (LTDC is only 24b).
It seems that this resolution is too high for this demo to run without external RAM.

closed 2025-01-17 by lvgl-bot