lvgl/lvgl #7439: lv_demo_ebike() glitches

tdjastrzebski, 2024-12-10:
### LVGL version

9.3.0.dev

### Platform

STM32U5G9, GCC 13 compiler on Windows, NEMA GFX enabled

### What happened?

[image]

uLipe, 2024-12-10:
@tdjastrzebski it would be possible to you  to check if there is log output? In special for LOG_WARNING and LOG_ERR tags?

tdjastrzebski, 2024-12-12:
@uLipe There are no warnings or errors in log. NEMA_GFX seems to be the culprit. Without NEMA_GFX it works ok. It also works well with just DMA2D enabled.
cc @liamHowatt

liamHowatt, 2024-12-12:
cc @MarkIoanAmat

MarkIoanAmat, 2024-12-12:
We will look into it and keep you updated.

MarkIoanAmat, 2024-12-18:
Update: we found the issue regarding this glitch, I will follow up with a new PR.
@tdjastrzebski @liamHowatt

kisvegabor, 2025-01-03:
Not stale

PR #7759 (2025-02-25): fix(nema): Broken build after draw unit refactoring, draw_label changes
Fixes #7691
Fixes #7723
Fixes #7439

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig). [... cut]
closed 2025-02-25 by kisvegabor