lvgl/lvgl #7442: App crashes when trying to access any widget that was created in another function even with globally define lv_obj_t* for that widget

cempod, 2024-12-11:
### LVGL version

v8.2.0

### Platform

Eclipse port simulator; Zephyr Rtos v3.7.0

### What happened?

After creating any widget outside of main function, when you try to do something with this widget app crashes with seg fault or, for example if you want to change text of label widget it fails in assert check `obj != NULL` in lv_label.c:87 

### How to reproduce?

https://github.com/cempod/lv_port_pc_eclipse/blob/help_me/interface/main_page.c

Above is the link with my fork of lv_port_pc_eclipse where I reproduced the issue. File `interface.main_page.c` contains commented out lines 32-34 to reproduce the issue.

liamHowatt, 2024-12-11:
[Try this](https://github.com/cempod/lv_port_pc_eclipse/compare/help_me...liamHowatt:lv_port_pc_eclipse:patch-1).

You need double-pointers for what you're trying to do.

cempod, 2024-12-11:
Ty very much, I'm literally crying out of happiness, it works as it should except that on lines 34 34 it works without additional `&` symbol, because it's lvgl's functions after all.

closed 2024-12-11 by cempod

liamHowatt, 2024-12-11:
Happy to help, glad it worked :slightly_smiling_face: 

> except that on lines 34 34 it works without additional `&` symbol

Ahh of course. Nice you were able to work around my mistake.