lvgl/lvgl #7450: could realize the 1/2 meter and 1/4 meter?

xinsuinizhuan, 2024-12-12:
### Introduce the problem

1/2meter as the:
[image]

1/4meter as the:
[image]

lizhaoming634, 2024-12-12:
Use a background image and use 
 lv_meter_set_scale_range(lv_obj_t * obj, lv_meter_scale_t * scale, int32_t min, int32_t max, uint32_t angle_range,
                              uint32_t rotation) , You can set "angle_range" to 180 degrees to achieve 1/2 of the meter

xinsuinizhuan, 2024-12-12:
> 180
    it is a circle, not half circle:
    lv_obj_t* meter = lv_meter_create(lv_scr_act());
    lv_obj_center(meter);
    lv_obj_set_size(meter, 200, 200);

    /*Remove the circle from the middle*/
    lv_obj_remove_style(meter, NULL, LV_PART_INDICATOR);

    /*Add a scale first*/
    lv_meter_scale_t* scale = lv_meter_add_scale(meter);
    lv_meter_set_scale_ticks(meter, scale, 6, 2, 10, lv_palette_main(LV_PALETTE_GREY));
    lv_meter_set_scale_major_ticks(meter, scale, 1, 2, 10, lv_color_hex3(0xeee), 8);
    lv_meter_set_scale_range(meter, scale, 0, 100, 180, 90);

xinsuinizhuan, 2024-12-12:
[image]

lizhaoming634, 2024-12-12:
use   lv_obj_set_style_border_opa(meter,LV_OPA_0, LV_PART_MAIN);

closed 2025-01-07 by lvgl-bot