lvgl/lvgl #7451: flex layout inside button does not click (9.3.0.dev)

tdjastrzebski, 2024-12-12:
### LVGL version

9.3.0.dev

### Platform

GCC 13 compiler on windows, STM32U5G9 board

### What happened?

I wanted to create a complex button containing two rows, each with two labels and exact spacing between them.
The button looks good, just as I want it to look but the problem is that this button is clickable only on its edges.
It seems like the flex layout (generic object) does not inherit button’s ‘clickability’.
Adding: `lv_obj_add_flag(div, LV_OBJ_FLAG_CLICKABLE)` does not help.

[image]

### How to reproduce?

lv_obj_t* cont = lv_screen_active();
lv_obj_t* btn;
lv_obj_t* lbl;
lv_obj_t* div;

btn = lv_button_create(cont);
lv_obj_set_size(btn, LV_SIZE_CONTENT, LV_SIZE_CONTENT);
lv_obj_set_flex_flow(btn, LV_FLEX_FLOW_COLUMN);
lv_obj_set_style_pad_row(btn, 10, 0);

// row 1
div = lv_obj_create(btn);
lv_obj_remove_style_all(div);
lv_obj_set_size(div, LV_SIZE_CONTENT, LV_SIZE_CONTENT);
lv_obj_set_flex_flow(div, LV_FLEX_FLOW_ROW);
lv_obj_set_style_pad_column(div, 20, 0);

lbl = lv_label_create(div);
lv_label_set_text_fmt(lbl, "long long text");
lbl = lv_label_create(div);
lv_label_set_text_fmt(lbl, "short");

// row 2
div = lv_obj_create(btn);
lv_obj_remove_style_all(div);
lv_obj_set_size(div, LV_SIZE_CONTENT, LV_SIZE_CONTENT);
lv_obj_set_flex_flow(div, LV_FLEX_FLOW_ROW);
lv_obj_set_style_pad_column(div, 20, 0);

lbl = lv_label_create(div);
lv_label_set_text_fmt(lbl, "medium 1");
lbl = lv_label_create(div);
lv_label_set_text_fmt(lbl, "medium 2");

liamHowatt, 2024-12-12:
Use `lv_obj_add_flag(div, LV_OBJ_FLAG_EVENT_BUBBLE)` to bubble the click event up to the parent.

tdjastrzebski, 2024-12-12:
@liamHowatt Thank you! It helped. Maybe this flag should be there by default? However, I have no opinion, having a real project and GUI to build now I need to quickly learn such LVGL nuances.

uLipe, 2024-12-12:
@tdjastrzebski glad that the help worked, I will be closing this one, but feel free to reopen in case of new discovery.

closed 2024-12-12 by uLipe

kisvegabor, 2024-12-18:
I think a more common solution would be to make the children not clickable:
```c
lv_obj_remove_flag(div, LV_OBJ_FLAG_CLICKABLE);
```

tdjastrzebski, 2024-12-18:
@kisvegabor Thank you, it works well. So, instead of receiving the event and then forcing it to bubble up it is better not to receive the event which then will be received (and handled) by the parent container. Makes sense.