lvgl/lvgl #7452: Optionsl Extensions and Include Patterns

vwheeler63, 2024-12-12:
### Introduce the problem

The following jumped out at me because of a redundancy in the `.c` file you will see below.  But the redundancy is only a minor issue (and probably should be handled), but looking into it exposed a possibly larger issue -- or possibly something I don't understand yet, so please guide me if so.

At the top of `lv_freetype.h` is this:
```c
/*********************
 *      INCLUDES
 *********************/
#include "../../lv_conf_internal.h"
#include "../../misc/lv_types.h"
#include "../../misc/lv_event.h"
#include LV_STDBOOL_INCLUDE

#if LV_USE_FREETYPE
...all relevant content is here
#endif /*LV_USE_FREETYPE*/
```

and at the top of the `lv_freetype.c` file is this:
```c
/*********************
 *      INCLUDES
 *********************/
#include "../../misc/lv_fs_private.h"
#include "lv_freetype_private.h"

#if LV_USE_FREETYPE

#include "lv_freetype_private.h"
#include "../../core/lv_global.h"

...all content here

#endif /*LV_USE_FREETYPE*/
--end-of-file--
```

What I am not understanding is this:  since there is no actual content below the `#if LV_USE_FREETYPE` conditional directive, what is the purpose for having `#include` directives outside the conditional directive?

(`#include "lv_freetype_private.h"` outside the conditional directive the in `.c` file appears redundant to me, but its resolution depends on the answer to the above question.)

Kind regards,
Vic

### Proposal

I am happy to resolve (or not) depending on the answer to the above question.

liamHowatt, 2024-12-12:
In my opinion:

- In headers, the only include that should come before `#if LV_USE_...` is `#include "../lv_conf_internal.h"`.
- In C files, the only include that should come before `#if LV_USE_...` is `#include "(the header with the closest correspondence to this C file)"`.

vwheeler63, 2024-12-12:
Additional:

So far my search has taken me to an `#include`-mapping utility in the compiler which shows me:
```
lv_freetype.c
  +-- lv_fs_private.h
        +-- lv_fs.h
              +-- lv_conf_internal.h
                    +-- lv_conf.h

So what I am seeing is that *something* has to include `lv_conf_internal.h` after which the *rest of the plan* is defined in the
```c
/* If lv_conf.h is not skipped, include it. */
#if !defined(LV_CONF_SKIP) || defined(LV_CONF_PATH)
    ...
#endif
```

block in the `lv_conf_internal.h` file.  So I now have an additional question:  what is LVGL designer's policy for where an LVGL source file should include the `lv_conf_internal.h` file?  Observation:  it does not seem to be extremely consistent:  out of 32 widgets, only 18 include this file in their `.h` files.  Or perhaps it is only included when a configuration macro matters to that code?

tdjastrzebski, 2024-12-12:
I mentioned it already somewhere else but using preprocessor macros after `#include` is generally discouraged in C/C++.
Just my 3 cents.

vwheeler63, 2024-12-12:
...re-studying [LVGL configuration](https://docs.lvgl.io/master/intro/add-lvgl-to-your-project/configuration.html)....

vwheeler63, 2024-12-12:
> In my opinion:
> 
>     * In headers, the only include that should come before `#if LV_USE_...` is `#include "../lv_conf_internal.h"`.
> 
>     * In C files, the only include that should come before `#if LV_USE_...` is `#include "(the header with the closest correspondence to this C file)"`.

That certainly makes the most sense to me, Liam.  Before I proceed, I would like to pick @kisvegabor 's brain on this.  Gábor, in my documentation endeavors, if I encounter oddities (like the redundancy in the `.c` file above) would it be appropriate for me to modify as per this policy?  Or is there more to "the plan" that I am not yet aware of?

vwheeler63, 2024-12-27:
Not stale.

vwheeler63, 2025-01-04:
@kisvegabor I find this calls for a paragraph in the coding standards about includes.  Something short similar to what Liam wrote above.  What do you think?

kisvegabor, 2025-01-07:
It makes sense to me. ([Comment](https://github.com/lvgl/lvgl/issues/7452#issuecomment-2539390274)). Short and simple.

PR #7524 (2025-01-03): fix(lv_freetype):  clean up #includes in lv_freetype.*
A redundant include in `lv_freetype.c` caught my attention and I found the include logic among the main 3 files needing some clean-up. The clean-up moved #includes so that only just the #includes needed to provide the `LV_USE_FREETYPE` macro were kept above that #if directive, and all others were moved below it.  Compilation was tested both with and without `LV_USE_FREETYPE` enabled and I can confirm nothing was broken.  (And the redundant #include was removed.)

Resolves #7452

### Notes
- Update the [Documentation](https://github.com/lvg [... cut]