lvgl/lvgl #7454: Introducing LVGL's UI editor

kisvegabor, 2024-12-12:
### Introduce the problem

Hi Guys,

I'm happy to share that we have published a [detailed video about our UI editor](https://youtu.be/ntAuLhuK3Ck): 

[image]

The main features of this tool include:
- **Component-oriented**: Various elements of the UI can be implemented as reusable components.
- **XML-based**: The components are described in an HTML-like form.
- **Instant preview**: The components can be previewed as you edit them.
- **Figma support**: LVGL's Figma plugin helps to quickly extract styles from any Figma elements.
- **Online preview**: A CI action converts the XML files into a website where they can be previewed.
- **Custom widget creation**: Unlike components, widgets have C logic. You can recompile the editor to make it contain the new widgets' code.
- **C export**: Both components and widgets can be exported to C, enabling seamless integration into your application, just like handwritten code.
- **Runtime XML loading**: Components can be loaded from XML at runtime without recompiling the firmware.

The target release date of v0.1 is 2025 January. v1.0 is planned at 2025 September. 

We would love to hear what you think about this tool and concept. Please leave your feedback here: https://github.com/lvgl/lvgl_editor/issues

C47D, 2024-12-13:
Congratulations @kisvegabor , it's amazing how far the project has grown, I am sure a lot of great things will come in the future. :smile_cat:

lizhaoming634, 2024-12-13:
I hope there are more tutorials, such as how to create individual widgets.

Staok, 2024-12-13:
Wonderful work. It is similar to Qt's QML programming, so it is become more convenient to modify the ui. Many dynamic events related to the ui can  be handled in the front end(in xml ?). How will XML interact with the C/C++ back end of the program? If it can be like QML, it will be much more convenient and more modern!!!

kisvegabor, 2024-12-13:
@lizhaoming634 

> I hope there are more tutorials, such as how to create individual widgets.

IT was just requested in a YouTube comment as well. (Maybe it was also you? :slightly_smiling_face: ) Anyway , we really should add some docs about it. However if the editor generated the skeleton the rest should the straightforward. 

@Staok 
> How will XML interact with the C/C++ back end of the program? If it can be like QML, it will be much more convenient and more modern!!!

Well, in QML you can use JavaScript so that it can be parsed at runtime.  In our case - working in C - it we need to approach it differently. I have something like this in mind:
```c
//In C
lv_xml_register_event_cb("led_on", my_led_on_cb);
```

```xml
<!-- In XML -->
<lv_button width="100" height="50">
   <events>
      <clicked>
         <call_function name="led_on" user_data="led1"/>
         <start_animation name="anim1"/>
         <set_subject name="subject" value="10"/>
         <update_widgget name="label1" property="text" value="LED 1 is ON"/>
      </clicked>
   </events>

   <lv_label text="LED ON" align="center"/>
</lv_button> 
 ```

Staok, 2024-12-14:
@kisvegabor 
what I am doing is a big project that contains more than 5k lv objs by using squareLine to construct the lvgl frent end(the ui), what I need if I trans the whole front end work to another way, these are the best urgent:
1、 Our UI designer use figma, if it can directly trans it to lvgl editor ui.
2、Lvgl default widgets are far more NOT enough for high custom UI, we have almost 100 custom "component" of squareLine, so if lvgl editor support this feature can be great, and, c/c++ backend need to flexibly modify the "component", and add event cb to it.
3、Keep all the lvgl obj "style" accessable to the xml editor, and the different "style" of different "state", we need high custom functions to design.
4、and so on...I just think out these this time

vwheeler63, 2024-12-16:
@kisvegabor 

Looks like an immense and beautiful piece of work!  I look forward to getting my feet wet with it!

kisvegabor, 2024-12-17:
@Staok Cool! We are aiming for this level of flexibility. :crossed_fingers:

kisvegabor, 2025-01-03:
I close it for now. 

We plan the release of v0.1 for next week, so a new video and announcement will come soon. :slightly_smiling_face:

closed 2025-01-03 by kisvegabor