lvgl/lvgl #7463: Shadow for button has bg image

fatihozen3, 2024-12-16:
### Introduce the problem

I use lvgl with 9.2.2 version on NXP imxrt platform. I have a triangle and want to add some styles on it.

```
    img_triangle = lv_button_create(lv_screen_active());
    lv_obj_remove_style_all(img_triangle);
    lv_obj_set_pos(img_triangle, 660, 131);
    lv_obj_set_size(img_triangle, 129, 102);
    lv_obj_set_style_bg_image_src(img_triangle, "F:/triangle.bin", LV_PART_MAIN | LV_STATE_DEFAULT);
 lv_obj_set_style_bg_image_recolor_opa(img_triangle, 255U, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_image_recolor(img_triangle, lv_color_hex(0x0eb4ff), LV_PART_MAIN | LV_STATE_DEFAULT);
	lv_obj_set_style_transform_scale_y(img_triangle, 280, LV_STATE_PRESSED);
	lv_obj_set_style_transform_scale_x(img_triangle, 280, LV_STATE_PRESSED);
	lv_obj_set_style_x(img_triangle, 20, LV_STATE_PRESSED);
	lv_obj_set_style_y(img_triangle, 240, LV_STATE_PRESSED);
    lv_obj_set_style_shadow_width(img_triangle, 25, 0);
    lv_obj_set_style_shadow_color(img_triangle, lv_color_hex3(0x888), 0);
```

I used above code block to see it on screen. As you see below picture, shadow's seen after 129x102 rectangle area not near triangle picture. Is there a way to see shadow's near unrectangle(triangle, circle etc.) images ? 

[image]

liamHowatt, 2024-12-19:
I don't believe so, unfortunately, but maybe someone will prove me wrong.

kisvegabor, 2024-12-22:
I can confirm @liamHowatt's comment. What you need is drop-shadow which is quite complex and slow. What we have now is box shadow which draws the shadow on an (rounded) rectangle. 

We are planning to add  blur support to LVGL and drop shadow can be considered a special version of blurring, so it might be added to LVGL in 2025.

closed 2024-12-26 by fatihozen3