lvgl/lvgl #7469: In which version is the LV_SYMBOL_DEF macro used?

jor1178, 2024-12-17:
### LVGL version

v9.2.2

### Platform

In which version is the LV_SYMBOL_DEF macro used?

### What happened?

In which version is the LV_SYMBOL_DEF macro used?

jor1178, 2024-12-17:
Is there this macro that the AI told me, is it false information?

vwheeler63, 2024-12-17:
`LV_SYMBOL_DEF` does not appear in either v9.x or v8.x as a macro on its own.  However, in both versions there is a file called `lv_symbol_def.h` which has a guard macro `LV_SYMBOL_DEF_H` and which file defines a number of symbols (icons) that are available in the included fonts.

In what context are you seeing LV_SYMBOL_DEF?  Are you getting some kind of error message?

jor1178, 2024-12-18:
I want to simplify a set of picture names, similar to C language ## concatenation variable names. I got the above information from the AI, so I want to make sure. For example:

uint8_t i=1;

LV_SYMBOL_DEF(img,i)=img1;

i=2;

LV_SYMBOL_DEF(img,i)=img2;

i=3;

LV_SYMBOL_DEF(img,i)=img3;

jor1178, 2024-12-18:
This is what the AI gave me：

#include "lvgl/lvgl.h"
 
// 假设所有图像资源都已经通过LV_IMAGE_DECLARE声明
 
static void dial_hand_update_display(lv_obj_t * cont)
{
    time_t timep;
    struct tm *tp;
    time(&timep);
    tp = localtime(&timep);
 
    lv_obj_t *img_hour1_src = lv_obj_get_child(cont, 1);
    lv_obj_t *img_hour2_src = lv_obj_get_child(cont, 2);
 
    // 获取小时数的十位和个位
    char hour1_char = (tp->tm_hour / 10) % 10 + '0'; // 十位
    char hour2_char = tp->tm_hour % 10 + '0';       // 个位
 
    // 根据字符设置图像源
    const lv_img_dsc_t *img_hour1 = lv_img_get_src(lv_img_src_t(LV_SYMBOL_DEF(H_ ## hour1_char)));
    const lv_img_dsc_t *img_hour2 = lv_img_get_src(lv_img_src_t(LV_SYMBOL_DEF(H_ ## hour2_char)));
 
    lv_img_set_src(img_hour1_src, img_hour1);
    lv_img_set_src(img_hour2_src, img_hour2);
 
    lv_obj_set_pos(img_hour1_src, 55, 50);
    lv_obj_set_pos(img_hour2_src, 175, 50);
}

jor1178, 2024-12-20:
ok,thanks!

closed 2024-12-20 by jor1178