lvgl/lvgl #7489: No assert malloc on multiple malloc calls

jfsimardStelpro, 2024-12-19:
### LVGL version

v9.1.0

### Platform

STM32G0C1CE with arm-none-eabi-g++ v13.2 rel1

### What happened?

Multiple malloc and malloc_zeroed calls without proper allocation assertions.

ex. :
- in lv_draw.c, functions lv_draw_create_unit() and lv_draw_add_task

kisvegabor, 2024-12-22:
Thank you for pointing it out. Could you open a Pull request to fix it?

jfsimardStelpro, 2025-01-06:
@kisvegabor , to be sure, you want me to fix it ?

uLipe, 2025-01-06:
@jfsimardStelpro , hello!

Thank you for pointing that, it is not a strict: "We want you to fix it". It is more the nature of an open source project, since you faced the issue and have them fresh into your environment, it is always welcome from your (or other contributors side) to open a PR that fix the issue in overall or partially, I'm mentioning partially, because the ASSERTION is typically a hot code path, so it is probably an incremental improvement of adding more and more assertion points while the code base evolves and asks for more checkings.

So if you have the idea where to start and put some asserts where they are missing, a pull request is welcome :)

uLipe, 2025-01-06:
@jfsimardStelpro thanks for the clarification, it will be surely added to the backlog queue.

It would be good, if could you provide use some of the paths where the assertion is missing.

Thank you!

jfsimardStelpro, 2025-01-06:
it is already mention in the description of the issue

kisvegabor, 2025-01-15:
Adding a malloc failed hook is actually good idea!

liamHowatt, 2025-01-28:
> so there's no good way of preventing this even with an allocation-failed hook.

Could the hook be passed the pointer that failed to be realloc'd so it has the choice to free it?

davidchisnall, 2025-01-29:
You’d need to restructure a lot of the code. Most of the uses of realloc in this library appear to be unsafe: they unconditionally overwrite the pointer. You could change the behaviour of lv_realloc to match BSD’s non-standard reallocf, which would address that problem and allow you to treat realloc failures as malloc failure, but I was quite surprised to see a library that targets embedded systems not *gracefully* handling allocation failure, since allocation failure is common in resource-constrained systems. I was also quite surprised to see dynamic thread creation, which will impact memory pressure and cause allocation failures to be even harder to reason about.

liamHowatt, 2025-01-30:
I understand what you mean about `realloc`. If `realloc` returns `NULL`, the still-alive pointer is leaked and replaced with `NULL` because of how people call it. i.e. `ptr = realloc(ptr, size)` in most cases. I personally think we could add `lv_reallocf` and it can start to be used. We already added BSD `strlcpy`.

> I was quite surprised to see a library that targets embedded systems not _gracefully_ handling allocation failure, since allocation failure is common in resource-constrained systems.

How would you handle the failure to allocate, say, a roller widget, or on the other hand, a temporary layer during drawing? It seems these allocations need to succeed or the application cannot proceed in any useful way. Is that wrong?

> I was also quite surprised to see dynamic thread creation, which will impact memory pressure and cause allocation failures to be even harder to reason about.

Threads are typically allocated by draw units once and reused. Their stacks are a single allocation whose lifetime is the duration of the program.

closed 2025-03-01 by lvgl-bot