lvgl/lvgl #7493: Sliding Inertia Effect

lsj-fire, 2024-12-20:
### Introduce the problem

Hello author, I would like to ask a question. I am using lvgl8.0 version, after accessing the cst816d touch driver, when implementing the list item sliding inertia, there will be no inertia when short slide, the list even did not move, only in the screen continued to slide for a long time, and then let go to have the effect of sliding inertia, this situation, please ask how to solve this problem!

kisvegabor, 2025-01-09:
Sorry for the late reply. For custom animation you can use `lv_obj_scroll_to_x` as an `exec_cb` and set the anim_path to `lv_anim_path_ease_out`.

Natural-Fire, 2025-01-16:
> Sorry for the late reply. For custom animation you can use `lv_obj_scroll_to_x` as an `exec_cb` and set the anim_path to `lv_anim_path_ease_out`.

thanks，I'll look into it.

closed 2025-02-10 by lvgl-bot