lvgl/lvgl #7499: Images not rendering in 9.2.2

rohmer, 2024-12-22:
### LVGL version

9.2.2

### Platform

ESP32 C++, 16 bit color

### What happened?

I created an image with the online converter, it gets declared ok but during render it spits this out:
```[11:24:49][D][openhouse:025]: [Warn]    (0.628, +628)    lv_draw_sw_blend_image_to_rgb565: Not supported source color format lv_draw_sw_blend_to_rgb565.c:399
[11:24:49]
[11:24:49][D][openhouse:025]: [Warn]    (0.641, +13)     lv_draw_sw_blend_image_to_rgb565: Not supported source color format lv_draw_sw_blend_to_rgb565.c:399
[11:24:49]
[11:24:49][D][openhouse:025]: [Warn]    (0.648, +7)      lv_draw_sw_blend_image_to_rgb565: Not supported source color format lv_draw_sw_blend_to_rgb565.c:399```

It shoudl be noted that other LVGL items display to the screen just fine.  For example Windows, Text, etc.

### How to reproduce?

Image description:
```
const lv_image_dsc_t SplashImage = 
        {
            {
                LV_COLOR_FORMAT_RGB565,
                0,
                2,
                64,
                64                
            },
            sizeof(splashImage_map),
            splashImage_map
        };
```
Use of image:
```
imgObject=lv_img_create(parent);
LV_IMG_DECLARE(SplashImage);
lv_img_set_src(imgObject,&SplashImage);
lv_obj_set_align(imgObject,LV_ALIGN_TOP_MID);
```

kisvegabor, 2024-12-22:
Please check if [`LV_DRAW_SW_SUPPORT_RGB565`](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L167) is enabled.

rohmer-brightwell, 2024-12-23:
It is:
    #define LV_DRAW_SW_SUPPORT_RGB565       1
    #define LV_DRAW_SW_SUPPORT_RGB565A8     1
    #define LV_DRAW_SW_SUPPORT_RGB888       1
    #define LV_DRAW_SW_SUPPORT_XRGB8888     1
    #define LV_DRAW_SW_SUPPORT_ARGB8888     1
    #define LV_DRAW_SW_SUPPORT_L8           1
    #define LV_DRAW_SW_SUPPORT_AL88         1
    #define LV_DRAW_SW_SUPPORT_A8           1
    #define LV_DRAW_SW_SUPPORT_I1           1

kisvegabor, 2024-12-23:
Could you attach the converted image?

Have you used the [v9 image converter](https://lvgl.io/tools/imageconverter)? 
[image]

rohmer, 2025-01-12:
Sorry, ended up getting sick, finally found the issue.  There was a mismatch between C++ and C, and it was trying to display it in an invalid format

closed 2025-01-12 by rohmer

kisvegabor, 2025-01-13:
I hope your feel better now. :slightly_frowning_face: