lvgl/lvgl #7501: Docs: Some examples not rendering because web version is not compiled with optional flags

timonsku, 2024-12-22:
### LVGL version

v9.2.0

### Platform

Web

### What happened?

Some of the examples that use new optional features like complex gradient are not rendering in the documentation page as the release was not compiled with those optional features enabled.

### How to reproduce?

Visit https://docs.lvgl.io/9.2/examples.html#radial-gradient-as-background a warning is shown instead of rendering the examples as expected.

timonsku, 2025-01-06:
still relevant

kisvegabor, 2025-01-15:
Our docs page is down for some unknown reasons, so I can't check it now. 😟

IIRC correctly I'v already fixed in in master.

timonsku, 2025-01-15:
Seems to be back online, still showing the warning that the feature is not enabled

timonsku, 2025-01-15:
DRAW_TRANSFORM_USE_MATRIX is still disabled for Master too though
https://docs.lvgl.io/master/examples.html#transform-object-using-a-3x3-matrix

timonsku, 2025-02-02:
still relevant

liamHowatt, 2025-02-07:
I think we can enable some of these. `USE_DRAW_SW_COMPLEX_GRADIENTS` should be completely fine to enable, I think. Any others?

I have one concern. `TRANSFORM_USE_MATRIX` is kind of experimental. If something indirectly causes matrix-based transformation to regress, all examples that use transformations will break. The examples in the docs all come from the same Emscripten object so all examples share the same configurations.

closed 2025-03-01 by lvgl-bot