lvgl/lvgl #7503: control the sliding step

lsj-fire, 2024-12-23:
### Introduce the problem

Hi author, I have a development question for you, how to control the sliding step of the list control?

kisvegabor, 2025-01-09:
What do you exactly mean by "sliding step"?

Natural-Fire, 2025-01-16:
> What do you exactly mean by "sliding step"?

Meaning how do I control the movement step of my control, moving up, down, left, right, like sliding a list control

Natural-Fire, 2025-01-16:
[image]

For example, moving from Track 1 to Track 3

Natural-Fire, 2025-01-16:
> What do you exactly mean by "sliding step"?
[image]

For example, moving from Track 1 to Track 3

liamHowatt, 2025-02-07:
What kind of input device are you using?

closed 2025-03-10 by lvgl-bot