lvgl/lvgl #7507: Shaped Window with Transparent Background (PNG)

symfund, 2024-12-24:
### LVGL version

9.3.0

### Platform

Wayland on Linux with Weston 14.0.1

[image]

[image]

### What happened?

The transparent background is not rendered as expected.

symfund, 2024-12-25:
[image]

As shown in CASE 2, the transparent background is not rendered correctly.

liamHowatt, 2025-01-08:
Interesting. It wasn't fixed by #7269. Case 3 also has a problem for me.

kisvegabor, 2025-01-09:
In Case 2, does `lv_display` has `LV_COLOR_FORMAT_ARGB8888`? See [here](https://docs.lvgl.io/master/details/base-widget/obj.html#transparent-screens).

symfund, 2025-01-10:
#7543 

[image]

@kisvegabor In case 2, the lv_display does have LV_COLOR_FORMAT_ARGB8888.

demo runs as
$ LV_WAYLAND_DISABLE_WINDOWDECORATION=1 LV_SIM_WINDOW_HEIGHT=120 LV_SIM_WINDOW_WIDTH=160 /usr/local/bin/lvglsim &

symfund, 2025-01-10:
A temporary solution
In function lv_wayland_window_create(), after creating the LVGL display driver via lv_display_create(), set display color format to LV_COLOR_FORMAT_NATIVE_WITH_ALPHA, and set display render mode to LV_DISPLAY_RENDER_MODE_DIRECT.

[image]

[image]

symfund, 2025-01-10:
@kisvegabor Thanks for your awesome LVGL, I can now implement the Analog Clock with LVGL 9.3.0 on Wayland.

kisvegabor, 2025-01-13:
Thank you for the feedback! 

I think we should start with the XRGB8888 color format by default because it's faster, but an `LV_EVENT_COLOR_FORMAT_CHANGED` event handler should be added to the Wayland to detect `lv_display_set_color_format` calls.  

cc @etag4048.

closed 2025-02-06 by lvgl-bot