lvgl/lvgl #7520: VGLite rendering on K230 is broken

MrThanlon, 2024-12-26:
### Introduce the problem

Enable LV_USE_DRAW_VG_LITE
[image]
Disable LV_USE_DRAW_VG_LITE
[image]

Any suggestions for debugging?

uLipe, 2024-12-26:
@MrThanlon could you check / enable the logging system on the LVGL to check if there are some warning? When I faced problems like this before, the VGLite draw module started to output messages on the log indicating there are some missing configuration parts on the driver side.

MrThanlon, 2024-12-27:
> @MrThanlon could you check / enable the logging system on the LVGL to check if there are some warning? When I faced problems like this before, the VGLite draw module started to output messages on the log indicating there are some missing configuration parts on the driver side.

I have enable all log, no error log, only VGLite USER log.
BTW, I test the benchmark demo, in the 6th scene "Rotated ARGB images", there should be two logo, but VGLite only get one logo and flash back, I debug the program and breakpoint at draw_execute function, it called by 4 times in order
- LV_DRAW_TASK_TYPE_FILL full screen white
- LV_DRAW_TASK_TYPE_IMAGE the logo
- LV_DRAW_TASK_TYPE_FILL full screen white (?)
- LV_DRAW_TASK_TYPE_IMAGE the logo
the result is only one logo in the screen, why the second LV_DRAW_TASK_TYPE_FILL happen?

MrThanlon, 2024-12-27:
It seems the K230 VGLite doesn't support path bounding_box, any idea to fix it?

closed 2025-01-03 by MrThanlon

kisvegabor, 2025-01-03:
Have you found a solution?