lvgl/lvgl #7521: Improved the efficiency of text wrapping

Neothai, 2024-12-26:
### Introduce the problem

### LVGL Version
v9.3

### Problem

Hello everyone!

First of all, thanks to LVGL, I can easily create beautiful UI pages on MCU.

Now I am developing a project using LVGL to design a UI page, which requires displaying a large number of texts, so I need to wrap the texts in the limited space. The problem now is that the text wrapping is in both "Thai and English", and the results of wrapping text in Thai are not very satisfactory. Now I classify the text wrapping styles into 3 styles:

- Wrapping text to fit the width of the object specified by lv_obj_set_width(), regardless of whether the last text that is cut to a new line is complete or meaningful.
- Wrapping text by checking from a group of words that are separated by spaces between the groups of words.
- Wrapping text by considering the correct size and grammar. The first order is to consider the size of the text that should be wrapped to fit the size of the object. The next order is to consider the meaning of the last word that will be cut. If the last word of each line is cut between words, those words must be put on the entire line. For example,

Example sentence: "สวัสดีทุกๆคน ยินดีที่ได้รู้จักทุกท่านในวันนี้"
Number of characters per line: 15

How to count the number of characters:
```
"ส" + "วั" + "ส" + "ดี" + "ทุ" + "ก" + "ๆ" + "ค" + "น" + " " + "ยิ" + "น" + "ดี" + "ที่" + "ไ" + "ด้" + "รู้" + "จั" + "ก" + "ทุ" + "ก" + "ท่" + "า" + "น" + "ใ" + "น" + "วั" + "น" + "นี้"
Total 29 characters
```

Incorrect text wrapping: (Because the word "ได้" cannot be separated)

```
สวัสดีทุกๆคน ยินดีที่ไ
ด้รู้จักทุกท่านในวันนี้
```
#

Correct text wrapping:

```
สวัสดีทุกๆคน ยินดีที่
ได้รู้จักทุกท่านในวันนี้
```
#

## Here is an image to illustrate the description:

[image]

xlbao123, 2024-12-27:
A complete implementation of the [Unicode TR14 algorithm](https://www.unicode.org/reports/tr14/) is required to solve all such issues.

Neothai, 2024-12-29:
From what I've researched, it seems that the best way to consider line wrapping is to extract "each word" from a full sentence. We may need a dictionary of words to analyze and extract each word from a sentence.

Here are some examples of line wrapping in different software:

## Chrome Browser:

https://github.com/user-attachments/assets/48f6297b-941b-4359-b704-66906f9c40e4

## Microsoft PowerPoint:

https://github.com/user-attachments/assets/68a972dc-c8bf-4412-a284-50d5e767f919

Neothai, 2025-01-15:
Still need everyone's attention

closed 2025-03-10 by lvgl-bot