lvgl/lvgl #7525: static function declarations do not get exported

kdschlosser, 2024-12-28:
### LVGL version

v9.2.2

### Platform

Windows 11
cmake
ninja
MSVC

### What happened?

@liamHowatt 
@kisvegabor 

Functions that are declared as static and are supposed to be apart of the public API do not get exported when compiling LVGL into a shared library. 

Some examples are in `src/core/lv_obj_style.h`

My understanding of the purpose of static functions could be flawed and if it is please correct me. a function is made static if the function is only supposed to be used "locally". When I say locally I mean by the projects code. Static function declaration is more commonly used in C source files so there are no name collisions between source files and any functions that are declared in a source file as static has a zero possibility of being used in some other source file. Basically any function that is declared/defined in a C source file and is ONLY used in that very same source file should be made static. If for some reason a private header file has function declarations in it and that header file is included in public header files then the functions that are declared in the private header files should be made static. This is so they don't get exported when compiling a shared library. private header files really should not be included in any public header files. They should only be included in other private header files or in source files. 

So public API functions should not be made static and all private functions should be made static. That is what my understanding is.

As a note, functions should not be defined in header files, only declared. the definitions should be in a source file. This keeps things a lot cleaner as far as documentation is concerned. it separate the documentation from the source code so things don't turn into a jumble that makes it hard to read.

In order for me to circumvent the problem I have to create wrapper functions that are not static so they get exported for use. a rather frustrating thing to have to do considering there are 292 functions in the LVGL API that are marked as static. 

@vwheeler63 would probably know what is the "industry standard" when it comes to this kind of a thing. If I am wrong in my understanding of the use of static functions please let me know.

### How to reproduce?

Here is a list of functions that are not marked as static and should be. This is not an all inclusive list as every option in LVGL is not turned on. I am sure there are functions in the disabled options that should also be made static as well.

<details>

<summary>Click me to see the list</summary>

```
LODEPNG_VERSION_STRING
code128_encode_gs1
code128_encode_raw
code128_estimate_len
gd_close_gif
gd_get_frame
gd_open_gif_data
gd_open_gif_file
gd_render_frame
gd_rewind
jd_decomp
jd_mcu_load
jd_mcu_output
jd_prepare
jd_restart
load_kern
lodepng_add_itext
lodepng_add_text
lodepng_can_have_alpha
lodepng_chunk_ancillary
lodepng_chunk_append
lodepng_chunk_check_crc
lodepng_chunk_create
lodepng_chunk_data
lodepng_chunk_data_const
lodepng_chunk_find
lodepng_chunk_find_const
lodepng_chunk_generate_crc
lodepng_chunk_length
lodepng_chunk_next
lodepng_chunk_next_const
[... 416 lines cut ...]
lv_obj_get_style_x
lv_obj_get_style_y
lv_obj_get_style_align
lv_obj_get_style_transform_width
lv_obj_get_style_transform_height
lv_obj_get_style_translate_x
lv_obj_get_style_translate_y
lv_obj_get_style_transform_scale_x
lv_obj_get_style_transform_scale_y
lv_obj_get_style_transform_rotation
lv_obj_get_style_transform_pivot_x
lv_obj_get_style_transform_pivot_y
lv_obj_get_style_transform_skew_x
lv_obj_get_style_transform_skew_y
lv_obj_get_style_pad_top
lv_obj_get_style_pad_bottom
lv_obj_get_style_pad_left
lv_obj_get_style_pad_right
lv_obj_get_style_pad_row
lv_obj_get_style_pad_column

etc....
```

</details>

kdschlosser, 2024-12-28:
https://stackoverflow.com/questions/2984316/dll-export-of-a-static-function

vwheeler63, 2024-12-28:
Hi, Kevin!  ( @kdschlosser )

Before I get into many paragraphs, after looking over the above carefully, I'm not sure what problem you are wanting to address.  Is it something missing in the generated API documentation?

kdschlosser, 2024-12-28:
just because a function in inlined doesn't mean it always gets inlined. simply changing the compiler optimization alters this behavior.

But at any rate the solution would be to create macros that would handle it, Is there a need for the function to be static? I don't think there is a need to have the function declared as static when using inline. 

https://stackoverflow.com/questions/7762731/whats-the-difference-between-static-and-static-inline-function

It's the static that is causing the issue not the inline and it can/should be removed as there is no issue with the same function existing in other translation units because it doesn't.

kdschlosser, 2024-12-28:
The LVGL simulator is not linked against a shared library. It is linking against a static one.

vwheeler63, 2024-12-28:
> The LVGL simulator is not linked against a shared library. It is linking against a static one.

I know -- I was using it to compare.

kdschlosser, 2024-12-28:
There has to be a compiler switch or a linker switch that controls this behavior.

kdschlosser, 2024-12-29:
what I also don't know is if the static functions get exported properly when compiling under Linux. If they do get exported then we can use a simple 

```c

#if defined(MSVC) && defined(DLL)
#define STATIC_INLINE
#else
#define STATIC_INLINE     static inline
#endif
```

and it would get used like so...

```c
STATIC_INLINE void lv_obj_set_style_margin_all(lv_obj_t * obj, int32_t value, lv_style_selector_t selector)
{
    lv_obj_set_style_margin_left(obj, value, selector);
    lv_obj_set_style_margin_right(obj, value, selector);
    lv_obj_set_style_margin_top(obj, value, selector);
    lv_obj_set_style_margin_bottom(obj, value, selector);
}
```

when compiling as a shared library using MSVC then `-DDLL` would need to be added to the compiler arguments which is pretty easily done..

vwheeler63, 2024-12-29:
> If they do get exported then we can use a simple

I like it!  I'm anxious to see what @kisvegabor and @liamHowatt think once they get back after the 6th....

kdschlosser, 2024-12-29:
it doesn't work with MSVC...

The reason why it doesn't work is because it's a definition and not a declaration. so when the header file gets imported into a C source file and then again into another C source file there are now 2 function definitions that are the same.

I had to move the definitions into a C source file and then make a declaration in the header file. Removing the `static inline` of course. That is the only way I was able to get it to compile and have the functions properly exported.

kdschlosser, 2024-12-29:
It's going to become a coding mess in order to make it work right because there will need to be a double set of code for the same functions, ones that are static inline and defined in the header file but guarded with a macro so it known to use those when compiling without making a shared library. The second will be defined in a c source file without the `static inline` and once again guarded with a macro for a shared library compilation.

The big ? is how does this all play out compiling on posix using clang or gcc or is this just an issue that is specific to MSVC. I am thinking it is going to be an MSVC specific bump in the road.

I even tried to force it by turning off inline functions completely using the /Ob0 switch and I also tried adding the functions to an export definition file. Neither worked.

vwheeler63, 2024-12-29:
> it doesn't work with MSVC...

Try
```c
#if defined(_MSC_VER)
```

vwheeler63, 2024-12-29:
> It's going to become a coding mess

Duplicate code definitely does *not* sound like a good idea.

kdschlosser, 2024-12-29:
no it doesn't work the functions do not get exported, and why would they? The function is supposed to be inlined which an oversimplification of it is it acts kind of like a macro does where it gets inserted into the place it is being called. 

With what I am doing I cannot link as a static library. It needs to be a shared library because it is being compiled all by itself with no other C code accessing it.

There is a binding that I wrote for LVGL that allows LVGL to be used in CPython on a desktop. To do that I am using the ctypes module in CPython as the binding layer to access the shared library. That makes it cross platform so it will work on anything that is able to run CPython so long as I can compile LVGL as a shared library. There is no way to access a static library from inside of CPython unless I write the binding code in C. Doing that would be a large task. It could be done using swig or cython but it would be a very large undertaking.

vwheeler63, 2024-12-30:
> ...it would be a very large undertaking.

Probably even better to be generated code.  

That said, if the static library is to be usable at all, the entire API must be available.  I guess it's over to @kisvegabor to determine how to remedy.  I can think of 2 workable solutions off the top of my head:

1. turn them into normal functions, or
2. turn them into macros.

I am thinking `#2` won't work for a Python binding, am I right?

kdschlosser, 2024-12-30:
>I am thinking #2 won't work for a Python binding, am I right?

no it will not. macros replace code.

I did however come up with an idea. Leave the inline static functions how they are just rename then to something else. Maybe prefixing them with an underscore. do a check to see if the compilation is for a shared library running MSVC using a macro that can be passed to the compiler at the command line. If the macro exists then wrapper functions that are not static inline get made that will make the needed calls to the static inline functions. and if that macro doesn't exist then macros gets created instead that point to the underscore prefixed functions.

That will be the cleanest way of doing it allowing everything to work like it should.

I am going to code that up. Going to need your assistance with it to handle the documentation end of things. It's going to be tricky to get it to work correctly so the documentation displays how it does right now. Maybe not use macros that point to the static inline functions maybe create static inline function wrappers only when the documentation is being compiled.

Gonna have to dink about with it so see what the best way is to handle that.

kdschlosser, 2024-12-30:
Macros are not available to any binding other than the C++ binding. That is because the C++ binding LVGL is able to be compiled with the binding and not as a shared library... The MicroPython binding is not able to use macros because the binding code gets generated from the preprocessor output which means all of the macros have been expanded so they are not visible to the code generator.

The question is can this be done...

```

static inline int32_t _obj_get_style_min_height(const lv_obj_t * obj, lv_part_t part)
{
    lv_style_value_t v = lv_obj_get_style_prop(obj, part, LV_STYLE_MIN_HEIGHT);
    return (int32_t)v.num;
}

#ifdef MSVC_SHARED_LIB
int32_t lv_obj_get_style_min_height(const lv_obj_t * obj, lv_part_t part) return _obj_get_style_min_height(obj, part);
#else
static inline int32_t lv_obj_get_style_min_height(const lv_obj_t * obj, lv_part_t part) return _obj_get_style_min_height(obj, part);
#endif
```

can you inline an inlined function like that?

kdschlosser, 2024-12-30:
I just remembered there is no documentation for these functions because they are generated using a script. you can instruct Doxygen to ignore the `_obj_get_style_min_height` function without any issue and you can change the code to read

```c
#if defined(MSVC_SHARED_LIB) && !defined(DOXYGEN)
int32_t lv_obj_get_style_min_height(const lv_obj_t * obj, lv_part_t part) return _obj_get_style_min_height(obj, part);
#else
static inline int32_t lv_obj_get_style_min_height(const lv_obj_t * obj, lv_part_t part) return _obj_get_style_min_height(obj, part);
#endif
```

so when doing a documentation build on Windows the documentation will capture the static inline portion of the function when generating the API documentation.

kdschlosser, 2024-12-30:
I just don't know how the compiler will handle the nested inline function calls. Will it optimize it correctly without making the compiled binary larger... That is something I do not know...

kisvegabor, 2025-01-07:
A few notes on `static inline` functions: 
They are `inline` to indicate they should be inlined (it's up to the compiler to really inline them).  In the past we have found that most compiler proof way of having inline functions in having them in the same C file where you want to inline them. To do this with generic functions used in multiple places we need to place the inline functions to the header files. But if a header file in included in 2 or more files the function will the defined multiple times in each C file, resulting in compiler error. That's why we need `static` too, to make the function local to the C file. 

I prefer inline functions over macros because they are easier debug, read, and they provide type checking for the parameters. 

There is no better option to workaround MSVC we can apply 
```c
#if defined(MSVC) && defined(DLL)
#define LV_STATIC_INLINE   /*Use and LV_ prefix*/
#else
#define LV_STATIC_INLINE     static inline
#endif
```
with a search and replace. `static inline` -> `LV_STATIC_INLINE` But I hope we don't need to introduce it just for MSVC.

kdschlosser, 2025-01-07:
GCC doesn't export static inline functions either. Compiling LVGL as a shared library would not work properly there either. I am pretty sure it's the same thing with clang as well. I mean think about it. static means it remains local and if it's static then it cannot be exported.

The only way to handle this is going to have a marker set if a shared libraruy is what is being compiled. This is pretty easily done by passing something like `-DLV_SHARED_LIB` to the compiler when building. and then in the code for LVGL you would have...

```c
#if LV_SHARED_LIB
#define LV_STATIC_INLINE
#else
#define LV_STATIC_INLINE  static inline
#endif
```

Just to let you know there are also things declared as `inline static` for some crazy reason.  IDK if there is a difference between the 2...

What you might consider going since most of the static inline functions are styles and are generated using a script. You could add onto that script so it generates a private header file with the static inline function with the names altered slightly. do a simple search of the LVGL files to see where the original function names are used and add an include to the new private header and replace the used function names with the new private ones. This way you would still get the bump in speed from the inline functions. I really don't want to see a loss in performance. LVGL is able to run on MCU's that are able to run Linux. Like the Omega Onion or something running OpenWRT. The inline functions would help performance wise especially on those kinds of devices.

kdschlosser, 2025-01-07:
Here is a thought....

```
static inline int32_t lv_obj_get_style_min_height_inline(const lv_obj_t * obj, lv_part_t part)
{
    lv_style_value_t v = lv_obj_get_style_prop(obj, part, LV_STYLE_MIN_HEIGHT);
    return (int32_t)v.num;
}

int32_t lv_obj_get_style_min_height(const lv_obj_t * obj, lv_part_t part) return lv_obj_get_style_min_height_inline(obj, part);
```

function calles made internally in lvgl will call the functions that end with _inline. and the user can use either depending on how they are linking to LVGL. if they are linking LVGL as a static lib then they can use the function that has the ` _inline` suffix, shared lib they use the function without the suffix..

What do you think???

kisvegabor, 2025-01-16:
We shouldn't have 2 different APIs depending on LVGL used a s shared library or "normally".

I don't understand something. If we have  shared library without the `static inline` functions, and we have the LVGL headers with the `static inline` functions what is the problem? The library is already precompiled, but when you include `lvgl.h` you can also use the `static inline` functions and they can be really inlined when you compile your app. If these were part of the lib how could the compile inline them? 

So it seems to me they shouldn't be exported, and it's correct.

kdschlosser, 2025-01-16:
You do not need to include a header file in order to access the functions inside of a shared library.

kisvegabor, 2025-01-30:
It's not correct. You always need to include the header files in order to use any functions not declared in the given C file. Otherwise you will get `implicit declaration of function` warning. To access enums, types, defines, global variables, etc including the realted headers is a must.

kdschlosser, 2025-01-30:
Ummm... yes it is correct.. You are thinking about it from the standpoint of a compile time piece of software.

Here is an example...

```python

import ctypes
import time

# Load the C library
libc = ctypes.CDLL("libc.so.6")

# Define the function prototype for clock_gettime
CLOCK_MONOTONIC_RAW = 4  # Use a monotonic clock that is not subject to NTP adjustments
class timespec(ctypes.Structure):
    _fields_ = [("tv_sec", ctypes.c_long), ("tv_nsec", ctypes.c_long)]

libc.clock_gettime.argtypes = [ctypes.c_int, ctypes.POINTER(timespec)]
libc.clock_gettime.restype = ctypes.c_int

def get_time():
    ts = timespec()
    libc.clock_gettime(CLOCK_MONOTONIC_RAW, ctypes.byref(ts))
    return ts.tv_sec + ts.tv_nsec * 1e-9

# Example usage
start = get_time()
# Your code here
end = get_time()

print("Elapsed time:", end - start)
```

That code is accessing the `clock_gettime` function in the libc shared library without any header file. It is Python code. High level programming languages can use FFI to access what is located inside of a shared library without ever needing to use a header file. This is because the function prototypes are all encoded into the shared library. No header file is needed. The return value and parameters are always of an integer type. whether it be an actual integer or a pointer which after all is still an integer that points to a memory location. in C typing only serves the purpose of making it easier for people to understand when writing the code.

example...

c```

typedef struct {
   int field;
} some_struct;

void some_function(some_struct *arg) 
{
    arg->field = 123456;
}

some_struct some_variable = {
    .field=0
};

some_function(&some_variable);

```

This statement would evaluate correctly.
`some_variable->field == 123456 == true`

c```
typedef struct {
   int field;
} some_struct;

void some_function(some_struct *arg) 
{
    arg->field = 123456;
}

uint8_t some_array[4] = { 0 };

some_function((some_struct *)some_array);
```

This would evaluate properly as well.
`(int)some_array == 123456 == true`

The type cast only serves a single purpose and that is to deal with the size properly. The only thing the actual compiled binary cares about when dealing with variables and arguments passed to a function is a location in memory if dealing with pointers and the size the actual type it doesn't care a hill of beans about. In the above Python example if I change the code to this..

```python

import ctypes
import time

# Load the C library
libc = ctypes.CDLL("libc.so.6")

# Define the function prototype for clock_gettime
CLOCK_MONOTONIC_RAW = 4  # Use a monotonic clock that is not subject to NTP adjustments
class timespec(ctypes.Structure):
    _fields_ = [("tv_sec", ctypes.c_long), ("tv_nsec", ctypes.c_long)]

libc.clock_gettime.argtypes = [ctypes.c_long, ctypes.POINTER(timespec)]
libc.clock_gettime.restype = ctypes.c_long

def get_time():
    ts = timespec()
    libc.clock_gettime(CLOCK_MONOTONIC_RAW, ctypes.byref(ts))
    return ts.tv_sec + ts.tv_nsec * 1e-9

# Example usage
start = get_time()
# Your code here
end = get_time()

print("Elapsed time:", end - start)
```

Where I changed the type passed as an argument and also changed the type of the return value both of them to `long` the code runs without any issue all exactly as it did when the type was set to `int` No difference. It's all about the memory locations and the size. There is no care in the world as to the "type" that is being used. as long as the size matches there will be no error. Now... If I change the type to an unsigned int it will run without any error but the results will be wrong if the value is < 0.

Remember when I was going on about placing structures into the private scope when they are used in public functions?? The reason why is the inability to determine the size of the structure. By removing that one single piece of the puzzle it makes it impossible to create a representation of that structure to use for the purposes of passing to a function inside of a shared library. The size MUST be known otherwise it will fail with memory access errors.

This code will also work

```python
import ctypes

# Load the C library
libc = ctypes.CDLL("libc.so.6")

# Define the function prototype for clock_gettime
CLOCK_MONOTONIC_RAW = 4  # Use a monotonic clock that is not subject to NTP adjustments
libc.clock_gettime.restype = ctypes.c_int

def get_time():
    array = (ctypes.c_uint8 * 8)()
    libc.clock_gettime(ctypes.c_int(CLOCK_MONOTONIC_RAW), array)

    # idk if I have the array indexing correct but ti serves the point to show 
    # that data types are not cared about
    tv_sec = array[3] << 24 | array[2] << 16 | array[1] << 8 | array[0]
    tv_nsec = array[7] << 24 | array[6] << 16 | array[5] << 8 | array[4]
    
    return tv_sec + tv_nsec * 1e-9

# Example usage
start = get_time()
# Your code here
end = get_time()

print("Elapsed time:", end - start)
```

as you can see I completely did away with the type `timespec` all together. compiled C code doesn't care about data types, it only cares about size and memory location. The types are only there to make it easier for people to understand not for the machine to. 

My point is the header file is completely irrelevant so long as the size is known when accessing a shared library using something like FFI. FFI is widely used in many different programming languages to be able to access compiled binaries written in C..

Here are some examples... Python, Ruby, Perl, JavaScript, Java, C#, F#, Lua, PHP, Groovy, Lisp, Objective-C, PowerShell, etc..... You do not need a header file in order to access the shared library from those languages. But the functions DO need to be exported so they are accessible.

kdschlosser, 2025-01-30:
The issue is you are thinking that a compiled shared library is ONLY going to be accessed from some other program that is written in C. Break away from that line of thought because there is a world of other programming languages out there that are able to access the contents of a shared library and no header file is used in order to do it.

Now seeing as how the functions are very simple functions they can be written to perform the same task in whatever programming language is being used to access the shared library. And that would be OK if there were 1, 2 maybe 10 of those kinds of functions.. The issue is there are a whole lot more than 10 of those kinds of functions in LVGL that would have to be written by hand. That's quite a large amount of extra work that would need to be done to write a binding for LVGL if the binding is being written in a higher level programming language.

kdschlosser, 2025-01-30:
One of the biggest issues with using inline functions is they behave much like a preprocessor macro does. The code inside of the function gets placed where the call is being made from. As an example if the same function is not written as inline and the function takes up say 10 bytes in memory if there are 200 different places in the program that call that function only 10 bytes of memory is going to be used. Where as if the function is inline you end up with 200 * 10 = 2000 bytes used instead of 10. Having a macro definition to control how those functions are declared is not a bad thing to have because it lets the user decide on what they want to do based on the resources that are available. Changing the compiler optimization is a global change. You cannot really target specific things in the code. That is what the macro would do. There is a macro that allows for a similar type of thing with the ESP32 and placing a function into IRAM. It's really no different.

kisvegabor, 2025-02-01:
Okay, I see! I've just checked that the non-style related `static inline` functions are either not exposed via `lvgl.h` or very specific and shouldn't be used in bindings.

It means we need to focus on styles only, which means we can modify `style_api_gen.py` to export these function as `static inline` and normal too, and enable the one based on a define like `LV_STYLE_API_INLINED`. When compiling an `so` file, this can be disabled, and enabled by default. 

BTW, will you use FFI in the Micropython binding?

kdschlosser, 2025-02-01:
When dealing with MicroPython external libraries are not dynamically linked to. This is because the library needs to be compiled specifically to run on the MCU. So the library must be statically linked to and that compiled the libraries code into the binary that gets written to the MCU..

This is where you really need to put some thought into how you want to deal with these functions. You are dealing with something that has 2 edges to it the first edge is public availability which as of right now is a problem. But if we correct that problem it creates an issue on the other edge which is the use of the functions internally. You loose the speed gains associated with using inline by making the functions publicly available. The easiest way to deal with this is to keep the existing inline functions but possibly moving them into the private scope and removing the `static inline` from the functions that are in the public scope. This would be pretty easy to handle I believe because most or even possibly all of the functions in question are generated using a script. The same script could be used to also check the C source file to replace the function names that are public names to the private ones.

I think the above is a bit on the extreme but still worth mentioning. However, I don't believe there is going to be any appreciable or noticeable speed difference if the functions are not `static inline`. I say this because the use case would be compiling LVGL to run on a PC is pretty much the only scenario where you would compile a shared library of LVGL. The only scenario where it could possibly make a difference is running on a credit card computer like a Raspberry PI because a PI sits in between a PC and an MCU in terms of performance. I do not believe there is enough people running LVGL on PI where LVGL is compiled as a shared library to even have to worry about it too much....

I feel that updating the script that generates the code for the styles so `static inline` gets replaced with a macro like you have suggested and that macro is able to be set from the command line...

I would have this placed at the top of the generated header files for the style functions.

```
#ifndef LV_STYLE_API_INLINE
#define LV_STYLE_API_INLINE   static inline
#endif
```

This way it can be set as empty by passing `"-DLV_STYLE_API_INLINE="` to the compiler. 

I don't think the name of the macro really needs the "D" on the end. It translates better when reading the function code.

could even use `LV_INLINE_STYLE`... 

I like `LV_INLINE_API` better because if there is a need in the future to do the same thing possibly to other functions that are not style functions you are not going to end up breaking API by removing the `STYLE` from the macro name. 

I forgot to mention or maybe I did already and we forgot about it.. `static inline` is in use as well as `inline static`. Have to check to see if `inline static` functions are in the public scope as well. Those should be corrected to `static inline` if they are in LVGL code to maintain consistency.

kisvegabor, 2025-02-06:
To make sure that we are on the same page, could you show and example for a single functions (e.g. `lv_obj_get_style_bg_color`) in each files with macros, `#if`s etc?

kdschlosser, 2025-02-06:
I think that simply doing this...

```c
#ifndef LV_INLINE_API 
#define LV_INLINE_API  static inline
#endif

LV_INLINE_API lv_color_t lv_obj_get_style_bg_color(...);
```

Would be more than sufficient for this.

so in the cmake script you add a user variable for compiling a dynamic library that can be set from the command line..

'''cmake
option(LV_DYNAMIC_LIBRARY
       "Use this parameter if compiling LVGL as a dynamic library" OFF)

if(LV_DYNAMIC_LIBRARY)
    target_compile_options(lvgl PUBLIC "\"-DLV_INLINE_API=\""
endif(LV_DYNAMIC_LIBRARY)

'''

So now a user can kind of have a dynamic library build from the command line by using the option. I say kind of because it cannot be done without editing the cmake files and adding the compiler/linker arguments to actually compile as a dynamic library. It would be nice if the options were added by the cmake script if that option was provided, at least for the 3 major compilers... CLANG, GCC and MSVC. Then nothing special would need to be done except for running cmake with that option set so the build gets set up properly and then a simple call to Ninja to compile or whatever build handler is wanting to be used and bank you now have a dynamic library that is able to be used by some other program or even a high level programming language using FFI

kdschlosser, 2025-02-06:
@liamHowatt 's input on how to go about doing this might be helpful.

kisvegabor, 2025-02-07:
```c
LV_INLINE_API lv_color_t lv_obj_get_style_bg_color(...);
```

It's not working if `LV_INLINE_API` is empty because this way `lv_obj_get_style_bg_color` will be defined 20+ times, each time the header is included.

kdschlosser, 2025-02-07:
OH that's right. It's because the function is defined in the header file and not in the source file....

I forgot about that...

Out of question, could a function be declared in a header file as static inline but the definition for the function is in the source file?

kdschlosser, 2025-02-10:
OK that makes sense. Then it is going to be more complicated to make it work properly and the code is going to be long and ugly unfortunately.

kdschlosser, 2025-02-11:
another way of handling it and it is able to be done all in the same file is like this...

```c

#ifndef LV_INLINE_API
#define LV_INLINE_API 1
#endif

static inline int32_t _lv_obj_get_style_width(const lv_obj_t * obj, lv_part_t part)
{
    lv_style_value_t v = lv_obj_get_style_prop(obj, part, LV_STYLE_WIDTH);
    return (int32_t)v.num;
}

// force doxygen to use the functions and that is where the documentation will be placed. 
#if LV_INLINE_API && !defined(DOXYGEN)
    #define lv_obj_get_style_width  _lv_obj_get_style_width
#else 
    int32_t lv_obj_get_style_width(const lv_obj_t * obj, lv_part_t part) { return _lv_obj_get_style_width(obj, part); 
#endif

```

and that will keep everything in the same header file and it will allow the inline functions to work like they should if using the INLINE API. The definition check for `LV_INLINE_API` would not be in the style header. It would be placed in `lv_conf_internal.h` this way the same type of logic can be used if there are static inline functions elsewhere that need to be wrapped either now or in the future.

We can do some crafty with doxygen so it will not document the the `_lv_*` functions and we can force it to use the `lv_*` functions instead. Can even add the `static inline` if that is wanting to be done.

Here is an attached generator script. I reformatted the thing so it is easier to to read. It was a pain to follow what was going on with the way it was done.

[scratch_1531.py.txt](https://github.com/user-attachments/files/18750724/scratch_1531.py.txt)

AndreCostaaa, 2025-02-15:
Just a question. Can't we just remove static inline functions from the public api?

I believe the speed difference is negligible and 90% of the time (in embedded) memory usage is more crucial than speed, having an inline public function could increase flash usage on the end-user's application _rapidly_ which could be confusing for the user

My proposal would be to move the current static inline function definitions to a .c file, remove the static inline attribute and keep the function signature inside the header file

kdschlosser, 2025-02-16:
you would end up having to change all of the code in LVGL so it would use the inline functions. The inline functions also need to be defined in a header file so they can be included. I know that a source file can also be included but that is not the right thing to do. This is a lot of work to do and the reason why this discussion is taking place. Trying to figure out the easiest way to go about correcting the problem without having to change large chunks of LVGL code. 

The best I am able to come up with is wrapping all of the inline functions in a macro and having a second set that are just declarations of the functions. Then create a source file once again using a macro to guard against compiling the definitions for the declarations that are in the header file. That's about as easy as it is going to get I believe. The problem is because the function is defined in the header file if you do away with the `static inline` you end up with naming collisions but in order to make the function inline it needs to be defined in a header but once again if the static is not added we end up with naming collisions. It's a damned of you do and damned if you don't scenario.

AndreCostaaa, 2025-02-16:
Not sure we understand eachother. I meant, giving up on static inline functions altogether. Just replace the current static inline functions with a normal function instead (with a signature in the header file and the declaration in a source file)

It's not that much work specially if most of these functions are generated using a script

kdschlosser, 2025-02-16:
Honestly we would need to run some tests to get real world impacts. I don't know what the cost is to call a function on an MCU. I don't know how many times a function would have to be called to make a meaningful impact. Inline functions have more than just the benefit of being faster. They also consume less stack memory. They do increase code size and the question becomes, by how much? Is that additional program size worth it for the performance gains gotten from the inline functions. We need to get LVGL running an on MCU noting the compiled binary size. Get a measurement in CPU ticks of how long it takes for the benchmark demo to run. This will get a nice snapshot because those inline functions we be called a very large number of times.

Then remove the inline from the functions and compile. See what the binary size difference is. Then tun the test again and see what the difference is in how fast the benchmark demo is able to complete.

Personally I do not believe that having the functions as inline is going to make any large impact in the performance. I do not believe that the tradeoff of increased program size to the performance gains makes it worth doing I do not believe that the functions are being called anywhere near as often as is thought. The reality is in most cases they are called when the UI first starts up and that's about it. Some of them may get called more often especially when using the animations but the majority of them I don't believe get used a whole lot.

kisvegabor, 2025-02-17:
Can someone run a quick benchmark to see the memory usage change and speed difference (using `lv_demo_benchmark`). 

If really there is no big difference  I also agree with removing `inline`s.

kdschlosser, 2025-02-17:
Not so simple as I am finding out. as it turns out running the benchmark is not a blocking call. It uses the timers in lvgl to change the different tests. Cannot perform a timing test using it because the end result will always end up being the same.

kisvegabor, 2025-03-04:
> Does the timer callback only get called one time or does it get called 10 times? If it gets called 10 times then it is not actually running once every millisecond and if it only gets called a single time then once again it is not getting called once every millisecond. How on earth is that actually being managed?

Will be called only once. Just like a low priority time interrupt. If a high priority interrupt runs for 100 ms, the low priority timer will be triggered only once. 

---

If call `lv_timer_handler` in every milliseconds 2 things can happen:
1. Rendering is fast, there will be some immediately returning calls  to `lv_timer_handler`. No questions here.
2. Rendering is slow and let's say 3 frames are skipped. Rendering time will be measured correctly as it's just the sum of times spent with rendring. FPS will be correct too as it's just count of hoe many time LVGL could render. Finally after a long run `lv_timer_handler` will run again immediately, as the period time already expired.

kdschlosser, 2025-03-04:
There is no easy way for me to step into the way it is currently being done to get an accurate measurement of how long it takes for the demo to run and the problem is how often lv_task_handler gets called is going to directly effect the results. The demo would need to have been written not using lv_timers and not rely on lv_task_handler being called by user code in order to be able to measure it properly.

When LVGL calculates the frame counts is it counting each time a frame buffer is flushed to the display as a "frame" or is it calculating x number of flushes between `lv_display_flush_is_last` calls returning True as a single frame. The latter would be the correct way to do it because a frame is an entire update not just a chunk of an update.

I am not able to set up a proper test using the benchmark demo without having to write it over again. We need to get nano second time resolution in order to get a measurement of what the inline performance impact is with and without it. and in order to do that the benchmark demo needs to be a blocking call. That would mean the demo would need to run without using timers. I am not sure how else to get an accurate measurement.

kisvegabor, 2025-03-06:
The benchmark demo is considered reliable in the a latest master. We can open up this topic but let"s discuss it in a seprate issue if there is a specific concern. 

If there is no change in the benchmark demo e.g. on ESP32, we can say that inline and non-inline  are the same. We can ignore nanosec differences.

closed 2025-04-02 by lvgl-bot