lvgl/lvgl #7527: enable CONFIG_LV_USE_FS_ARDUINO_ESP_LITTLEFS=y. BUILD ERROR!!!

ngowangvinh, 2024-12-30:
### LVGL version

v9.2.2

### Platform

ESP-IDF 5.3.2 (using VScode extension)
Arduino 3.1.0~1 as component
lvgl 9.2.2 as component

### What happened?

Error when build: ```
W:/ESP_IDF/project/test_lvgl_lovy_esp32_wrover/managed_components/lvgl__lvgl/src/libs/fsdrv/lv_fs_arduino_esp_littlefs.cpp:5:10: fatal error: LittleFS.h: No such file or directory
    5 | **#include "LittleFS.h"**
```

### How to reproduce?

ADD Arduino and lvgl as components in ESP-IDF VSCODE Extension project.
```
dependencies:
  idf:
    version: '>=4.1.0' 
  espressif/arduino-esp32: ^3.1.0~1
  lvgl/lvgl: ^9.2.2
```
set sdkconfig:
CONFIG_LV_FS_DEFAULT_DRIVE_LETTER=70
CONFIG_LV_USE_FS_ARDUINO_ESP_LITTLEFS=y
CONFIG_LV_FS_ARDUINO_ESP_LITTLEFS_LETTER=70
Build project -> error occur!

liamHowatt, 2025-01-08:
You need to add LittleFS to your project too, as it its source is not included in LVGL.

ngowangvinh, 2025-01-10:
I solved the problem by add `REQUIRES  arduino-esp32` (line 64) in to file esp.cmake in /env_support/cmake.

liamHowatt, 2025-01-14:
Nice. `arduino-esp32` has a dependency on LittleFS so it makes sense. https://github.com/espressif/arduino-esp32/blob/fcce93a02949c50ce7ab565884b933fb474b3437/idf_component.yml#L119

Since that cmake file is generic and not specifically Arduino, we probably shouldn't add that line in LVGL. Do you think it'd make sense to add something like `if(arduino) REQUIRES arduino-esp32`?

ngowangvinh, 2025-01-16:
I tried to use below code to conditioning register component. But can not sort out the problem.
```
 if(CONFIG_LV_USE_FS_ARDUINO_ESP_LITTLEFS OR CONFIG_LV_USE_FS_ARDUINO_SD)
      idf_component_register(SRCS ${SOURCES} ${EXAMPLE_SOURCES} ${DEMO_SOURCES}
          INCLUDE_DIRS ${LVGL_ROOT_DIR} ${LVGL_ROOT_DIR}/src ${LVGL_ROOT_DIR}/../
                        ${LVGL_ROOT_DIR}/examples ${LVGL_ROOT_DIR}/demos
          REQUIRES esp_timer arduino-esp32) 
  else()
      idf_component_register(SRCS ${SOURCES} ${EXAMPLE_SOURCES} ${DEMO_SOURCES}
            INCLUDE_DIRS ${LVGL_ROOT_DIR} ${LVGL_ROOT_DIR}/src ${LVGL_ROOT_DIR}/../
                          ${LVGL_ROOT_DIR}/examples ${LVGL_ROOT_DIR}/demos
            REQUIRES esp_timer) 
  endif()
```
I refer here: [https://github.com/espressif/esp-protocols/blob/542547d38bcd656aab3e8fd6f35dcfa5500dfcda/components/esp_websocket_client/CMakeLists.txt](url)

I just know a little about cmake, so I don't know how  to write a right code to sort out the problem.

liamHowatt, 2025-02-04:
Hmm, looks correct. What was the error?

closed 2025-02-28 by lvgl-bot