lvgl/lvgl #7530: lv_cache_entry_private.h is included in a public header somewhere.

kdschlosser, 2024-12-30:
### LVGL version

v9.2.2

### Platform

`lv_cache_entry_t` is being used in `lv_image_decoder.h` and the `lv_cache_entry_private.h` header file is not included anywhere in the `lv_image_decoder.h`. So there is someplace in come header files that including `lv_cache_entry_private.h` and is chaining allowing `lv_cache_entry_t` to be available in the `lv_image_decoder.h` file.

### What happened?

Nothing is broken as a result of this, 

I know that the cache stuff was supposed to have been moved to being private so I am thinking that something got overlooked.

vwheeler63, 2024-12-30:
@kdschlosser 

Tip (if you weren't already aware):  In Visual Studio, temporarily add `/showIncludes` as a compiler command line option to the project:  ProjectProperties > C/C++ > Command Line > Additional Options:  add `/showIncludes` to whatever is there.

Edit  `/lvgl/src/draw/lv_image_decoder.c`  and hit Ctrl-F7 to compile just that one file.  The Output window will display a tree structure (as a list) of the `.H` files that get included showing the sequence.

However, what I am seeing for `lv_image_decoder.c` is a partial definition that delays actual connection with the `struct _lv_cache_entry_t` until the linker takes over:

```
lv_image_decoder.c:  `#include "lv_image_decoder_private.h"`
      +-- #include "../misc/cache/lv_cache.h"
            +-- #include "lv_cache_lru_rb.h"
                  +-- #include "lv_cache_private.h"
                        +-- #include "../lv_types.h"
                        +-- struct _lv_cache_entry_t;     // <--- the work-around
// ...and from here onward, it is "legal" to use type `lv_cache_entry_t`
```

I suspect this connection was unintentional, whereas directly including `lv_cache_entry_private.h` from `lv_image_decoder.h` is probably the direct (clean) way of handling it (in my opinion).  (Though I was not part of the "private scheme" design, so I may be missing something.)

kdschlosser, 2024-12-30:
The thing is all of the "cache" code was supposed to have been moved out of the public scope. I am not sure if the use of `lv_cache_entry_t` in the `lv_image_decoder.h` file was an oversight, something that got missed and ended up not being moved into the private scope. The thing is because `lv_cache_entry_t` is declared/defined in a private header file but it is accessible from a public header file that means that EVERYTHING that is in the `lv_cache_private.h` file is actually not private. 

your tree for the include propagation shows exactly what I am saying...

```
            +-- #include "lv_cache_lru_rb.h"  <===== PUBLIC HEADER FILE
                  +-- #include "lv_cache_private.h" <====== PRIVATE HEADER FILE INCLUDED IN A PUBLIC HEADER FILE
```

kdschlosser, 2024-12-30:
What I am doing is actually going to map out exactly what includes need to exist in each of the public header files based on what is being used. There are things that are used in some of the header files where there is no inclusion made for where that component is located. It is working because of the way the inclusions are propagating. This makes it unclear how something is actually available. That is something that will be able to be corrected because of what I am going. Remove the inclusions that are being done when there is nothing that is being used from the included file and add the ones that things are being used from even if the only purpose ends up being code readability. The compiler will optimize all of that away. 

I think that it makes it easier from a maintenance standpoint. What are your thoughts on that?

vwheeler63, 2024-12-30:
> What are your thoughts on that?

On projects I control, it is firm policy to keep the include list clean.  If you weren't already using it, Visual Studio has coloring in its "Intelli-whatever-it's-called" that could be helpful.  Example from `lv_obj_style.h`:

[image]

and as you can see, it darkens `#include "../misc/lv_bidi.h"` which is a signal you can delete that include, since its contents are (according to the editor) not being used.  And indeed, deleting it and then compiling `lv_obj_style.c` confirms that it compiles successfully.  And `lv_obj_style.c` is showing something similar with `lv_anim_private.h`:

[image]

kdschlosser, 2024-12-30:
but if you are including lv_area.h in the lv_color.h file and you include lv_color.h in another header file and in that other header file you are using lv_area_t it is going to work even tho there is not an explicit include of lv_area.h. That is the kind of thing that I am talking about. While the include is technically speaking not needed it does make it easier to understand where things are coming from when looking at the code. This kind of thing should also be done in the C source files as well.

I think it should be done even if it is only for the purpose of code readability. I know it's not going to make the compiled size any larger or anything like that. It will make it easier from a maintenance standpoint because if you are working on lv_color.h and you remove the include to lv_area.h it is not going to end up breaking anything if the include to lv_area.h has been placed into that other header.

kdschlosser, 2024-12-30:
I believe that is the reason why as in your example you see includes that are actually not being used in a specific header file. It is because of the propagation that might be taking place where some other header file is including the one that has the inclusion that looks like it is not being used. but it is being used but not directly. Too much of that kind of a thing make a really nasty tangles mess that can break very easily when changes are made. Sends person of and running to fix what could end up being a cascading code failure caused by a nested inclusion.

kdschlosser, 2024-12-30:
Does Visual Studio only identify if the include is not being used in that specific header or does it also check further up the tree to see if it is being used from propagation?

kdschlosser, 2024-12-30:
Here is an example. `core/lv_obj_class.h` only need to have a single inclusion which is `misc/lv_types.h` there are only 3 things that are needed from that header file. `lv_obj_class_t`, `lv_event_t` and `lv_obj_t`

here are the inclusions that are being done in the `lv_obj_class.h` file

```c
#include "../misc/lv_types.h"
#include "../misc/lv_area.h"
#include "lv_obj_property.h"
```

kdschlosser, 2024-12-30:
It would be nice if LVGL had a Visual Studio Solution packaged with it. I would use it if there was one. I don't want to be bothered with making one. 

The windows simulator is not up to date with the master branch and would need to be updated. Last I knew it was updated to 9.0.0, maybe it has been updated since I last looked at it some 3 or 4 months ago.

kdschlosser, 2024-12-30:
nope still has not been updated...

https://github.com/lvgl/lv_port_pc_visual_studio/releases

vwheeler63, 2024-12-31:
> I believe that is the reason why as in your example you see includes that are actually not being used in a specific header file.

That is certainly possible.

> Does Visual Studio only identify if the include is not being used in that specific header or does it also check further up the tree to see if it is being used from propagation?

I don't know.

> nope still has not been updated...

There is an issue #7412 about the projects not getting updated when LVGL file structure is updated.  However, until that is fully handled, there is an easy work-around:

1. Use Visual Studio to open `LVGL.MaintainerTools.sln`;
2. If it isn't already done, set the project `LvglProjectFileUpdater` as the "main" project (the one that runs when you hit [F5]);
3. Run it.  It updates 3 other `.vcxproj` files against the current LVGL file structure, and does so correctly.
4. Close the solution and then re-open `LVGL.sln` and you will find it now compiles.

kisvegabor, 2025-01-03:
I think it was fixed by https://github.com/lvgl/lvgl/pull/7220

closed 2025-01-28 by lvgl-bot