lvgl/lvgl #7547: Legacy bin image detection

fatihozen3, 2025-01-06:
### Introduce the problem

I use LVGL v9.1.0 and NXP IMXRT1170 in my project. I use raw fs to detect images and fonts on my flash memory.

I use offline image converter with below command
python LVGLImage.py --ofmt BIN --cf ARGB8888 --compress NONE --align 1

I  could use images successfully till I add new 12 images, total binary image size was 14.2 MB before new images, 14.9 MB after new images adding. After adding new images I get warning and error as below, this image were being used successfully before I add other images.
[image]

What should I do ? The main question is why this is happening after adding new images, is it related to size of the bin image or my flash component, I need an advice here.

kisvegabor, 2025-01-15:
Hi,

Do you use indexed images?

kisvegabor, 2025-01-15:
cc @XuNeo

XuNeo, 2025-01-15:
> python LVGLImage.py --ofmt BIN --cf ARGB8888 --compress NONE --align 1

The command line arguments are correct, which conflicts with what the log said. 
So you can firstly verify file content of `F/topbuttoncircle2.bin`, to check if it's same as generated file content.

closed 2025-02-10 by lvgl-bot