lvgl/lvgl #7549: On ESP32, the scale seems to be out of balance, with every component becoming large

fatcatkk, 2025-01-06:
### Introduce the problem

platform = espressif32 @ 6.5.0
board = esp32-s3-devkitc-1
framework = arduino
lib_deps = 
	bodmer/TFT_eSPI@^2.5.43
	lvgl/lvgl@8.4.0

A few months ago, the code was working fine. But recently it suddenly became like this, and no code or configuration was changed in the process.
[image]

closed 2025-01-06 by fatcatkk

kisvegabor, 2025-01-09:
What was the solution?

fatcatkk, 2025-01-09:
> What was the solution?
It looks like I've forgotten that I've changed some of the configurations.I started a new project, followed the normal process, and the system was up and running.