lvgl/lvgl #7551: Suggestions on the Roller Component

zxdgithub001, 2025-01-06:
### Introduce the problem

Can the roller add more elements, such as pictures, and scroll them together with the text? Or is there any other component that can create a list where each item contains pictures and text and can scroll in a loop?      

I hope to receive your reply. Thank you very much.

kisvegabor, 2025-01-15:
Hey,

You can just create use the `flex_column` layout to place the items below each other and use `scroll_snap` to make item snap to the center. (Our docs page is down  for some unknown reason now yo I can't ad links to these. :slightly_frowning_face: )

It won't support the recoloring of the center item though. Is it requirement for you?

lizhaoming634, 2025-02-08:
https://github.com/lvgl/lvgl/pull/7714
Can you try this PR?

closed 2025-03-09 by lvgl-bot