lvgl/lvgl #7552: LVGL9.2.2 lv_file_explorer.c has Minor bugs that affect functionality.

lifeng5478, 2025-01-06:
### LVGL version

V9.2.2

### Platform

[image]
this! After the folder is entered, it cannot be re-entered. I fixed Like This! For your information!

### What happened?

When the file manager enters the secondary directory, it reverts again and cannot be entered again.

### How to reproduce?

static void browser_file_event_handler(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t * obj = lv_event_get_user_data(e);
    lv_file_explorer_t * explorer = (lv_file_explorer_t *)obj;

    if(code == LV_EVENT_VALUE_CHANGED) {
        char file_name[LV_FILE_EXPLORER_PATH_MAX_LEN];
        const char * str_fn = NULL;
        uint32_t row;
        uint32_t col;

        lv_memzero(file_name, sizeof(file_name));
        lv_table_get_selected_cell(explorer->file_table, &row, &col);
        str_fn = lv_table_get_cell_value(explorer->file_table, row, col);

        str_fn = str_fn + 5;
        if((lv_strcmp(str_fn, ".") == 0))  return;

        if((lv_strcmp(str_fn, "..") == 0) && (lv_strlen(explorer->current_path) > 3)) {
            strip_ext(explorer->current_path);
            /*Remove the last '/' character*/
            strip_ext(explorer->current_path);
            lv_snprintf((char *)file_name, sizeof(file_name), "%s", explorer->current_path);

        }
        else {
            **_if(lv_strcmp(str_fn, "..") != 0) {
                if(explorer->current_path[lv_strlen(explorer->current_path) - 1] != '/') {
                    strip_ext(explorer->current_path);
                    strip_ext(explorer->current_path);

                    lv_snprintf((char *)file_name, sizeof(file_name), "%s/%s", explorer->current_path, str_fn);
                }
                else {
                lv_snprintf((char *)file_name, sizeof(file_name), "%s%s", explorer->current_path, str_fn);
                }
            }_**
        }

        lv_fs_dir_t dir;
        if(lv_fs_dir_open(&dir, file_name) == LV_FS_RES_OK) {
            lv_fs_dir_close(&dir);
            show_dir(obj, (char *)file_name);
        }
        else {
            if(lv_strcmp(str_fn, "..") != 0) {
                explorer->sel_fn = str_fn;
                lv_obj_send_event(obj, LV_EVENT_VALUE_CHANGED, NULL);
            }
        }
    }
    else if(code == LV_EVENT_SIZE_CHANGED) {
        lv_table_set_column_width(explorer->file_table, 0, lv_obj_get_width(explorer->file_table));
    }
    else if((code == LV_EVENT_CLICKED) || (code == LV_EVENT_RELEASED)) {
        lv_obj_send_event(obj, LV_EVENT_CLICKED, NULL);
    }
}

lifeng5478, 2025-01-06:
It was meant to be。

uLipe, 2025-01-06:
@lifeng5478 , since you are aware of the fix, would you open a pull request to the upstream? 

Thank you!

ellchr00, 2025-01-07:
Also another issue which I came across is that if you use your QAM (Quick Access Menu)
And choose for example HOME then explorer->current_path will be e.g. C:/home/user/\0
-> changing to FileSystem the new Path C: will be copied (show_dir @611) into explorer->current_path    the char * will then be C:\0home/user/\0
BUT @615 the \0 in position 3 will be overridden by / and we are back to the old path which is wrong.

so either lv_strlcpy has to add an extra \0 but i suggest as safer quick fix to add an \0 when adding the '/'.
Something like ->
@line 615:
if((*((explorer->current_path) + current_path_len) != '/') && (current_path_len < LV_FILE_EXPLORER_PATH_MAX_LEN)) {
        *((explorer->current_path) + current_path_len) = '/';
+        if (current_path_len < LV_FILE_EXPLORER_PATH_MAX_LEN-1){
+           *((explorer->current_path) + (current_path_len + 1)) = '\0';    
+      }
    } 

Although ugly I hope I made myself clear .....

lifeng5478, 2025-01-07:
https://github.com/lifeng5478/lvgl   
You can refer to this！

line 502:
`        else {
            if(lv_strcmp(str_fn, "..") != 0) {
                if(explorer->current_path[lv_strlen(explorer->current_path) - 1] != '/') {
                    lv_snprintf((char *)file_name, sizeof(file_name), "%s/%s", explorer->current_path, str_fn);
                }
                else {
                    lv_snprintf((char *)file_name, sizeof(file_name), "%s%s", explorer->current_path, str_fn);
                }
            }
        }
`
line 611:

`    /*
    size_t current_path_len = lv_strlen(explorer->current_path);
    if((*((explorer->current_path) + current_path_len) != '/') && (current_path_len < LV_FILE_EXPLORER_PATH_MAX_LEN)) {
        *((explorer->current_path) + current_path_len) = '/';
    }
    */
`
This paragraph needs to be commented out

lifeng5478, 2025-01-07:
I don't know what's going on, I can't commit it, so I forked one in my warehouse

lifeng5478, 2025-01-07:
[image]

[image]

ellchr00, 2025-01-07:
OK I will have a look at the commit .... (have only found this thread here). The question is what the definition of a path is and if it needs a '/' at the end or not (and who is relying on it )

uLipe, 2025-01-07:
@lifeng5478 that is the process, if you have the correction, fork the lvgl into your own github account, then commit the changes on your fork, after doing that you may be able to create a pull request for evalution to put the commit on the lvgl upstream.

Please let me know if you have any other question regarding contribution.

C47D, 2025-01-07:
Can you test the same scenario using the latest commit? There have been multiple fixes in the file Explorer widget.

ellchr00, 2025-01-08:
Just from code review, I would state the current 9.3.0-dev code (https://github.com/lvgl/lvgl/blob/master/src/others/file_explorer/lv_file_explorer.c#L621) has similar issues. The code expects a "/" at each directory so in this regard the changes made are not in line .... (there need to be a decision on definition of currentpath "/" at end or not)

1. changing from a a directory c:/users/xy/ via QAM to c: will end into c:/users/xy as there is no \0 added after  the adding of / at lines 621
2. Clicking into empty table will end up with an empty string +"/" added to explorer->currentpath sending it into noman's land.

Just my 2c

C47D, 2025-01-08:
I'm using the `lv_example_file_explorer_1` as reference. From this I understand that the second time I click on the `d3bug` directory I shouldn't be able to open it. Is my understanding correct? Because I wasn't able to replicate the issue.

> When the file manager enters the secondary directory, it reverts again and cannot be entered again.

[Screencast from 07-01-25 19:41:03.webm](https://github.com/user-attachments/assets/25265ae0-7d82-4228-aeb6-704ac6bc903c)

lifeng5478, 2025-01-08:
> I'm using the `lv_example_file_explorer_1` as reference. From this I understand that the second time I click on the `d3bug` directory I shouldn't be able to open it. I wasn't able to hit the bug, maybe I'm missing something.我正在使用`lv_example_file_explorer_1`作为参考。由此我明白，第二次点击 `d3bug` 目录时，我应该无法打开它。我没能找到错误，也许我错过了什么。
> 
> > When the file manager enters the secondary directory, it reverts again and cannot be entered again.当文件管理器进入辅助目录时，它会再次恢复并且无法再次进入。
> 
>  Screencast.from.07-01-25.19.41.03.webm
> 截屏视频.from.07-01-25.19.41.03.webm

You are version 9.3, 9.3 has a lot of problems, it is not recommended, 9.2.2 is acceptable.

ellchr00, 2025-01-08:
@lifeng5478 I looked more closely to your change proposal and found another bug:

Now there is not anymore a "/" at the end of the explorer->current_path and therefore using ".." will jump up 2 directories (2x strip)
Removing one of it would solve this problem, but the question remain if there are other "user" of this variable so I would propose to fix it in a way that the "/" stays at the end for legacy reasons .....

Also in your proposal (at least on my system) clicking outside the table (short directory, str_fn = "") there is still a "/" applied to the end [as due to "%s/%s"] and therefore mixing those 2 systems with possible undesired side effects (e.g. using "..")

Cheers

lifeng5478, 2025-01-09:
> @lifeng5478 I looked more closely to your change proposal and found another bug:我更仔细地查看了您的更改提案，发现了另一个错误：
> 
> Now there is not anymore a "/" at the end of the explorer->current_path and therefore using ".." will jump up 2 directories (2x strip)现在资源管理器 >current_path 的末尾不再有 “/”，因此使用 “..” 将跳转 2 个目录（2 倍条带） Removing one of it would solve this problem, but the question remain if there are other "user" of this variable so I would propose to fix it in a way that the "/" stays at the end for legacy reasons .....删除其中一个可以解决这个问题，但问题仍然存在，如果这个变量还有其他 “user”，所以我建议以一种方式修复它，因为遗留原因，“/”留在末尾.....
> 
> Also in your proposal (at least on my system) clicking outside the table (short directory, str_fn = "") there is still a "/" applied to the end [as due to "%s/%s"] and therefore mixing those 2 systems with possible undesired side effects (e.g. using "..")此外，在您的提案中（至少在我的系统上）单击表格外部（短目录，str_fn = “”）仍然有一个 “/” [由于 “%s/%s”]，因此将这两个系统混合在一起，可能会产生不需要的副作用（例如使用 “..”）
> 
> Cheers  干杯

This may not work, as getting the current path will also result in errors. Did you try modifying that? It should be feasible.

ellchr00, 2025-01-09:
Sorry I don't understand. What do exactly mean by "this" ?
It just depends on how the ending of current_path is defined (with or without "/"). I tried here both versions and they did work for me. 

At the moment I am running your proposed changes plus the above mentioned addition:

[image]
and
[image]

works for me well. The problem I see is that current_path is part of the API :(const char * lv_file_explorer_get_current_path(const lv_obj_t * obj) [But as far as I could see no definition of the return value] and by changing the return value to be without "/" other things my break even if it is not useful to have this "/" at the end.

Me personally I am fine with both solutions it just has to be defined properly and work ...

kisvegabor, 2025-01-30:
Is it stale?

ellchr00, 2025-01-30:
I am no contributor, so I don't know the procedure of commiting/reviewinghere. As far as I know there was a commit prepared (see above), which in my opinion does  break the API. So there is a decision needed if there should be a break in compatibility, namely
the content of const char * lv_file_explorer_get_current_path(const lv_obj_t * obj). The path used to have a "/" at the end and with the new commit it sometimes has and sometimes not.

johnzhy, 2025-02-06:
我用你的方法还是不能重新进入目录，经过调试，我找到了更好的解决方案：
`
606    size_t current_path_len = lv_strlen(explorer->current_path);
607    if((*((explorer->current_path) + current_path_len) != '/') && (current_path_len < LV_FILE_EXPLORER_PATH_MAX_LEN)) {
608        *((explorer->current_path) + current_path_len) = '/';
609       // 添加这句
6010        *((explorer->current_path) + current_path_len + 1) = '\0';
    }
`

kisvegabor, 2025-02-06:
I believe that the file explorer and `lv_menu` should be converted to a demo. They are too specific on their own and they should serve only as a reference implementation for the users. Do you agree with it?

If so, we can really move these to the `demos` folder  so that we can worry less about API breaks.

closed 2025-03-09 by lvgl-bot