lvgl/lvgl #7553: `lv_canvas_set_px()` in indexed `LV_COLOR_FORMAT_I8` case, `shift` is used uninitialized.

vwheeler63, 2025-01-06:
### LVGL version

v9.3

### Platform

any

### What happened?

When the color format is indexed `LV_COLOR_FORMAT_I8`, in the call to `lv_canvas.c::lv_canvas_set_px()`, in the branch starting with

```c
if(LV_COLOR_FORMAT_IS_INDEXED(cf)) {
   ...
```

the value of the `shift` variable is not set before it is used.  It *looks* like it might need to be set to 0, but I am not familiar with the data format, and so need to refer this to the originator of the algorithm:  @kisvegabor .

One other observations:  Since the LV_COLOR_FORMAT_... values are enumerators, a `switch` statement could be much more efficient for this function, for compilers that can optimize `switch` statements into a jump table.

vwheeler63, 2025-01-06:
@kisvegabor Hi.  This will close when the related PR is merged, but before I put this issue to rest, what do you think about the idea of converting that multiple if-elseif-else into a switch statement for efficiency?

kisvegabor, 2025-01-07:
There were 2 `#`s in `Resolves ##7553` in the [PR](https://github.com/lvgl/lvgl/pull/7556#issue-2771328553) therefor the issue wan't closed automatically.

closed 2025-01-07 by kisvegabor