lvgl/lvgl #7570: Loading JPG images from Arduino SD

gilphilbert, 2025-01-08:
### LVGL version

v9.2.2

### Platform

ESP32-S3 DevKit C
Arduino Framework
SD card reader (generic)
3.5" SPI display, using TFT_eSPI driver

### What happened?

I want to load an image file from an SD card connected to the ESP32. The SD card includes music (MP3s) and images. When the MP3 loads, I want to load the matching image from the SD card.

The SD card is already initialized as it uses a custom pin configuration:
```cpp
pinMode(SD_CS, OUTPUT);
SPIClass * sdSPI = new SPIClass();
sdSPI->begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS);
if (!SD.begin(SD_CS, * sdSPI)) {
  Serial.println("Error accessing SD card");
  return false;
}
```

My `lv_conf.h` file includes the following lines. I'm not sure if the SPI frequency or CS pin are required here, since it seems that this has been changed prior to 9.2.2 being released, with the user expected to initialize the library themselves (#6643). I'm using the TinyJPG decoder, so that's enabled too:
```h
/*API for Arduino Sd. */
#define LV_USE_FS_ARDUINO_SD 1
#if LV_USE_FS_ARDUINO_SD
  #define LV_FS_ARDUINO_SD_LETTER 'A'     /*Set an upper cased letter on which the drive will accessible (e.g. 'A')*/
  #define LV_FS_ARDUINO_SD_CS_PIN 48
  #define LV_FS_ARDUINO_SD_FREQUENCY 40000000
#endif
...
/* JPG + split JPG decoder library.
 * Split JPG is a custom format optimized for embedded systems. */
#define LV_USE_TJPGD 1
```

The `lv_image` instance is created at the start of the program:
```cpp
lv_obj_t * o_track_art;
...
o_track_art = lv_image_create(image_cont);
lv_obj_set_size(o_track_art, LV_PCT(100), LV_PCT(100));
lv_obj_remove_style_all(o_track_art);
```

Then I set the image src once the MP3 file is selected and loaded:
```cpp
if (meta.image != "") {  // <---  meta.image = "/filename.jpg"
  String _img = "A:" + meta.image;
  lv_image_set_src(o_track_art, _img.c_str());
}
```

This works fine with a C Array image file using the image converter Python script, so it's somewhere in the Arduino SD library, or the JPG library, I think.

I can't work out what's going on and there's no logging to help, any help is appreciated! Thanks

liamHowatt, 2025-01-10:
Yes, it was changed in v9.2. `LV_FS_ARDUINO_SD_CS_PIN` and `LV_FS_ARDUINO_SD_CS_FREQUENCY` were removed.

Relevant issue: #6643

> I'd simply remove the initialization and the related configs and let it to be done by the user. (We do the same with FatFS)

Relevant PR: #6725 (see what was changed: https://github.com/lvgl/lvgl/pull/6725/files)

https://docs.lvgl.io/master/details/libs/arduino_sd.html

> You will need to initialize the SD card before LVGL can use it (i.e. `SD.begin(0, SPI, 40000000)`).

gilphilbert, 2025-01-11:
Thanks, I have initialized the SD card, and I have successfully read files from it, so I know the connection to the SD card is good.
```cpp
pinMode(SD_CS, OUTPUT);
SPIClass * sdSPI = new SPIClass();
sdSPI->begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS);
if (!SD.begin(SD_CS, * sdSPI)) {
  Serial.println("Error accessing SD card");
  return false;
}
```

I removed the unncessary lines from the `lv_conf.h` file.

So I have:
- The SD card initialized
- `LV_USE_TJPGD` enabled
- `LV_USE_FS_ARDUINO_SD` enabled
- `LV_FS_ARDUINO_SD_LETTER` set to `'A'`
- The source pointing to `A:/filename.jpg` (which I've already checked exists using the SD library directly)

What am I missing? Thanks for any help

gilphilbert, 2025-01-12:
After some investigation, it seems to be something to do with TJPG and LODEPNG, as I can open images in BIN format just fine when I use the `LVGLImage.py` to convert my images. Using JPG or PNG would be much more simple, if anyone has any pointers on why these might be an issue. I don't get any errors, the images just don't load.

One oddity I did discover is that if I include the BIN file in the  image (`lv_image_set_src`) and draw the display with `lv_load_screen()` the screen draw is very slow - maybe 1-1.5 seconds and I can watch the screen draw from top to bottom. But if I set the source for the image _after_ I load the screen, the render is much faster (the screen render is immediate, the image appears maybe .5s later). Not sure if that's clear but this is slow...
```cpp
    lv_label_set_text(o_track_title, _title.c_str());

    if (_img != "") {
      lv_image_set_src(o_track_art, _img.c_str());
    }

    lv_screen_load(o_main_screen); // load the complete screen
```

...and this is fast...
```cpp
    lv_label_set_text(o_track_title, _title.c_str());

    lv_screen_load(o_main_screen); //load the screen

    if (_img != "") {
      lv_image_set_src(o_track_art, _img.c_str()); //now set the source image
    }
```

Thanks!

liamHowatt, 2025-01-14:
Can it load very small JPG images? I want to eliminate memory as a possible limiting factor.

That slowness behavior is strange. I'm not sure why that would happen. :thinking:

gilphilbert, 2025-01-19:
I thought the same, but the image I'm trying to load is only 120x120 pixels and that doesn't load. Would I get errors if that was the issue, or would it just gracefully fail? I'm running `lv_timer_handler();` in a FreeRTOS task. Initially I was getting "canary" issues until I increased the stack depth - when I increased the stack depth the BIN files started showing, but I'm still not getting jpeg images. I will try increasing the stack depth further and see if that makes the jpegs appear.

liamHowatt, 2025-02-10:
> Would I get errors if that was the issue, or would it just gracefully fail?

I am not sure, unfortunately. Ideally you should see errors logged but maybe it was left out.

> I will try increasing the stack depth further and see if that makes the jpegs appear.

Let me know how it goes.

gilphilbert, 2025-02-11:
I just tried it again, I increased stack depth to the point that the ESP ran out of memory and that didn't fix it. I'm not getting any errors either, so I'm not sure where to go at this point. BIN files work fine, but that's not an ideal solution. Any suggestions on how to troubleshoot?

closed 2025-03-09 by lvgl-bot