lvgl/lvgl #7577: Timer (lv_timer)example: error: invalid use of incomplete typedef

w3llschmidt, 2025-01-09:
### LVGL version

9.3.0

### Platform

Handwritten C with FB on Linux (Raspberry)

### What happened?

If i tried this example code:

https://docs.lvgl.io/master/details/main-components/timer.html#creating-a-timer

I got:

`main.c:94:31: error: invalid use of incomplete typedef ‘lv_timer_t’ {aka ‘struct _lv_timer_t’}
   94 |   uint32_t * user_data = timer->user_data;
      |                               ^~
`

### How to reproduce?

`void my_timer(lv_timer_t * timer)
{
  /* Use the user_data */
  uint32_t * user_data = timer->user_data;
  printf("my_timer called with user data: %d\n", *user_data);

  /* Do something with LVGL */
  if(something_happened) {
    something_happened = false;
    lv_button_create(lv_screen_active());
  }
}

...

static uint32_t user_data = 10;
lv_timer_t * timer = lv_timer_create(my_timer, 500,  &user_data);
`

w3llschmidt, 2025-01-09:
My includes:

`#include <unistd.h>
#include <pthread.h>
#include <time.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <locale.h>
#include <fcntl.h>
#include <linux/fb.h>
#include <sys/mman.h>
#include <sys/ioctl.h> 

#include "lvgl/lvgl.h"
#include "lvgl/src/drivers/display/fb/lv_linux_fbdev.h"
#include "lvgl/src/drivers/evdev/lv_evdev.h"

#include "ui.h"
#include "ui_helpers.h"
#include "global.h"

#include "bcm2835.h" // see makefile
`

AndreCostaaa, 2025-01-09:
The example is fixed in #7575 

Simply replace 

```c
uint32_t * user_data = timer->user_data;
``` 
with
```c
uint32_t * user_data = lv_timer_get_user_data(timer);
```

w3llschmidt, 2025-01-09:
Life can be so simple, Thanks :)

PR #7575 (2025-01-13): docs: fix usage of timer user data
EDIT:

Closes #7577
closed 2025-01-09 by w3llschmidt