lvgl/lvgl #7586: SegFault when trying to create a menu within a menu.

phelter, 2025-01-10:
### LVGL version

v9.2

### Platform

Visual Studio Simulator with X11 on linux but should be on any.

### What happened?

Accidentally was creating a menu within a menu.

```c
lv_obj_t *menu = lv_menu_create(lv_scr_act()); // Create the menu on the active screen
lv_obj_t *menu2 = lv_menu_create(menu); // <- this fails with no assert.  Should be checking that you cannot do this.
``
The construction of the second menu fails in the object initialization.

Assuming LVGL would check usage of proper objects being created within the menu. i.e. only the menu_[cont|separator|section]
If you create another object type it crashes without info

Would assume that the `lv_menu_create` would confirm that it isn't being created in an inappropriate place - i.e. another menu.

### How to reproduce?

```c
lv_obj_t *menu = lv_menu_create(lv_scr_act()); // Create the menu on the active screen
lv_obj_t *menu2 = lv_menu_create(menu); // <- this fails with no assert.  Should be checking that you cannot do this.
```

liamHowatt, 2025-01-14:
A menu should be passed to `lv_menu_page_create`. Here is the correct usage:

```c
                auto *menu = lv_menu_create(screenLvObj());
                auto *page = lv_menu_page_create(menu, "menu title");
                auto *obj_in_page = lv_obj_create(page);
```

phelter, 2025-01-14:
Yes I understand that - however the documentation is in the process of being updated to clarify that limitation.  The code also has a bunch of checks for other various things for proper input.  Please add additional checks so that if an improper object type is passed to the `lv_menu_page_create(wrong, "menu title")` that it fails early inside the `lv_menu_page_create` instead of 4 layers down and hard to debug the reasoning for the misuse of the API.  See: https://github.com/lvgl/lvgl/pull/7604/files#r1915485171

phelter, 2025-01-16:
Thanks for this !!!! Much appreciated.

phelter, 2025-01-16:
You can close if you like.

closed 2025-01-16 by liamHowatt