lvgl/lvgl #7589: ESP32 PlatformIO Arduino doesn't work with LV_OS_FREERTOS

eaglesemanation, 2025-01-11:
### LVGL version

v9.2.2

### Platform

ESP32C6 with PlatformIO using [Tasmota platform-espressif32](https://github.com/tasmota/platform-espressif32) for `framework=arduino` support

### What happened?

Due to the change in #6793 build fails with error indicating that atomic.h is missing. Considering that `lv_freertos.h` still uses macros to check for ESP_PLATFORM - I'm not sure why this was done.

### How to reproduce?

`platformio.ini`:
```ini
...
build_flags =
    -D LV_USE_OS=LV_OS_FREERTOS
...
```

This patch resolves compilation issues, so I kinda assume it works as intended:
```patch
18c18,22
< #include "atomic.h"
---
> #if (ESP_PLATFORM)
>     #include "freertos/atomic.h"
> #else
>     #include "atomic.h"
> #endif
```

I'm not sure if I could easily reconfigure include paths through platformio.ini, and I don't want to touch CMake until really necessary, I'm traumatized by it.

liamHowatt, 2025-01-14:
Please open a PR for it!

cc @cosmindanielradu19 any objections?

cosmindanielradu19, 2025-01-16:
> Please open a PR for it!
> 
> cc [@cosmindanielradu19](https://github.com/cosmindanielradu19) any objections?

You had the same concern on #6793 and according to [this](https://github.com/lvgl/lvgl/pull/6793#discussion_r1767498306), I've thought that your suggestion would fix the issue.

nicusorcitu, 2025-01-16:
https://github.com/lvgl/lvgl/pull/6793#discussion_r1770559873 it was fixed on a much proper way. The file name is the same, only the path is different.

What if on other platform the file "atomic.h" has a different path?

eaglesemanation, 2025-01-16:
That might be a more proper solution, I'm honestly not sure, but at the very least it's inconsistent. Here is a check for ESP_PLATFORM that is still in use: https://github.com/lvgl/lvgl/blob/62291fa1744811a0ad6969a65d1f426387719914/src/osal/lv_freertos.h#L26-L34

nicusorcitu, 2025-01-17:
IMO that should be removed. It does not make sense to have such inclusions.

liamHowatt, 2025-02-10:
> You had the same concern on [#6793](https://github.com/lvgl/lvgl/pull/6793) and according to [this](https://github.com/lvgl/lvgl/pull/6793#discussion_r1767498306), I've thought that your suggestion [...]

Thanks for reminding me about it.

@eaglesemanation does the CMake logic not work for you when you compile?

closed 2025-03-09 by lvgl-bot

AndreCostaaa, 2025-05-28:
Hi @uwwint, the proper fix here is to add the "freertos" folder as an include directory to your compiler options using the `-I` flag

uwwint, 2025-05-28:
@AndreCostaaa this comes down to granularity. 
Include flags in PIO are a project wide setting. Meaning that header names can clash because everything is flattened. Granted with freertos headers that *should* not happen - however espidf is vast, arduino-core is vast and then there might be a whole lot of other libraries in the project that don't even know they are running atop of rtos, so I understand the rationale to put rtos in its own directory by esp-idf. 

Right now when you have a PIO project with espidf and libdep on lvgl^9.2.2 it will now not compile out of the box. If -I is the way to go, I suggest reverting [c647c39](https://github.com/lvgl/lvgl/commit/c647c3924b96fe1f0fe44bf19f860d437878adc8#diff-0956730379d2b9c1d9d0526e8ed5c3bb1aa86bbf83ba3b4132abc7b85ed8a5dc)