lvgl/lvgl #7591: There is a zero division error in lv_scale

lizhaoming634, 2025-01-13:
### LVGL version

9.3

### Platform

visual studio simulator

### What happened?

I hope to use scale to implement a loaded animation, when I use lv_stcale_det_master_tick_ every (scale_ loader, 0);
At this time, there will be a zero division error.

### How to reproduce?

static void draw_event_cb(lv_event_t* e)
{
    lv_obj_t* obj = lv_event_get_target(e);
    lv_draw_task_t* draw_task = lv_event_get_draw_task(e);
    lv_draw_dsc_base_t* base_dsc = lv_draw_task_get_draw_dsc(draw_task);
    lv_draw_line_dsc_t* line_draw_dsc = lv_draw_task_get_line_dsc(draw_task);
    if (base_dsc->part == LV_PART_ITEMS) {
        if (line_draw_dsc) {
            int32_t value_of_line = lv_map(base_dsc->id1, 0, 100, 0, 360);
            int32_t ratio = lv_map(value_of_line, 0, 360, LV_OPA_TRANSP, LV_OPA_COVER);
            line_draw_dsc->color = lv_color_mix(lv_color_make(13, 195, 253), lv_color_make(37, 57, 115), ratio);
        }
    }
}
static void time_cb(lv_timer_t* t)
{
    lv_obj_t* scale_loader = lv_timer_get_user_data(t);
    static int i = 0;
    i = (i + 1) % 36;
    lv_scale_set_rotation(scale_loader, 10 * i);
}

void lv_example_scale_8(void)
{
    lv_obj_t * scale_loader = lv_scale_create(lv_screen_active());
    lv_obj_set_size(scale_loader, 300, 300);
    lv_scale_set_mode(scale_loader, LV_SCALE_MODE_ROUND_INNER);
    lv_obj_center(scale_loader);

    lv_scale_set_label_show(scale_loader, false);
    lv_obj_set_style_arc_opa(scale_loader, LV_OPA_TRANSP, LV_PART_MAIN);
    lv_obj_set_style_length(scale_loader, 10, LV_PART_ITEMS);

    lv_scale_set_total_tick_count(scale_loader, 100);
    lv_scale_set_major_tick_every(scale_loader, 0);
    lv_scale_set_angle_range(scale_loader, 360);
    lv_obj_add_event_cb(scale_loader, draw_event_cb, LV_EVENT_DRAW_TASK_ADDED, NULL);
    lv_obj_add_flag(scale_loader, LV_OBJ_FLAG_SEND_DRAW_TASK_EVENTS);

    lv_timer_create(time_cb, 100, scale_loader);
}

lizhaoming634, 2025-01-13:
[image]
[image]
By adding judgments here, problems can be avoided.

[image]
By adding the id1 and id2 of minor_tick_dsc here, I can achieve the desired effect.

lizhaoming634, 2025-01-13:
https://github.com/user-attachments/assets/e1bd4cc3-c9b3-4526-947d-07c21977e171

liamHowatt, 2025-01-16:
I think your fix makes sense! If you are interested please sent a PR, otherwise I will. Thanks!

lizhaoming634, 2025-01-17:
> I think your fix makes sense! If you are interested please sent a PR, otherwise I will. Thanks!我认为你的修复是有道理的！如果您有兴趣，请发送 PR，否则我会发送。 谢谢！

I am not sure if it is possible to add id1 and id2 for minor_tick_DSC, so I may need to trouble you to send the PR.

lizhaoming634, 2025-01-29:
我希望在LVGL9中有类似V8的meter的功能
lv_meter_indicator_t * lv_meter_add_scale_lines(lv_obj_t * obj, lv_meter_scale_t * scale, lv_color_t color_start,
lv_color_t color_end, bool local, int16_t width_mod)
{
LV_ASSERT_OBJ(obj, MY_CLASS);
lv_meter_t * meter = (lv_meter_t *)obj;
lv_meter_indicator_t * indic = _lv_ll_ins_head(&meter->indicator_ll);
LV_ASSERT_MALLOC(indic);
lv_memset_00(indic, sizeof(lv_meter_indicator_t));
indic->scale = scale;
indic->opa = LV_OPA_COVER;
﻿
indic->type = LV_METER_INDICATOR_TYPE_SCALE_LINES;
indic->type_data.scale_lines.color_start = color_start;
indic->type_data.scale_lines.color_end = color_end;
indic->type_data.scale_lines.local_grad = local;
indic->type_data.scale_lines.width_mod = width_mod;
﻿
lv_obj_invalidate(obj);
return indic;
}

[image]

lizhaoming634, 2025-01-29:
I found that the calculation of the scale lines in LVGL9 seems to be inaccurate

[image]

lizhaoming634, 2025-01-30:
static int32_t angle_offset = 0; 

static void draw_event_cb(lv_event_t* e)
{
    lv_obj_t* obj = lv_event_get_target(e);
    lv_draw_task_t* draw_task = lv_event_get_draw_task(e);
    lv_draw_dsc_base_t* base_dsc = lv_draw_task_get_draw_dsc(draw_task);
    lv_draw_line_dsc_t* line_draw_dsc = lv_draw_task_get_line_dsc(draw_task);
    if (base_dsc->part == LV_PART_ITEMS && line_draw_dsc) {
        int32_t tick_idx = base_dsc->id1; 
        int32_t static_angle = lv_map(tick_idx, 0, 100, 0, 360); 
        int32_t dynamic_angle = (static_angle + angle_offset) % 360; 

        int32_t ratio = lv_map(dynamic_angle, 0, 360, LV_OPA_TRANSP, LV_OPA_COVER);
        line_draw_dsc->color = lv_color_mix(
            lv_color_make(0x2B, 0xFE, 0xE4), 
            lv_color_make(0xFD, 0x2B, 0xFB), 
            ratio
        );
    }
}

static void color_anim_cb(void* var, int32_t v)
{
    angle_offset = v;          
    lv_obj_invalidate(var);    
}

void start_color_animation(lv_obj_t* obj)
{
    lv_anim_t a;
    lv_anim_init(&a);
    lv_anim_set_var(&a, obj);
    lv_anim_set_exec_cb(&a, color_anim_cb);
    lv_anim_set_values(&a, 0, 360);   
    lv_anim_set_time(&a, 10000);       
    lv_anim_set_repeat_count(&a, LV_ANIM_REPEAT_INFINITE);
    lv_anim_set_path_cb(&a, lv_anim_path_linear);
    lv_anim_start(&a);
}

void lv_example_scale_8(void)
{
    lv_obj_t * scale_loader = lv_scale_create(lv_screen_active());
    lv_obj_set_size(scale_loader, 300, 300);
    lv_scale_set_mode(scale_loader, LV_SCALE_MODE_ROUND_INNER);
    lv_obj_center(scale_loader);

    lv_scale_set_label_show(scale_loader, false);
    lv_obj_set_style_length(scale_loader, 10, LV_PART_ITEMS);
    lv_obj_set_style_length(scale_loader, 10, LV_PART_INDICATOR);

    lv_scale_set_total_tick_count(scale_loader, 101);
    lv_scale_set_major_tick_every(scale_loader, 0);
    lv_scale_set_angle_range(scale_loader, 360);

    static lv_style_t section_minor_tick_style;
    static lv_style_t section_major_line_style;

    lv_style_init(&section_minor_tick_style);
    lv_style_init(&section_major_line_style);

    lv_style_set_line_width(&section_minor_tick_style, 4U); /*Tick width*/
    lv_style_set_line_width(&section_major_line_style, 4U); /*Tick width*/

    lv_scale_section_t* section = lv_scale_add_section(scale_loader);
    lv_scale_section_set_range(section, 0, 100);
    lv_scale_section_set_style(section, LV_PART_ITEMS, &section_minor_tick_style);
    lv_scale_section_set_style(section, LV_PART_INDICATOR, &section_major_line_style);
    lv_obj_add_event_cb(scale_loader, draw_event_cb, LV_EVENT_DRAW_TASK_ADDED, NULL);
    lv_obj_add_flag(scale_loader, LV_OBJ_FLAG_SEND_DRAW_TASK_EVENTS);
    start_color_animation(scale_loader); 
}
I rewrote an example.

liamHowatt, 2025-01-31:
I think I understand! I opened a PR: #7695. Does it do everything you need?

liamHowatt, 2025-01-31:
> I found that the calculation of the scale lines in LVGL9 seems to be inaccurate

How so?

lizhaoming634, 2025-02-04:
> I think I understand! I opened a PR: [#7695](https://github.com/lvgl/lvgl/pull/7695). Does it do everything you need?我想我明白了！我打开了一个 PR： [#7695](https://github.com/lvgl/lvgl/pull/7695) 。它能满足您的所有需求吗？

Thank you for the PR! It perfectly covers all my requirements.

lizhaoming634, 2025-02-04:
> > I found that the calculation of the scale lines in LVGL9 seems to be inaccurate我发现 LVGL9 中刻度线的计算似乎不准确
> 
> How so?  怎么会这样呢？

I'm not entirely sure, but in LVGL8, the coordinate calculations for meter icks seem to be handled with higher precision.

[image]

liamHowatt, 2025-02-28:
I see there is a 1px gap. Maybe it can be improved if we change the default tick offset by 1?

PR #7695 (2025-03-25): feat(scale): add id1 id2 for tick line draw descriptors. fix zero division
Fixes #7591

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code- [... cut]
closed 2025-03-25 by kisvegabor